arendst/Tasmota #1925: Connecting to MQTT by a hostname

mpapierski, 2018-02-15:
Hi!

I'm running MQTT server on my Raspberry PI on local network. Unfortunately in my home network DHCP leases on wifi are quite short. So often every few days the leases are expiring and IP address of my RPi is changed.

To solve this problem I configured my raspberry pi to have a hostname "raspberrypi.local" which is perfectly fine if I'm trying to reach it through other computers in the network. I tried to set "raspberrypi.local" as MQTT host in my Sonoff devices, but unfortunately Sonoff Tasmota is not able to resolve this hostname and therefore its unable to connect to it. The only workaround I found is to advertise MQTT server on local network through Avahi and reset MqttHost on all devices:

```sh
for device sonoff{1..10}; do curl http://$device.local/cm?cmnd=MqttHost%200; done
```

This makes Sonoff a lookup in Avahi to find mqtt service in local network. And it works. Until there is another IP address change on RPi. Then Sonoff Tasmota doesn't do Avahi lookup again because IP address is cached.

Ideally, though, I'd like my devices to connect by a hostname instead of MQTT host, but for some reason it doesn't work. 

Logs from the device itself:

```
21:09:15 MQT: Attempting connection...
21:09:21 MQT: Connect failed to raspberrypi.local:1883, rc -2. Retry in 10 sec
21:09:32 MQT: Attempting connection...
21:09:38 MQT: Connect failed to raspberrypi.local:1883, rc -2. Retry in 10 sec
21:09:49 MQT: Attempting connection...
21:09:55 MQT: Connect failed to raspberrypi.local:1883, rc -2. Retry in 10 sec
21:10:06 MQT: Attempting connection...
21:10:12 MQT: Connect failed to raspberrypi.local:1883, rc -2. Retry in 10 sec
21:10:23 MQT: Attempting connection...
21:10:29 MQT: Connect failed to raspberrypi.local:1883, rc -2. Retry in 10 sec
21:10:40 MQT: Attempting connection...
21:10:46 MQT: Connect failed to raspberrypi.local:1883, rc -2. Retry in 10 sec
21:10:57 MQT: Attempting connection...
21:11:03 MQT: Connect failed to raspberrypi.local:1883, rc -2. Retry in 10 sec
21:11:14 MQT: Attempting connection...
21:11:20 MQT: Connect failed to raspberrypi.local:1883, rc -2. Retry in 10 sec
21:11:31 MQT: Attempting connection...
21:11:38 MQT: Connect failed to raspberrypi.local:1883, rc -2. Retry in 10 sec
21:11:48 MQT: Attempting connection...
21:11:55 MQT: Connect failed to raspberrypi.local:1883, rc -2. Retry in 10 sec
```

I know a little about programming so I'd love to send a pull request if anybody has time to point where should I look to do a fix.

BTW I'm using Sonoff-Tasmota 5.11.1 downloaded from the Releases page.

Chipz0r, 2018-02-15:
Why not set your raspberry pi to a static ip address ?

mpapierski, 2018-02-15:
@Chipz0r Obviously this will work. I think using hostnames is pretty basic solution for such problem as well.

Chipz0r, 2018-02-15:
Of course, valid solutions.

zerog2k, 2018-02-16:
https://github.com/arendst/Sonoff-Tasmota/blob/development/sonoff/sonoff.ino#L575-L579

mpapierski, 2018-02-16:
@zerog2k Thanks for response

Not sure if it should be a special case i.e. resolve ("discover") or not resolve. I'd go for always trying to resolve a configured hostname, and then connect to the resolved IP address without overriding a configured mqtt host. If someone configures MQTT with an IP address then the resolve is basically a noop. What do you think?

RaymondMouthaan, 2018-02-17:
I am not sure whats goes wrong in your setup, but this i my setup:

- For each sonoff tasmota device i have a dhcp binding (reservation) configured in my dhcp server based upon mac address, so that a device always gets the same ip address.
- Same thing goes for the MQTT server running is a docker container on a raspberry pi. In the dhcp server i have a dhcp binding for the rpi, based upon its mac address.
- Also i've set up a dns server where for each sonoff a hostname is defined like mysonoff.mydomain which resolves with it's ip address. Same thing goes for the MQTT server ie. mymqtt.mydomain.

Using this setup i can access the web interface of the sonoff using http://mysonoff.mydomain. In the MQTT config screen, i have entered mymqtt.mydomain as hostname so that the sonoff connects to the mqtt broker by its name. This works perfectly. The Tasmota firmware i am using is Sonoff-Tasmota 5.12.0b.

zerog2k, 2018-02-18:
Thinking about it in a different way, the mqtt server is more or less expected to be a "fixed" server resource. It's IP probably _should_ not change throughout the lifecycle of a unit. Seems like a changing mqtt server ip could be problematic for other mqtt clients, e.g. iot devices, in general.

In general using DHCP for a _server resource_ isn't great, but if it's done, it's usually with a DHCP static lease, i.e. best of both worlds: ip remains static (never changing), but host can use dhcp to set it, and it's controlled centrally at dhcp server. Are you able to set a static dhcp lease on your mqtt server with your router?

tobox, 2018-02-19:
Using DNS-Names instead of IP addresses is almost always the better solution. Especially when it comes to renumbering (IP prefix changes), or service separation (move mqtt service to a different machine).

mpapierski, 2018-02-19:
Those are all valid points regarding configuring DHCP leases, or static IP address on RPI. On the other hand I still insist that there is a problem in the firmware as described in my earlier comment.

To recap:

- configuring MQTT host with "raspberrypi.local" doesn't work.
- configuring MQTT host with '' (empty string) does mDNS resolution BUT the IP address is cached, and later reconnecting is done to the cached IP address.

I still think MQTT hostname should be always resolved to an IP address, and the IP address should never be cached, so reconnections would be perfectly fine.

The case with empty MQTT host (mDNS lookup) is not necessary and should be explicit option (a checkmark "Do a lookup for MQTT service in local network"). And every time it should be discovered on reconnect.

SGF-Lon, 2018-02-19:
> thread[2].>

Tasmota 5.10.0 works fine with the following definition in
user_config.......
#define MQTT_HOST            "xxxxxx-mqtt"          // [MqttHost]

Links:

  1. https://github.com/arendst/Sonoff-Tasmota/issues/1925#issuecomment-366659367
  2. https://github.com/

mpapierski, 2018-02-24:
@SGF-Lon thanks for response - but this doesn't solve the problem. The explained issue will be still the same.

amontefusco, 2018-12-04:
>>>does mDNS resolution BUT the IP address is cached, and later
>>>reconnecting is done to the cached IP address which is wrong 
>>> because IP address of the service might be changed already.
This feature/bug is still there with the last release.

Moreover, I want to stress a couple of cases in support of mDNS search, and above all SD-DNS discovery, of the MQTT server:

1. isolated networks: i.e., networks operating on an ISOLATED broadcast domain: in such environment no services are available, just mDNS and SD-DNS, so the end nodes should be able to reach the MQTT server using Service Discovery; this is my case (avahi-browse -D -a -r output shown)

= wlan0 IPv4 Mosquitto MQTT server on cm3home _mqtt._tcp local
  hostname = [cm3home.local]
  address = [169.254.55.66]
  port = [1883]
  txt = ["info=Publish, Publish! Read all about it! mqtt.org"]

so, here two  features are in order:
a. discover the service _mqtt._tcp
b. assign a link-local IPv4 to the interface when DHCP  is not responding

2. even in case a DHCP service is available, discovering the MQTT broker via SD-NDS is very useful where you have more than one broker to implement redundancy.

cades, 2018-12-05:
hope MQTT host domain name support +1

ascillato, 2018-12-05:
already implemented and working. please see wiki at rules or the full explanation at: https://github.com/arendst/Sonoff-Tasmota/issues/4527

dachshund-digital, 2021-05-15:
Maybe implemented... but does NOT work.  My entire network is integrated with FQDN, I don't use static IPs and am in the process of retiring IPv4 completely.  Then I start using the tasmota firmware on ESP8266 based devices, and even the latest image 9.4 refuses to work with a valid FQDN.  I use REAL DNS in a master/slave configuration, not mDNS, no SD-DNS, I have a REAL DHCP server that provides IPv6 addressing.  But tasmota only works if I use the actual IPv4 address to my broker... really?  Come on, support real solutions, that are RFC based.  IF I enter a valid FQDN name in the host name field IS SHOULD ALWAYS RESOLVE.  From what I read from the reference issue above, that is NOT implemented?  I have to leave the entry blank or some such?