arendst/Tasmota #1926: Sonoff Basic HASS Device Type when using Discovery

MichaelSchemmer, 2018-02-15:
Hi there,

Firstly, thanks for this awesome firmware! It's really working well in my environment!

I've a question regarding the HomeAssistant Discovery component. It works and my HASS instance detects the device, but it picks up as a switch and its role is a light. 

The web console on the Sonoff displays the following:
MQT: homeassistant/switch/ESPEntrance/config
Is there any way to change the switch value to light? Or is there a  MQTT configuration topic that can be changed on the Sonoff?
I understand that the discovery component sends all the config data to the HASS instance.....

Thanks alot and keep up the great work!

rlust, 2018-02-15:
I am trying to use this in hassio also. How are you seeing the MQTT info in the sonoff for the hass config? I have set OptionSet 19 to on. Do you have the device set as a Wemos?

Thanks!

MichaelSchemmer, 2018-02-15:
Hi there, you can login to the built in webserver and restart the module. Once you've done so, click on the console tab and the info will then be displayed under MQT: homeassistant/switch/sonoff/config

rlust, 2018-02-16:
Hello
Did you have to do anything in hassio?

Thanks!

MichaelSchemmer, 2018-02-16:
Oh, yeah, sorry, add the following to your MQTT broker (I used the internal one):
  discovery: true
  discovery_prefix: homeassistant

pascalsaul, 2018-02-18:
Duplicate from: https://github.com/arendst/Sonoff-Tasmota/issues/1784

emontnemery, 2018-02-18:
See comment in #1784.

closed 2018-05-23 by MichaelSchemmer