arendst/Tasmota #1928: Problems with 2 pull up resistors on one sonoff basic

mestes136, 2018-02-15:
I am trying to use a reed sensor and a switch on one sonoff basic. The light switch is on gpio 14 and the reed switch is on gpio 3. When one is switched on they are both activated. My theory is because it’s backfeeding through one resistor to 3.3v pin then to the other pin. Is there a way around this? Pull down resistor on one and pull up on the other maybe? Bypass caps?

Frogmore42, 2018-02-15:
How do you have it wired and how do you have it configured?  What size (ohms) resistors are you using?
It is not a problem to have multiple pull up resistors as long as they are wired correctly.

mestes136, 2018-02-15:
10k and one end of both to the pin closest to the button and the other to the pin in use. I know there has to be another way to wire it my brain just isn’t coming out with it.

mestes136, 2018-02-15:
Would adding a diode to each solve this or is there a simpler solution?

Frogmore42, 2018-02-16:
Normally, a pull-up goes from the positive voltage supply 3V3 in this case to one pin.  You need another pull-up for the other pin.  This tutorial might help you visualize it: https://learn.sparkfun.com/tutorials/pull-up-resistors

mestes136, 2018-02-16:
Yes that is how I have made one pull up, when needed, work up until this point. But now I need a pull up to 2 pins so useing 2 different 10k resistors, one on gpio 14 and one on gpio 3 with the other ends of the resistors both on 3.3v, when only one switch/sensor is activated thay are both being read as activated........That’s a much better web site then the one I had to read 3 times to understand btw!

mestes136, 2018-02-16:
i Have found ONE of the solutions. I moved the sonoff closer to the wall switch so the resistor was no longer needed on that one and just have one circuit with a pull up. Still think there has to be another solution to this problem though. thank you for your help and the better reference site too!!!

Frogmore42, 2018-02-16:
If you have a switch on a GPIO, you should always have a pull up on the input pin and have the switch short the input pin to ground, or a pull down and short to VCC/power (but the pull up is usually better).  If you don't have a pull up the pin will be floating and might register as zero or as one when the switch is open.  Some MCUs have switchable pull UPS but they are generally very weak and will not work well with long wires.

I still don't understand how you had it hooked up and if your problem is hardware or software.