arendst/Tasmota #1929: witty 8266 analog0, Need help

Igor122017, 2018-02-16:
Good day. I need help, I am do tasmota and got the ADC data from the photoresistor and also set the motion sensor on GPIO14 - the data is (the lighter the ADC parameter is, the adjustment range is from 0 .... 1024.) Now the question is how to manage these data, how to make it so that when the> 300 the motion sensor sets GPIO14 to 1, and if the value is <300 GPIO14, 0 is left ... help please!

Igor122017, 2018-02-20:
help me anybody, how to make it so that when the value of the photoresistor (analog0)> 300 - GPIO14 - when driving was 1 and at a value of <300 - with motion remained GPIO14 - 0?

Oxyandy, 2018-02-20:
NodeRed, Domoticz etc == MQTT clients with a broker to interpret the data -- blockly or another script to parse the resulting output

Igor122017, 2018-02-20:
all right, through MQTT it can be done, it's clear to me and I know .... the question is different, can Witty work with these data without MQTT and other brokers?

Igor122017, 2018-02-20:
I mean, is it possible to resolve the analog0 in this firmware on the basis of the data and/or forbid the output of 1/0 to GPIO14?

closed 2018-03-01 by Igor122017