arendst/Tasmota #1933: Sonoff to Control Up/Down Motor

areid1234, 2018-02-16:
Hi all,

Sorry to start this discussion again!

I have a motor which raises and lowers a TV. The TV turns on when raised, the motor stops at the top and bottom of the lift.

I would like to add a Sonoff/Sonoff Dual to the circuit. I saw on a previous note in this thread that a dual could switch "on" the up and switch "on" the down. I have already bought a Sonoff with a singular relay and have placed this within the circuit in several places but with no luck!

I don't know what the brains of the board do but it effectively uses the RF transmitting to a remote to move this up and down.

I am a complete novice but i anyone has any advice and which wires to connect to the Sonoff to get this going I would be grateful! I have also ordered an RF version which could work but I would prefer this wired in

Thanks!

[image]

[image]

[image]

[image]

reloxx13, 2018-02-16:
Dude, its rare to see so much garbage.

Put on your shoes on and drive to a hardware store and buy f**king isolation tape for a few cents.

230Volts are not a game!

does the up/down switch stay in his up/down position or moves it back to center?

[image]

https://www.conrad.de/de/relaisplatine-bestueckt-1-st-conrad-components-rel-pcb6-1-1-wechsler-230-vac-503334.html

reloxx13, 2018-02-16:
if sonoff is on, relais will "simulate" a button UP press and hold.
if sonoff is off, relais will "simulate" a button DOWN press and hold.

areid1234, 2018-02-23:
Hi Reloxx

Thanks for your diagram. I have wired this circuit up as below with the Relais you have recommended. Sonoff is on, circuit board LED is on, motor moves up and down but the relais doesn't seem to be working. I bought 2 relais and they both aren't working. Can you see any problems with my diagram that I could be doing wrong? Or any other issues it may be?

Thanks again for your help!

[image]

reloxx13, 2018-02-23:
> does the up/down switch stay in his up/down position or moves it back to center?

areid1234, 2018-03-06:
When the Sonoff is ON to give power to the relay, it switches the lift up (I have wired up to the NC) but it does nothing when you turn the Sonoff OFF. The remote works the same still!

areid1234, 2018-03-06:
I have also tried this with the Dual. But this is how it seems to work...

Relay 1 ON no action
Relay 1 OFF Lift moves up

Relay 2 ON No action
Relay 2 OFF No action

[image]

radsimu, 2019-10-21:
the problem with dual is that you may end up powering both state of the motor, and the circuit board might not be prepared for that. Did you manage to somehow configure the sonoff to avoid that? (have it automatically lock the other channel, when a channel is opened - is that even possible with Sonoff?)