arendst/Tasmota #1935: Sonoff Basic won't receive an IP

s1572, 2018-02-16:
I flashed my Sonoff Basic with Tasmota 5.12 using the Arduino IDE 1.8.5, the supplied libraries, esp8266 v2.4.0 and a USB-TTL adapter without any error message.

On the first try it connected successful and I even could access the web interface to adjust the mqtt config. But since a restart of my sonoff basic it just won't receive an IP from my router.

I tried to reflash it with different Tasmota, IDE and esp8266 library versions without success, still the same fault. I even tried different power supplies, mains power and I even started over with a default user_config with only my SSID and password. A configuration via WPS shows the same fault.

I even changed the 2.4GHz channel, restarted my router, installed the newest firmware and turned off the 5GHz - still the same.

It shows the name as "PC-MA-CA-DR-ES-S1-23" (with the real mac of course) so it wont even receive the right name for the sonoff. I tried to manual configure the IP via the user_config and via the router web interface, but that didn't help either.

I attached a screenshot of my router interface, I can translate it if needed:
https://i.imgur.com/6yBJfwc.png

ferdydek, 2018-02-17:
Don't you think your router may be at fault here? Maybe you disabled DHCP on it? Or added the mac addresses mask of ESP to blacklist on DHCP ? 
Did you try on another sonoff so we can confirm the issue isn't linked with particular unit ?

s1572, 2018-02-17:
I though about that, but I don't think so. I rechecked the DHCP settings and even extended the address space to ensure that there is an IP "left" for the sonoff. The Mac blacklist is empty and set to "allow all", I even tried to assign a specific (free) IP to this Mac address.
Unfortunately I don't have another sonoff device on hand, I ordered a S20 but that'll take a while to arrive. I tested a new orange pi zero and a raspberry pi 3 but they worked without a problem.

jeylites, 2018-02-18:
Downgrade you esp8266 v2.4.0 library to V2.3.0 seem to fix the issue.

s1572, 2018-02-18:
I already tried that. My current config is:
> sonoff 5.12
> esp8266 v2.3.0
> Adafruit BME680 v1.0.5
> Adafruit Sensor v1.0.2.02
> ArduinoJson v5.11.2
> IRremoteESP8266 v2.2.1.02
> NeoPixelBus v2.2.9
> OneWire v2.3.3.06
> PubSubClient v2.6.09
> TasmotaSerial v1.1.0
> TSL2561 Arduino Library v1.0.0

So I basically just the libraries from the current release.

alton76, 2018-02-18:
What Arduino version? I had issue with IDE 1.8.5 on OSX

s1572, 2018-02-18:
I'm running 1.8.5 on Windows 10. I tried it with the same config and IDE 1.8.4 with the same result, any suggestion which IDE version should work for me?

jeylites, 2018-02-18:
some steps I could think of...

1) Change CFG_HOLDER in user_config and re-flash the device 

Eg: say currently it's 0x20161209, change it to something else 0x20161205, then re flash

[CFG_HOLDER](https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting#config-problems-can-cause-boot-loops-items-set-in-user_config-to-not-appear-etc)

2) Clearing the configuration flash area

[ESPTOOL](https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting#esptool-usage)

s1572, 2018-02-18:
Thank you! That worked for me. I can't tell which one, since I tried both at the same time, but my sonoff finally get's an IP again.

closed 2018-02-18 by s1572