arendst/Tasmota #1939: Sonoff SC Sensor Data JSON Format

reloxx13, 2018-02-16:
Heya, 

just got the Sonoff SC JSON from you and it looks like this:

```
  "StatusSNS": {
    "Time": "2018-02-16T16:18:49",
    "Temperature": 25,
    "Humidity": 83,
    "Light": 10,
    "Noise": 20,
    "AirQuality": 100,
    "TempUnit": "C"
  },
```

Would it be possible to change it to the layout all other sensor got?

e.g.:
```
  "StatusSNS": {
    "Time": "2018-02-16T16:18:49",
    "SonoffSC": {
        "Temperature": 25,
        "Humidity": 83,
        "Light": 10,
        "Noise": 20,
        "AirQuality": 100
    }
    "TempUnit": "C"
  },
```

Other sensors: [JSON Status Responses#Sensors](https://github.com/arendst/Sonoff-Tasmota/wiki/JSON-Status-Responses#sensors)

arendst, 2018-02-17:
Will do

closed 2018-02-17 by arendst

pablobolomey, 2018-08-26:
What way do i get the single sensor values from the sonoff SC 
 state_topic: 'office/sonoffSC01/SENSOR'
    value_template: '{{ value_json.SonoffSC.xxxxxxxx] }}'

ascillato, 2018-08-26:
Information in the wiki at https://github.com/arendst/Sonoff-Tasmota/wiki/Home-Assistant#dht22-sensor