arendst/Tasmota #1941: Pow & DS18x20 possibly?

AlexTransit, 2018-02-17:
If I connect a temperature sensor to "Sonoff Pow" will it work?
reconfigure the config is not a problem.
a little later I will try and write.

reloxx13, 2018-02-17:
why not?^^

smadds, 2018-02-17:
The issue with the POW is that, unlike the other Sonoffs, the logic circuitry is not isolated from the main voltage. Using any sensors therefore poses a safety issue, as they are not normally insulated enough to protect from mains voltages. I believe that for this reason the sensor options have been disabled on the POW

reloxx13, 2018-02-18:
so where does it get power from if it is isolated 👎 

just connect a sensor on one of the gpios and go. its an esp like any other sonoff.

tobox, 2018-02-19:
Don't connect any sensors to POW unless you really know what you are doing. 

Chances are 50% that you get an electric shock (230V) touching your DS18x20.

The power meter IC makes the POW a special sonoff, as it has no clean separation between the AC mains and the ESP Vcc.

tobox, 2018-02-19:
> just connect a sensor on one of the gpios and go. its an esp like any other sonoff.

No, it is not. Compare the schematics available from Itead.

AlexTransit, 2018-02-19:
The power section is different only with a smoothing filter
Sonoff Basic - half (only C8)
Sonoff Pow - full (C2 & C3)

https://www.itead.cc/wiki/images/5/52/Sonoff_POW_Schematic.pdf
https://www.itead.cc/wiki/images/f/ff/Sonoff-Schematic.pdf

smadds, 2018-02-19:
On page 3 of the POW schematic you can see the "L" live mains input connected to the HLW chip via a 1k resistor. The chip does not isolate the incoming mains from the logic level outputs, so is dangerous.

230V through 1k would give up to 230mA - many time the safe value of a leakage trip, and enough to do you serious damage.

As @tobox says - if you don't know what you're doing stay clear!

As a suggestion - why not separate your sensors from your controls? It means you can put the sensors at the right position without long cables. I've been building boxes around the Wemos D1, BMP280 and PoE power converters (I have high level data sockets in the building).  

[image]

AlexTransit, 2018-02-20:
connect any sensors to Pow  very bad idea.
internal shematic connected Line to local ground! 
this can be seen in the photo.
[image]

closed 2018-02-20 by AlexTransit

SimoneBnc, 2018-05-13:
Isn't it the same for the basic Sonoff to have the GND connected to the Neutral?

smadds, 2018-05-14:
Sonoff basic is a 2-wire device, so no possibility of connecting ground. It's a shame, since it makes it harder to connect it inline with 3-wire devices. (Best way I have found is to not cut the cable and have the earth running inside the box.)

The issue with neutral is more complex. Neutral is actually ground - it's connected to the sheathing of the mains cable laid underground by your electricity supplier and is connected to ground at the power supply end of the cable. You can touch neutral at any time and, as long as there are no faults (i.e. it's not floating) you'll be fine. 

There are, however, 2 problems:
1) In much of the world you cannot guarantee which pin will be live and which neutral. (e.g. connecting Sonoff basic to a 2-pin EU or US plug it's a 50:50 chance!)
2) Residual current breakers look for matching current going from the live and returning to neutral. Any imbalance will trip the breaker.

It's one of those "in theory it should be ok", but in practice there are just too many potential safety issues for any sane person to consider doing it!