arendst/Tasmota #1943: Cannot connect to wifi. Password max length or SSID character?

TimTaler007, 2018-02-17:
I could not connect the Sonoff with my regular wifi network. It works with the guest wifi.
Is there a known problem with the length of the wifi password or special characters in the SSID?
My regular wifi uses 38 signs in the password. Only numbers and normal letters, no special character or blanc. 
The not working SSID is "WLAN_Router_HOME" and password like 
"908214309823424029didyoutrytohackwifiX"
The working guest wifi SSID "WLAN Guest" and password like "theguestwifi". Password is much smaller and SSID has a blanc instead of underline.

Jason2866, 2018-02-17:
Try a shorter password and you will see.
The SSID ist not the problem my SSID is nearly the same as yours: Jason_Home_WLAN

TimTaler007, 2018-02-18:
Thanks for your reply. Sounds like a problem with the password. I have to take time to change the password because so many devices must be updated too...not the funniest job to type so long passwords with the TV remote for example.

mad-ady, 2018-05-02:
It seems I too hit this bug. My paranoid wifi has a 64 character password (drat, now the hackers know it too!), and the sonoff won't connect to it (it connects to the secondary wifi which has a shorter password). I've checked the code and in version 5.12 wifi passwords are saved in a 65 character array (defined in settings.h, as part of struct SYSCFG):

```
char   sta_ssid[2][33];
char   sta_pwd[2][65];
```
The last character must be there to accommodate a null (0x00), so as I see it passwords of 64 characters should work. But in my case it doesn't.

I will try to change the defaults, build and flash a custom version and test it a bit to see how it goes. I suspect that the configuration settings will be messed up afterwards because they will live at wrong offsets...

mad-ady, 2018-05-03:
I ran some tests today and it seems it works without issues. I also have an AP with a space in the middle of the SSID and I was worried it wasn't supported. So I created a new Access Point with no space and the old 64 character password and forced the sonoff to switch wifi connections and it worked. I also tried with the SSID with space in it and it also worked.
The only thing is - the sonoffs want to connect to WIFI2 by default. That's not a problem as long as they can connect to WIFI1 when needed.

closed 2018-07-10 by ascillato2

whitehoose, 2020-01-23:
I've just finished upgrading my network, Old was virgin media hub connected to an old BT router running WRT and configured as an access point.  I have 5 sonoff switches connected to various stuff and all worling fine - my ssid is a single word (no spaces) same on both router and old AP.
 Passwird is a snappy upper/lower case phrase "i like to B3 different".  Been working for ages.

THEN I added a comfast 300 mb/sec ap - and existing switches worked for a few hours then the new AP locked up and all the sonoffs stopped working,  Unable to add new.
 I introduced a new SSID for the new AP - and the old sonoffs started working - bbut still couldn't add new.  Removed comfast - everything working as always.

I need the new AP - so settled for 2 SSIDS and no new switches for the time being but moving from house to garage is complicated if I'm logging in to a site as I walk between zones

bugith, 2021-03-07:
We incorrectly use the word "key" instead of "passphrase", because the key itself is always 64(**hex**) characters in length whatever is the length of the passphrase. The WPA**2** passphrase length must be between 8 minimum and 63 maximum, each included, and to build the passphrase we can use any of the [95 ascii-1967 printable characters](https://en.wikipedia.org/wiki/ASCII#Printable_characters) from 20(**dec**) to 126(**dec**) included.
Here is a [converter](http://jorisvr.nl/wpapsk.html) passphrase to KEY, usefull for situations where we have not a full keyboard or we have problems with characters we want to use.
Any device not complying with this [IEEE802-11i](https://en.wikipedia.org/wiki/IEEE_802.11i-2004) standard is/was designed for the trash can, e.g. the RA10X50_C_S ipcam and IcSee companion management application.