arendst/Tasmota #1945: Sonoff T1 2ch hold button

ageorgios, 2018-02-17:
When holding a button on the Sonoff T1 (EU) it beeps and restarts around on 3-4 seconds.

This is my configuration:
backlog setoption13 1; setoption0 0; setoption32 100;

ascillato, 2018-02-17:
Hi

What is the question? 

As in the wiki, it is a feature of this firmware

Please, see https://github.com/arendst/Sonoff-Tasmota/wiki/Button-usage

ageorgios, 2018-02-17:
wow you are right.

Is there a way to disable this?

ascillato, 2018-02-17:
Yes.
Read the wiki
;)

ageorgios, 2018-02-17:
please help is a little bit long

ascillato, 2018-02-17:
There are a lot of options for the behaviour of the button. As I don't know what you want just go to https://github.com/arendst/Sonoff-Tasmota/wiki/Commands and check for the one that it is better for you.

ageorgios, 2018-02-17:
I think there is no way of disabling the reset behaviour

emontnemery, 2018-02-18:
Use command setoption13 1

closed 2018-02-19 by ageorgios

SergiusTheBest, 2021-04-02:
@ageorgios Did you manage to disable those beeps?