arendst/Tasmota #1947: Sonoff touch 3-gang triggered by rf signals

MaximeSt, 2018-02-17:
Hi,

I have a Sonoff touch with tasmota installed. When you long press a button it goes into rf learning mode. By normal usage it learned the rf signal of other sockets but I can't seem to delete or unlearn these signals. Now the sonoff touch is unusable. 

My question is: How can I unlearn al the rf signals?

Thank you for the help!

Jason2866, 2018-02-17:
Reset 2
This command will reset your device and deletes all the data and configs

MaximeSt, 2018-02-19:
I did this command in the console of the device and it reseted, but the rf signals still trigger the buttons.

arendst, 2018-02-19:
Reset 1/2 won't work for RF signals as they are controlled by a second microchip. You'll need to follow iTead RF procedures and press a button somewhere on the board.

MaximeSt, 2018-02-19:
Ok I will try that, thank you for your help!

MaximeSt, 2018-02-19:
@arendst , do you have an idea where I can find this documentation? I searched before for this but only got information on how to pair rf devices and not how to unlearn them on the sonoff touch. Thanks in advance.