arendst/Tasmota #1948: Sonoff T1 EU 2-Gang reboots every few seconds, settings like module type are lost.

qorron, 2018-02-17:
this is my first T1 (already flashed about 40 sonoff devices, 30+ are installed in my flat and work well)
I manage to set it to module 29 (T1 2-Gang), it reboots as T1 but soon after that, regardless if I press a button or not, it resets and comes back as sonoff basic.
after that it keeps rebooting every few seconds.

what am I missing?

jeylites, 2018-02-18:
What Tasmota version are you running?
Did the default reboot occur while connected to the computer USB power or at install?

From time to time, I run into a similar issue. I'm trying together some info if you having a different setup.

My Setup 
Tasmota 5.12b
Arduino 1.8.4 OSX
ESP Board V2.30

Herschdorfer, 2018-02-18:
I have a T1 3-Gang and i have similar issues. I can configure it as a 3-gang and MQTT etc. but after some time i just get a reboot and all settings are lost.

emontnemery, 2018-02-18:
It's due to long press detected on one of the switches.
 Are you powering from AC or via the debug port? I see this issue frequently when powering from the debug port.
Send command setoption13 1 as suggested in the wiki.

Herschdorfer, 2018-02-18:
AH that could be.
I am emulating the relay board and power via 5V adapter. Does it need a higher voltage?

emontnemery, 2018-02-18:
I actually suspect it's some sort of watchdog function to detect malfunction of relay board when using original firmware.
According to the wiki the relays are indirectly controlled by additional switch, and the additional switch can sense trust state.

Herschdorfer, 2018-02-18:
Hmm that's not good. I wanted to use is with my own relay and power...

Ok then I need to reverse engineer this.

Cysign, 2018-02-18:
As far as I know the esp8266 has to be powered be 3.3V. Using 5V might damage your chip!
Or does the T1 have a buck-converter for 5->3.3V?

Herschdorfer, 2018-02-19:
The T1 has a 8285 but this one is also 3v3. But the board has a converter. When you power it via J2 it's fine. 

I updated the Wiki also about the J2 yesterday.

qorron, 2018-02-19:
the other 2 T1 from the same order worked just fine.
I reflashed the faulty one just to be sure but the behaviur did not change.

I had all modules hooked up to the programmer all the time.
will try the faulty one with an AC connection later.

detail on the side: my programmer only switches to 3.3V on the data pins, not the vcc pin.
however... I managed to program all ob my modules with it and haven't had any problems yet.
the modules report the vcc as 4.0-something, perhaps my programmer being so crappy, it can't hold 5V, made that possible. I tried to get a good one but they were hard to come by at the time, I started all this.
will get a proper one...

here is a serial log of what the faulty one had to say:
(I programmed my wifi, mqtt and other stuff into the firmware so I don't have to set up everything everytime)
```
00:00:00 Project sonoff Sonoff (Topic sonoff, Fallback DVES_A15FBB, GroupTopic sonoffs) Version 5.12.0
00:00:00 WIF: Connecting to AP1 wald in mode 11N as sonoff-8123...
00:00:04 WIF: Connected
00:00:04 HTP: Web server active on sonoff-8123 with IP address 192.168.1.110
18:41:42 MQT: Attempting connection...
18:41:42 MQT: Connected
18:41:42 MQT: test/sonoff/tele/LWT = Online (retained)
18:41:42 MQT: test/sonoff/cmnd/POWER = 
18:41:43 MQT: test/sonoff/tele/INFO1 = {"Module":"Sonoff Basic","Version":"5.12.0","FallbackTopic":"DVES_A15FBB","GroupTopic":"sonoffs"}
18:41:43 MQT: test/sonoff/tele/INFO2 = {"WebServerMode":"Admin","Hostname":"sonoff-8123","IPAddress":"192.168.1.110"}
18:41:43 MQT: test/sonoff/tele/INFO3 = {"RestartReason":"External System"}
18:41:43 MQT: homeassistant/switch/sonoff/config =  (retained)
18:41:44 MQT: test/sonoff/stat/RESULT = {"Reset":"Reset and Restarting"}
18:41:44 MQT: test/sonoff/stat/RESULT = {"POWER":"ON"}
18:41:45 MQT: test/sonoff/stat/POWER = ON
18:41:45 CFG: Use defaults
18:41:47 APP: Restarting

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 

tail 8

chksum 0x2d

csum 0x2d

v60000608

~ld
⸮

00:00:00 Project sonoff Sonoff (Topic sonoff, Fallback DVES_A15FBB, GroupTopic sonoffs) Version 5.12.0
00:00:00 WIF: Connecting to AP1 wald in mode 11N as sonoff-8123...
00:00:03 WIF: Connected
00:00:03 HTP: Web server active on sonoff-8123 with IP address 192.168.1.110
00:00:06 MQT: Attempting connection...
00:00:06 MQT: Connected
00:00:06 MQT: test/sonoff/tele/LWT = Online (retained)
00:00:06 MQT: test/sonoff/cmnd/POWER = 
00:00:07 MQT: test/sonoff/tele/INFO1 = {"Module":"Sonoff Basic","Version":"5.12.0","FallbackTopic":"DVES_A15FBB","GroupTopic":"sonoffs"}
00:00:07 MQT: test/sonoff/tele/INFO2 = {"WebServerMode":"Admin","Hostname":"sonoff-8123","IPAddress":"192.168.1.110"}
00:00:07 MQT: test/sonoff/tele/INFO3 = {"RestartReason":"Software/System restart"}
00:00:07 MQT: homeassistant/switch/sonoff/config =  (retained)
00:00:08 MQT: test/sonoff/stat/RESULT = {"POWER":"ON"}
00:00:08 MQT: test/sonoff/stat/POWER = ON
18:42:04 MQT: test/sonoff/tele/STATE = {"Time":"2018-02-18T18:42:03","Uptime":"0T00:00:14","Vcc":4.094,"POWER":"ON","Wifi":{"AP":1,"SSId":"wald","RSSI":84,"APMac":"92:2A:A8:51:8A:FB"}} (retained)
18:44:06 MQT: test/sonoff/stat/RESULT = {"Module":"29 (Sonoff T1 2CH)"}
18:44:07 APP: Restarting

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 

tail 8
[... 75 lines cut ...]

csum 0x2d

v60000608

~ld
⸮

00:00:00 Project sonoff Sonoff (Topic sonoff, Fallback DVES_A15FBB, GroupTopic sonoffs) Version 5.12.0
00:00:00 WIF: Connecting to AP1 wald in mode 11N as sonoff-8123...
00:00:04 WIF: Connected
00:00:04 HTP: Web server active on sonoff-8123 with IP address 192.168.1.110
18:47:41 MQT: Attempting connection...
18:47:41 MQT: Connected
18:47:41 MQT: test/sonoff/tele/LWT = Online (retained)
18:47:41 MQT: test/sonoff/cmnd/POWER = 
18:47:42 MQT: test/sonoff/tele/INFO1 = {"Module":"Sonoff Basic","Version":"5.12.0","FallbackTopic":"DVES_A15FBB","GroupTopic":"sonoffs"}
18:47:42 MQT: test/sonoff/tele/INFO2 = {"WebServerMode":"Admin","Hostname":"sonoff-8123","IPAddress":"192.168.1.110"}
18:47:42 MQT: test/sonoff/tele/INFO3 = {"RestartReason":"Software/System restart"}
18:47:42 MQT: homeassistant/switch/sonoff/config =  (retained)
18:47:43 MQT: test/sonoff/stat/RESULT = {"POWER":"OFF"}
18:47:43 MQT: test/sonoff/stat/POWER = OFF
18:47:51 MQT: test/sonoff/tele/STATE = {"Time":"2018-02-18T18:47:50","Uptime":"0T00:00:15","Vcc":4.094,"POWER":"OFF","Wifi":{"AP":1,"SSId":"wald","RSSI":82,"APMac":"92:2A:A8:51:8A:FB"}} (retained)

```

emontnemery, 2018-02-19:
The log is clear: The reset+restart is due to long press of button.
Please try connecting to the original relay board.

jeylites, 2018-02-21:
Do a flash clean using Esptool  and re flash. The other tip will be to set a unique default configuration on each T1. For mine I had it set with a unique static IP, Default module types, MQTT etc... if the hardware fails, I could rest assured that it will default to a seething that works.

[Troubleshooting](https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)

emontnemery, 2018-02-24:
@qorron Did you try with the original relay board?

qorron, 2018-02-24:
@emontnemery yes, I just tried it. it works fine while on AC.
any ideas why the other two worked fine on usb/serial?

alirezatabatabaee, 2019-03-12:
I have the same issue , with touch button the board resets every 5 second !!!

gspatzenegger, 2019-03-19:
I have the same issue with the sonoff T1. Using platformio, I changed the board type to esp8285 in the platformio.ini file, section [common] board = esp8285. Now the reboots have gone.