arendst/Tasmota #1949: Two SHT3X on I2C bus - feasible?

johngouk, 2018-02-17:
I want to connect 2 SHT3X temp/humidity sensors to a single ESP8266 on a single I2C bus, using separate I2C addresses (44 and 45). The devices with addresses set appropriately are detected and work fine individually. The SHT3X code currently only supports one device, but in principle I could modify it to support 2 on a single bus. Anyone aware of any a) existing efforts to do this b) potential gotchas, design/structural limitations beyond the SHT3X module itself? I was thinking of just looping against all the detected addresses and calling the appropriate MQTT/Web/Domoticz etc. routines, as it does now, adding a 0/1 indicator somewhere to each message for address 44/45.

Thanks for reading :-)

Frogmore42, 2018-02-18:
As you have seen the implementation does not support this.  It does support using multiplei2c  sensors, but each must be of a different type.  I have a few devices with BMP280 and HTU21 sensors to work around this issue.  The standard firmware only supports a single DS18B20, but there is a special version that supports multiple.

So, bottom line is that it is possible to support multiple sensors, look at the DS18x20 code for an example of how it can be done.

johngouk, 2018-02-19:
OK, I modified the module to support two sensors before the last comment arrived, such that it works as previously with just one, but produces modified JSON and Web content for two. Because the processing model calls it in the expectation of it being a single sensor, for JSON and Web that's ok as it can just put the data from 2 devices into the same output. However, I'm not sure what would happen for Domoticz, which gets two temp/humidity calls - I don't use it so haven't tested that.

[image]

Here's a JSON sample:

 tele/sonoff/0371/SENSOR = {"Time":"2018-02-19T13:08:50","SHT3X":[{"Address":"0x44","Temperature":19.8,"Humidity":49.3},{"Address":"0x45","Temperature":19.3,"Humidity":51.1}],"TempUnit":"C"}

If this would be useful, I'm happy to set up a branch/pull request etc. for moderation etc.

arendst, 2018-02-19:
Currently domoticz support only allows one temp sensor. 

Considering the addition of the SHTC3 sensor I think the code needs to support up to three sensors at the same time. Will need a re-write with array filled with found sensor info like DHT driver.

I, or someone else needs time to do it.

johngouk, 2018-02-19:
That DHT code looks fairly similar to what I did, except it appears to set up multiple single-sensor JSON objects in mqtt_data, rather than my array-of-sensors JSON. I think it only calls Domoticz T/H once, for the first one. Using single JSON objects makes the code a lot simpler, I had checked for a valid read from each sensor and added an STH3X element for a sensor only when they were good. It wouldn't be too hard to make this work for all three, on the same basis. I'll have a go, and get back to you. 

Would we need to use the pre-tele-period call to pre-read the sensors, prior to the xxxShow call, as in the DHT code? They respond pretty quickly, unlike the DHTs, so there's no attempt to do so at the moment, but if there are 3 of them it could get a bit long. How long could a xxxShow call take before anything broke?

arendst, 2018-02-19:
Yes pls give it a go.

Breaking is trial and error but I prefer to get the whole JSON string within 800mSec and that is with all sensors present. I normally test with Chrome Developer tools on the Ajax main page.

johngouk, 2018-02-20:
OK, done that. Seems to work just fine with two SHT3X sensors, I don't have a SHTC3 to test with. Using Ajax dev tools, the Waiting (TTFB) minimum time is about 80msec, so it can't be taking any longer than that for the Web format xxxShow call, which involves taking both measurements.

Here's a screen shot of the Web Main Menu:
[image]

and the JSON looks like:

7:34:02 MQT: tele/sonoff/0371/SENSOR = {"Time":"2018-02-20T17:34:02","SHT3X-0x44":{"Temperature":20.9,"Humidity":46.9},"SHT3X-0x45":{"Temperature":20.4,"Humidity":48.3},"TempUnit":"C"}

Happy to modify content/format if this isn't in keeping with other modules. I note that the DHT stuff uses Sensor names like "DHT20-nn" where "nn" is pin number in base 10. I used Hex because that's how the sensors generally express their address, so anyone using them would be more likely to know that.

What do I do? Create a pull request on the development branch, opening a new branch for this?

one-love-420, 2018-05-04:
With the way the addresses get tagged on the end(ie. "-0X44"), home assistant has json syntax errors with the way it reads the "-" is there another way to have the addresses show without it having the "-" 
Or is this more of a Home Assistant bug?

johngouk, 2018-05-05:
Since the values are just  JSON-encoded strings, I’d have thought it was a HA problem. I just used a separator value that I liked and seemed appropriate. IIRC, it started out as “_”, but that didn’t work for me!

one-love-420, 2018-05-05:
Ok that's good to know that "_" didn't work for you because I just compiled it with using the "_". It compiled fine i just haven't flashed yet to test and see if was going to fix the problem. It does seem to be more of a HA bug than anything else, because HA does receive the mqtt statements just doenst like the way the addresses are tagged along.
Error:
starting version 3.2.4
Testing configuration at /config
Failed config
  sensor.mqtt: 
    - Invalid config for [sensor.mqtt]: invalid template (TemplateSyntaxError: expected token 'end of print statement', got 'x44') for dictionary value @ data['value_template']. Got '{{value_json.SHT3X-0x44.Temperature}}'. (See ?, line ?). Please check the docs at https://home-assistant.io/components/sensor.mqtt/
    - platform: mqtt
      name: Temperature
      state_topic: tele/nodemcu/SENSOR
      unit_of_measurement: °F
      value_template: {{value_json.SHT3X-0x44.Temperature}}
    - Invalid config for [sensor.mqtt]: invalid template (TemplateSyntaxError: expected token 'end of print statement', got 'x44') for dictionary value @ data['value_template']. Got '{{value_json.SHT3X-0x44.Humidity}}'. (See ?, line ?). Please check the docs at https://home-assistant.io/components/sensor.mqtt/
    - platform: mqtt
      name: Humidity
      state_topic: tele/nodemcu/SENSOR
      unit_of_measurement: %
      value_template: {{value_json.SHT3X-0x44.Humidity}}
Successful config (partial)
  sensor.mqtt:

johngouk, 2018-05-05:
You might be just as well [modifying the MQTT template](https://www.home-assistant.io/docs/configuration/templating/#processing-incoming-data), which means you wouldn't have to modify the code every time it gets released...

one-love-420, 2018-05-10:
I messed with modifying templates for hours and gave up on going that route. The only way I seem to make it work in HA is using " _ " as the divider between the addresses instead of the " - "
If there is reason it has to be the " - " I vote for it to be changed but if not, it's not hard for me to change it myself when I want to update

Frogmore42, 2018-05-10:
I don't have/use HA, but was curious to see how to do this.  Using JSFiddle to test extracting the value, I came up with this for the template:
 {{value_json["SHT3X-0x44"].Temperature}}

What did you try to make it work (that didn't)?

one-love-420, 2018-05-10:
I can extract and view the json value, but when using the .yaml HA uses and when adding the sensor in the configuration.yaml it gets upset and has syntax errors at the "-" when changing it to the underscore it no longer throws the syntax error in HA and all code works as usual, like multiple sensors with addresses. They just show up as SHT3X_0x44. Hope this is making sense.

one-love-420, 2018-05-10:
I have it posted up a couple previous post but here it is again: 

- platform: mqtt
name: Temperature
state_topic: tele/nodemcu/SENSOR
unit_of_measurement: °F
value_template: {{value_json.SHT3X-0x44.Temperature}}

What works when I have it change into the .ino also is 

- platform: mqtt
name: Temperature
state_topic: tele/nodemcu/SENSOR
unit_of_measurement: °F
value_template: {{value_json.SHT3X_0x44.Temperature}}

The value_template doesn't like having the - as the divider between name and address so it needs a different symbol as the divider. I used the template too built into HA to see which symbols it doesn't throw errors and the underscore works. So then I tried to compile using arduino ide with the change to using the underscore.  Compiled with no errors and flashed, with several days of testing everything works fine with it being the underscore.

Frogmore42, 2018-05-10:
change it to this:
platform: mqtt
name: Temperature
state_topic: tele/nodemcu/SENSOR
unit_of_measurement: °F
value_template: {{value_json["SHT3X-0x44"].Temperature}}

this is the line I changed (from):
value_template: {{value_json.SHT3X-0x44.Temperature}}

To:
value_template: {{value_json["SHT3X-0x44"].Temperature}}

closed 2018-07-09 by stale[bot]

chedxb, 2019-06-08:
I attached 2 pcs of SHT3X sensors with different addresses ({"I2CScan":"Device(s) found at 0x44 0x45"})

However on the web server and MQTT, I see only one of them (SHT3X-0x45 Temperature and SHT3X-0x45 Humidity)

I use Tasmota, version 6.5.0

Do I need to change the settings and/or the code in order to be able to see both sensors?

Best,

Sava

johngouk, 2019-06-09:
> Do I need to change the settings and/or the code in order to be able to see both sensors?

You shouldn't have to. I wanted it to detect 1-3 sensors on different I2C addresses, and AFAIK it does. I am still running 5.12 because it works and I didn't get around to doing anything else or different yet. There appear to have been various issues with multiple sensors of different types conflicting/not being picked up etc. in conjunction with SHT3X, but I haven't really studied it.

chedxb, 2019-06-10:
It is interesting that I2CScan sees both devices but 2nd one is not visible on the web server and the MQTT.

Any idea how I can debug the problem?

johngouk, 2019-06-10:
Firstly, try each sensor separately - x44 and x45. Do you have other sensors of any sort connected? Try with/without. At least that gives you a clue... You should also be able to set the Log level to DEBUG, which should tell you if it found them in the initialisation phase. I can't recall how to do that, but I'm sure you can RTM. I agree it is curious that the I2CScan sees them, it might be worth checking out the code for that to see what it's doing that is different!