arendst/Tasmota #1950: 4CH Pro - button 4 not working

Leseratte10, 2018-02-17:
I have flashed a 4CH Pro with tasmota, and unfortunately, button 4 doesn't work. 

I can trigger relay 4 using the web interface, and I can trigger all other relays using buttons 1-3, but when I press button 4, nothing happens. Is it possible that I have a newer version of the 4CH Pro that has the fourth button on another GPIO pin?

Is there something like a setting that makes Tasmota print any button press on any GPIO to the console so I can check this? I'd rather not disassemble the 4CH again and check the button traces as I'd have to unplug the device from all this cable mess and it was not easy to correctly connect all the wires with that little space - but if there is no other way, I could do so.

Leseratte10, 2018-02-17:
After re-flashing the current version of Tasmota, button 4 now works and switches the relay. I have no idea why it didn't work before.

closed 2018-02-17 by Leseratte10

tobox, 2019-02-27:
I had a very similar problem today. I flashed a 4CH Pro (non R2) using esptool and everything worked, except for hardware buttons 2 & 3. I traced all wires, studied the schematic and finally found your post. The problem persisted for at least 5 reboots. 

I updated tasmota over wifi and the problem was gone. Strange.