arendst/Tasmota #1955: Sonoff disconnects from Wifi after connecting to mqtt

Ismily, 2018-02-17:
Hey everyone,

This is my first time using Tasmota and Home Assistant. I just started learning about all of these things so I'm very very noob. 
I have a problem when connecting my Sonoff Basic which has tasmota to mqtt on home assistant. Whenever I put the IP for my raspberry pi which has hassio and mqtt in the sonoff, the sonoff looses connection with the WiFi and the led starts to blink. It sometimes reconnects after a while then disconnects immediately. 
I followed this [guide](https://www.youtube.com/watch?v=c2IUIUnxjUY&t=858s) step by step for setup. 

Some extra info: I don't know how to test if mqtt is working correctly. I have access to the pi using SSH. Don't know if this is all you guys need to be able to help. Let me know if there is anything else you'd like to know. Please Help! and thank you in advance!

jeylites, 2018-02-18:
Is the IP address the same as during MQTT HASIO install? If not revert back to the previous IP address during install and see what happens.

Ismily, 2018-02-18:
Thanks for the quick response. I would love to try that. But how exactly do I do that ?! Sorry I'm very new to this

alton76, 2018-02-18:
Nevermind, can you send the log of the sonoff?
And some details on version etc

Ismily, 2018-02-18:
So this is what the last this the console shows soon as i change the IP of mqtt to the pi running hassIO, then it disconnects from WiFi and the led starts blinking:

08:00:07 MQT: tele/sonoff/INFO1 = {"Module":"Sonoff Basic","Version":"5.12.0","FallbackTopic":"DVES_8B4183","GroupTopic":"sonoffs"}
08:00:07 MQT: tele/sonoff/INFO2 = {"WebServerMode":"Admin","Hostname":"sonoff-0387","IPAddress":"192.168.0.53"}
08:00:07 MQT: tele/sonoff/INFO3 = {"RestartReason":"Software/System restart"}
08:00:07 MQT: homeassistant/switch/sonoff/config = {"name":"Sonoff","command_topic":"cmnd/sonoff/POWER","state_topic":"stat/sonoff/RESULT","value_template":"{{value_json.POWER}}","payload_off":"OFF","payload_on":"ON","availability_topic":"tele/sonoff/LWT","payload_available":"Online","payload_not_available":"Offline"} (retained)
08:00:07 MQT: stat/sonoff/RESULT = {"POWER":"ON"}
08:00:07 MQT: stat/sonoff/POWER = ON
08:00:08 MQT: stat/sonoff/RESULT = {"POWER":"ON"}
08:00:08 MQT: stat/sonoff/POWER = ON
08:00:15 MQT: tele/sonoff/STATE = {"Time":"1970-01-01T08:00:15","Uptime":"0T00:00:14","Vcc":3.471,"POWER":"ON","Wifi":{"AP":1,"SSId":"Fifi","RSSI":100,"APMac":"BC:64:4B:38:BF:D5"}}
08:00:16 MQT: tele/sonoff/SENSOR = {"Time":"1970-01-01T08:00:16","Switch1":"ON"}
08:00:34 RSL: stat/sonoff/RESULT = {"POWER":"OFF"}
08:00:34 RSL: stat/sonoff/POWER = OFF
08:00:36 RSL: stat/sonoff/RESULT = {"POWER":"ON"}
08:00:36 RSL: stat/sonoff/POWER = ON
08:00:36 RSL: stat/sonoff/RESULT = {"POWER":"OFF"}
08:00:36 RSL: stat/sonoff/POWER = OFF
08:00:45 WIF: Connecting to AP1 Fifi in mode 11N as sonoff-0387...

Running latest version of Tasmota. As I'm writing this it reconnect then disconnects again. This is what I see in console when it reconnects:

00:00:00 Project sonoff Sonoff (Topic sonoff, Fallback DVES_8B4183, GroupTopic sonoffs) Version 5.12.0
00:00:00 WIF: Connecting to AP2 indebuurt2 in mode 11N as sonoff-0387...
00:00:07 WIF: Connect failed as AP cannot be reached
00:00:07 WIF: Connecting to AP1 Fifi in mode 11N as sonoff-0387...
08:00:13 WIF: Connected
08:00:13 DNS: Initialized
08:00:13 HTP: Web server active on sonoff-0387.local with IP address 192.168.0.53
08:00:15 MQT: Attempting connection...
08:00:15 MQT: Connected
08:00:15 MQT: tele/sonoff/LWT = Online (retained)
08:00:15 MQT: cmnd/sonoff/POWER = 
08:00:15 MQT: tele/sonoff/INFO1 = {"Module":"Sonoff Basic","Version":"5.12.0","FallbackTopic":"DVES_8B4183","GroupTopic":"sonoffs"}
08:00:15 MQT: tele/sonoff/INFO2 = {"WebServerMode":"Admin","Hostname":"sonoff-0387","IPAddress":"192.168.0.53"}
08:00:15 MQT: tele/sonoff/INFO3 = {"RestartReason":"Software/System restart"}
08:00:15 MQT: homeassistant/switch/sonoff/config = {"name":"Sonoff","command_topic":"cmnd/sonoff/POWER","state_topic":"stat/sonoff/RESULT","value_template":"{{value_json.POWER}}","payload_off":"OFF","payload_on":"ON","availability_topic":"tele/sonoff/LWT","payload_available":"Online","payload_not_available":"Offline"} (retained)
08:00:15 MQT: stat/sonoff/RESULT = {"POWER":"OFF"}
08:00:15 MQT: stat/sonoff/POWER = OFF
08:00:16 MQT: stat/sonoff/RESULT = {"POWER":"OFF"}
08:00:16 MQT: stat/sonoff/POWER = OFF
02:52:12 MQT: tele/sonoff/STATE = {"Time":"2018-02-18T02:52:12","Uptime":"0T00:00:22","Vcc":3.471,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Fifi","RSSI":100,"APMac":"BC:64:4B:38:BF:D5"}}
02:52:12 MQT: tele/sonoff/SENSOR = {"Time":"2018-02-18T02:52:12","Switch1":"ON"}
02:52:23 WIF: Connecting to AP1 Fifi in mode 11N as sonoff-0387...

jeylites, 2018-02-18:
Is your RPI, router, sonoff in the same range IP? 
Are you able to toggle the sonoff relay from to web browser?

Sonoff: 192.168.0.53 is what you have for the sonoff

Ismily, 2018-02-18:
my RPI and sonoff are in the same range and I am able to toggle the sonoff from the web browser yes

jeylites, 2018-02-18:
I had similar issues with MQTT in the past, say I have my RPI IP at 192.168.1.10 and install MQTT, it somehow takes that as default IP. Ather the install if I try to change the IP to something else for example 192.168.1.30, it goes into a similar problem as yours. However reverting back to 192.168.1.10 fixed the issue. 

Where did you get your copy of MQTT from?

Ismily, 2018-02-18:
From the hassio add-ons page. Downloaded the mosquitto broker

jeylites, 2018-02-18:
can't say much about hassio add-ons, I don't use it. Try rebuilding RPI (Clean Install) with a different MQTT install, I'm using Peter Scargill's  script as linked [Script](https://tech.scargill.net/the-script-step-by-step/)

You could just install prerequisite, mosquito, modpass and leave everything else to uncheck...

 Before rebuilding check download an MQTT sniffer on Android phone or computer to see if the messages are coming in and out.

RVC69, 2018-02-18:
I have the same problem wifi connecting/disconnecting/reboot with sonoff rf, tasmota 5.12.0 and home assistant
downgrade to tasmota 5.11.1 is working.

Ismily, 2018-02-18:
Easiest way to downgrade?

Ismily, 2018-02-18:
Upgraded the firmware and it solved it! New update came out like 18hrs ago guess it had the fix. Thank you everyone for your help

alton76, 2018-02-18:
Just wondering what upgrade did you do?

jeylites, 2018-02-18:
@Ismily what was the issue?

Ismily, 2018-02-18:
@alton76  sonoff.ino 5.12.0b - downloaded it and reflashed the sonoff. 

@jeylites the issue was that every time I put my mqtt IP the sonoff would disconnect from WiFi. Seems to work fine now after upgrade.

benjaminprojas, 2018-04-22:
I can confirm. I had the exact same issue when connected to MQTT.  I just upgraded to v5.12.0m and everything now just works perfectly.

jefonseca, 2018-05-06:
It seems to happen in 5.13.1 with binary file in sonoff basic.

Flashed in GNU/Linux using this commands:
```
esptool.py -p /dev/ttyUSB0 erase_flash
esptool.py -p /dev/ttyUSB0 write_flash -fs 1MB -fm dout 0x0 sonoff.bin
```

I tried this versions in order to reproduce the problem (multiple times):
5.11.1 - Working Ok
5.12.1 - Working Ok
5.13.1 - Works, but when connection to MQTT Broker establishes, the network connection begin to fail

The message on the web console
```
19:55:32 DNS: Initialized
19:55:33 MQT: Attempting connection...
19:55:33 MQT: Connected
19:55:33 MQT: tele/DVES_XXXXXX/LWT = Online (retained)
19:55:33 MQT: cmnd/DVES_XXXXXX/POWER = 
19:55:49 DNS: Initialized
19:55:50 MQT: Attempting connection...
19:55:50 MQT: Connected
19:55:50 MQT: tele/DVES_XXXXXX/LWT = Online (retained)
19:55:50 MQT: cmnd/DVES_XXXXXX/POWER = 
19:56:12 DNS: Initialized
19:56:13 MQT: Attempting connection...
19:56:13 MQT: Connected
19:56:13 MQT: tele/DVES_XXXXXX/LWT = Online (retained)
19:56:13 MQT: cmnd/DVES_XXXXXX/POWER = 
19:56:29 DNS: Initialized
19:56:30 MQT: Attempting connection...
19:56:31 MQT: Connected
```

If i shutdown the MQTT Broker (or cut connection between the sonoff and the broker); the network connection on the sonoff works perfectly.

belidzs, 2018-05-09:
I1m having the same issue, however it only affects my Sonoff Basic unit. My Sonoff S20 and Sonoff Pow units are working perfectly with 5.13.1

ethan4, 2018-07-15:
Same issue here, five devices connected, all on 6.1.1

I can connect to the device's web interface shortly after they are booted but after a few minutes the web interface goes unresponsive. Kicking the device off the wifi (from my UniFi controller) seems to wake it up for a while but the same thing happens again after a while. 

MQTT seems to stay connected normally as devices are responsive to control via my broker after multiple days of uptime.

ascillato, 2018-08-23:
Hi,

Closing this issue as Ismily has stated the issue is solved by flashing last version. Please, ask to reopen if needed. Thanks.

closed 2018-08-23 by ascillato2