arendst/Tasmota #1956: ENERGY_Today

Lu-Fi, 2018-02-18:
Could there be a Problem with ENERGY_Today ? 
Counter does not reset here per day.

ENERGY_Yesterday is working as expected.

Frogmore42, 2018-02-18:
You'll need to provide a little more detail.  A snippet of the console, an indication of exactly what is not working for you, is this something new, etc.

Having said that, POW devices depend on having the correct time to do daily calculations, so you could be seeing one of two known issues.  1st is if the time is not correct, the daily value will not reset.  You need to get NTP working again to fix that issue.  There are a few threads on that issue.  I have found that removing all power from the POW for about 30min and then plugging it back in, will fix that, but that hasn't worked for some people.  The other known issue is that some older versions don't reset until there is new usage.

Lu-Fi, 2018-02-20:
Here you can see the Power Usage per Day for my freezer.

[image]

This is the yctual State after 5.12 update:

[image]

Regards

LuFi

arendst, 2018-02-20:
Pls provide output of command ``status 0``.

Does it happen every day or sometimes? Is it only Energy Today or also Energy Yesterday?

Lu-Fi, 2018-02-20:
This is one of 7 different POW i have. 
All of them having the same issue.

17:25:59 CMD: status 0
17:25:59 MQT: stat/SonoffPow_2/STATUS = {"Status":{"Module":6,"FriendlyName":"Sonoff","Topic":"SonoffPow_2","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":0,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
17:25:59 MQT: stat/SonoffPow_2/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://domus1:80/api/arduino/sonoff.ino.bin","Uptime":"0T08:52:54","Sleep":50,"BootCount":60,"SaveCount":267,"SaveAddress":"F6000"}}
17:25:59 MQT: stat/SonoffPow_2/STATUS2 = {"StatusFWR":{"Version":"5.12.0","BuildDateTime":"2018-02-09T14:54:51","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
17:25:59 MQT: stat/SonoffPow_2/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"192.168.178.40","LogPort":514,"SSId1":"xxx","SSId2":"xxx","TelePeriod":300,"SetOption":"00000009"}}
17:25:59 MQT: stat/SonoffPow_2/STATUS4 = {"StatusMEM":{"ProgramSize":485,"Free":516,"Heap":23,"ProgramFlashSize":1024,"FlashSize":4096,"FlashMode":3}}
17:25:59 MQT: stat/SonoffPow_2/STATUS5 = {"StatusNET":{"Hostname":"POW-KUE-TIEFK","IPAddress":"192.168.190.53","Gateway":"192.168.190.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.190.1","Mac":"5C:CF:7F:B4:A5:67","Webserver":2,"WifiConfig":2}}
17:25:59 MQT: stat/SonoffPow_2/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.178.40","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_B4A567","MqttUser":"DVES_USER","MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
17:25:59 MQT: stat/SonoffPow_2/STATUS7 = {"StatusTIM":{"UTC":"Tue Feb 20 16:25:59 2018","Local":"Tue Feb 20 17:25:59 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":99}}
17:25:59 MQT: stat/SonoffPow_2/STATUS9 = {"StatusPTH":{"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
17:25:59 MQT: stat/SonoffPow_2/STATUS10 = {"StatusSNS":{"Time":"2018-02-20T17:25:59","ENERGY":{"Total":251.046,"Yesterday":1.125,"Today":5.229,"Power":4,"Factor":0.22,"Voltage":228,"Current":0.078}}}
17:25:59 MQT: stat/SonoffPow_2/STATUS11 = {"StatusSTS":{"Time":"2018-02-20T17:25:59","Uptime":"0T08:52:54","Vcc":3.204,"POWER":"ON","Wifi":{"AP":1,"SSId":"IoT","RSSI":94,"APMac":"56:C7:BF:16:84:D8"}}}

Lu-Fi, 2018-02-20:
And yes, ENERGY_Yesterday is working.

arendst, 2018-02-20:
Ah, it's probably caused by having sleep set to 50. Change it to 0 and see what happens over the next days.

Lu-Fi, 2018-02-20:
ok, i will report tomorrow

arendst, 2018-02-21:
You might want to read this #1842 regarding energy saving.

Lu-Fi, 2018-02-21:
thx, very informational post: #1842  !!

closed 2018-02-21 by Lu-Fi