arendst/Tasmota #1958: Sonoff Touch switches go on AP mode if wifi gateway to server goes down

ornolio, 2018-02-18:
I've 18 Sonoff Touch switches all flashed with Tasmota Program Version 5.11.1 Build Date & Time 2018-01-07T21:51:09, except one which is at last version. 

Topology
All S.Touch are linked to a MQTT  (mosquito) dedicated server (Raspberry PI 3) through a single Internet gateway Router-wifi AP (Access Point). 
The server is on the same LAN linked to the gateway Internet Router using a UTP cable.

Status
When the  Internet Router-AP is powered off, the status of all the S.Touch:
S1) are still switchable, but the touch must be last more long (5 secs instead of less than 1 sec)
S2) are blinking (including those settled with Led off) with a 1s period as, if they are in a waiting for OTA flashing mode 
S3) are on the wifi AP list of my PC (and any mobile phone in the range) and each one switch can be WIFI linked
S4)When Internet Router wifi AP is on and it appears in my PC AP list, in  several  minutes (max 5 min) all the S.Touch (one by one, progressivly) disappears from my PC AP list, while returning automatically in a normal status connected to the  Internet Router-AP and, in turn to the MQTT server.

Questions: 
Q1)Which could be the reason for S1) and how it can be recovered (normal switching touching 1sec)?
Q2)is there a way to configure each S.Touch to let them NOT to run in AP mode ?

Thank You

SodoffBaldrick, 2018-02-18:
https://github.com/arendst/Sonoff-Tasmota/wiki/Commands
Is it WifiConfig that interests you?

I set mine to "WifiConfig 4"

ornolio, 2018-02-18:
Thank you SodoffBaldrick for your help, you addressed me on the right path. In my case now I'm using WifiConfig 5

closed 2018-04-20 by ornolio