arendst/Tasmota #1960: lightbulb state of mqtt msg is not complete

kevinshane, 2018-02-18:
Hi, when I power on/off the lightbulb, in this case is a ailight, I always get the mqtt payload msg like this:
```
Topic: stat/ailight/RESULT
Data: {"POWER":"ON"}
```
not a complete state of the Dimmer and the Color(because when turning on or off, the Dimmer and Color also get changed). The `stat/xxx/RESULT` topic should always returns a complete data like this:
when turning on: `{"POWER":"ON","Dimmer":100,"Color":"FFFFFFFF"}`
when turning off: `{"POWER":"OFF","Dimmer":0,"Color":"00000000"}`

Please fix the msg payload, because it will cause error when only turn on/off the lights, the msg only returns the power state but missing the Color and Dimmer state.

arendst, 2018-02-18:
Try ``Backlog Power On; State``

kevinshane, 2018-02-21:
hi @arendst Thanks for sharing the commands, when I execute `Backlog Power On; State` on web console, I get all info including Dimmer and Color, which is very nice. But how to publish this command via mqtt? In my openhab item settings it look like this
```
Switch ailight "ailightPower" <colorlight> (gMbedroom)
    { mqtt=">[broker:cmnd/ailight/POWER:command:*:default],
            <[broker:stat/ailight/RESULT:state:JSONPATH($.POWER)]" }

Dimmer ailightDimmer "ailightDimmer" <colorlight> (gMbedroom)
    { mqtt=">[broker:cmnd/ailight/Dimmer:command:*:default],
            <[broker:stat/ailight/RESULT:state:JSONPATH($.Dimmer)]" }
```

I am sure I need to change the mqtt `<` info for the Dimmer item, but how?
What's the correct way to subs a topic listening to the command `Backlog Power On; State`

kevinshane, 2018-03-02:
Hi @arendst  Theo, please take a look at the correct status of turning on/off the lights, #2014 also has the same problem.
when using `Backlog State` in mqtt , it does gives the full state, but the `stat/ailight/RESULT` also received `{"Backlog":"Appended"}` which still cause many errors if items listen to this topic
Also when turning on/off the lights, the `Dimmer` and `Color` stat not get update.
For example, when POWER off, the Dimmer still at `100` state. and Color still at previous state(not 00000000 when off)