arendst/Tasmota #1961: MagicHome Wifi after flashing sonoff.bin

timota, 2018-02-18:
Hi All, 
sorry for noob question. I recently flashed sonoff.bin firmware on MagicHome(zj-esp-ir-b v2.3) via esptool. Below output:
--------------------------------------------------------------------------
D:\ESP\esptool-2.2.1>python esptool.py --port COM6 write_flash -fs 1MB -fm dout
0x0 sonoff.bin
esptool.py v2.2.1
Connecting....
Detecting chip type... ESP8266
Chip is ESP8285
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Compressed 496960 bytes to 342591...
Wrote 496960 bytes (342591 compressed) at 0x00000000 in 30.4 seconds (effective
130.7 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting... 
--------------------------------------------------------------------------
But after module restart - i dont see any WiFi hotspot to flash main firmware. What im doing wrong or do i need to do an extra step that i missed ? 

Please help and sorry for noob question.

oxcxsxixbxaxn, 2018-02-18:
I had the same issue...  I did see a hotspot sometimes, but it wasn't visible long and didn't accept connection.
I ended up following the guide to build a custom firmware including my own home wifi details in user_config.h
Now I can flash any devive and have it connect straight to my wifi.  Then just need to look at router to see which IP it got...
https://github.com/arendst/Sonoff-Tasmota/wiki/Beginner-Guide---Create-your-own-Firmware-Build

timota, 2018-02-18:
got it. Will build my firmware. Thanks for help.

timota, 2018-03-02:
resolved

closed 2018-03-02 by timota