arendst/Tasmota #1964: Ultrasonic distance measurement

nosaj66au, 2018-02-18:
Hi Guys,
    Any plans for ultrasonic distance measurement sensors? I have about 4 application I could use it for now if it ran on Tasmota. I'm currently playing with this one at the moment "Jsn-sr04t Ultrasonic Module Distance Measuring Transducer Sensor Waterproof" code is very small and simple.
Jason

MimbaMonkeyHouse, 2018-02-20:
I'd also be interested in this for various water tanks!

Oxyandy, 2018-02-20:
Water tanks ? I am using float switches.. the so far have been working great

MimbaMonkeyHouse, 2018-02-20:
Thanks for the tip, but I've not seen floating switches adequate to check water level in 3m deep - 8000l tanks. I was looking at these for swimming pool water level where amplitude is not as wide as in rain collection tanks...

nosaj66au, 2018-02-21:
Hi all, Not that i wanted too, but I have now have loaded ESPEasy onto a Wemos Mini and the used the waterproof Ultrasonic module no need for volt shiffers the module runs fine on 3.3v. Bingo!!! Even have it running on Solar and 3400mA battery.  
Jason

MimbaMonkeyHouse, 2018-02-21:
Cheers for that nosaj66au. I may try that then! What solar panels are you using?

nosaj66au, 2018-02-21:
Hi MimbaMonkeyHouse, 
https://www.bigdiscount.com.au/solar-panel-monocrystalline-cells-3w-5w-8w-9v.html
the 8w version most likly overkill a I have enable sleep mode which draws 4.9mA and jumps to 120mA to measure and send the MQTT message.

MimbaMonkeyHouse, 2018-02-21:
Nice