arendst/Tasmota #1969: Magic Home controller turn on randomly

LucaAghemo, 2018-02-19:
Hello,
I flashed a Magic Home led strip controller, following instructions. Everything went well, I setup the device via web, had to set it as 37 Arilux because it's one of the new series, but since that it's fine, everything work, rgb settings, dimmer etc.
The only problem is that the lestrip turn on randomly, usually after a few minute I turned it off.
It doesn't send any signal on mqtt, that is it doesn't update the status (remain off), it turns on with the last color setting but in full brightness, even if the state before the last time I turned it off was dimmed very low.
I thought it was a power problem, something like a bad wire that sometimes loose contact and turn off and on the controller, but this is not the case IMHO, because if I disconnect from the power and reconnect it actually turn on, but it update the mqtt status and in Home Assistant I see the switch changing. This doesn't happen in the "random" episodes.
So now I feel unsecure about using it, because I don't want it to turn on the strip while I'm away or, even worst, in my bedroom while I'm sleeping!
And speaking about that, is there a way to set the initial state to off? So that, if the power goes off and back on during the night I'm not awaken from the light?
Any suggestion? Thx in advance.

reloxx13, 2018-02-19:
Try to set PowerOnState to off, if this fixes your problem the device is restarting/crashing.

What says Last Restart Reason on Information Page?

reloxx13, 2018-02-19:
Maybe one other GPIO Pin is not correct with your board

```
{ "Arilux LC01",     // Arilux AL-LC01 (ESP8285) - https://www.banggood.com/nl/ARILUX-AL-LC01-Super-Mini-LED-WIFI-Smart-RGB-Controller-For-RGB-LED-Strip-Light-DC-9-12V-p-1058603.html
                       //  (PwmFrequency 1111Hz)
     GPIO_KEY1,        // GPIO00 Optional Button
     0,
     GPIO_LED2_INV,    // GPIO02 RF receiver control
     0,
     GPIO_ARIRFRCV,    // GPIO04 IR or RF receiver (optional)
     GPIO_PWM1,        // GPIO05 RGB LED Red
     0, 0, 0, 0, 0, 0, // Flash connection
     GPIO_PWM2,        // GPIO12 RGB LED Green
     GPIO_PWM3,        // GPIO13 RGB LED Blue
     GPIO_USER,        // GPIO14 RGBW LED White (optional - set to PWM4 for Cold White or Warm White)
     0, 0, 0
  },
```

you could try to remove the gpios you dont need in the code (replace with 0) and reflash.

LucaAghemo, 2018-02-19:
Sorry for the noob question: how do I set PowerOnState to off? Via console?
I'll check the GPIO pins. With the 34 MagicHome setting colours where wrong, no red and green and blue inverted. Setting 37 Arilux fixed. The problem is that I checked for some minute the controller after last flashing, everything was fine, so I unsolded the wires from the board. I was fool enough not to keep it connected for a while to REALLY check.
If I have to resold on that chip I might close myself in a small dark room and start crying... :-)

Jason2866, 2018-02-19:
No need to resolder. You can flash it over the website of the sonoff. just typen in the IP Adress in your browser select new firmware file and your gone...

reloxx13, 2018-02-19:
and be sure you set up correct wifi setting in the user_config.h incase of config loss.

read the wiki command page for power on state.

also, you should atleast read it once to know what u can do.

LucaAghemo, 2018-02-19:
First of all thx for your kind answer.
I gave the command in the console, that was accepted, but that didn't change the problem.
Actually, I discovered that I have a problem in the mqtt broker, that loose reconnect with my devices.
I didn't know that because the other device don't change the status when it happen, but monitoring the broker on the raspberry via mosquitto_sub showed the lost and restabilished connections.
So now I have two problem, keeping the led in the same state if it loose the connection (like my other sonoff and stuff with tasmota) and understanding why mqtt goes up and down.
Ain't life a bitch? :-)

Jason2866, 2018-02-20:
Yes you can flash the generated firmware from arduino over the web console

LucaAghemo, 2018-02-26:
A small update, I flashed a new Tasmota firmware OTA, everything went well but the led strip continue to turn on randomly, probably everytime it looses connection with the mqtt server.
The PowerOnState off command in the console seems ok, nevertheless at power on it still turn on.
Any advice please?

Jason2866, 2018-02-26:
Could you post the console messages from start?
Which version do you use, Tasmota and Arduino IDE?

aspitznyc, 2018-03-06:
@LucaAghemo Just a quick question ... are you using a physical switch to turn it on/off? Like one connected to GPIO04? On another Sonoff devices, you get a random turn on because the logic pin starts floating and it doesn't know if it's on the ON or OFF state due to interference with the cable for the physical switch.
If that's your case, it's solved by soldering a resistor and/or a capacitor.