arendst/Tasmota #1970: Support for Geekcreit 2 Channel AC 85V-250V APP Remote Control WIFI Wireless Switch Socket For Smart Home

bvdvoorn, 2018-02-19:
I have tried to load the Tasmota software on the Geekcreit 2 Channel AC 85V-250V APP Remote Control WIFI Wireless Switch Socket [https://www.banggood.com/Geekcreit-2-Channel-AC-85V-250V-APP-Remote-Control-WIFI-Wireless-Switch-Socket-For-Smart-Home-p-1114958.html?cur_warehouse=CN](url).

I used the SonOTA methode and loading did work without any problem. But after finishing the switch is not working and the wifi not connecting.

Any idea what I can do to correct/resolve this?

arendst, 2018-02-19:
Find out what GPIO is connected to what button and relay ;-). Then using the Generic module check your findings.

Jason2866, 2018-02-20:
What is the advantage of this modul? It is expensiver than a sonoff 2Ch

bvdvoorn, 2018-02-20:
The relay is to be connected to a lower power source. With the sonoff 2Ch the relay and unit are connected to same power source.

rqsnst, 2018-02-22:
that is an interesting module , did you find out how to configure ?

Jason2866, 2018-02-25:
@bvdvoorn Some moduls doesnt connect when using a universal build. For example many LED modules needs a preconfigured firmware.... Solder the needed wires and flash "your" firmware. Perhaps all the needed signals are at the 4 hole pin connector. After that you can play as @arendst said.

KLelong, 2018-06-06:
I did some reverse engineering, but I'm stuck.
I successfully could flash tasmota, but it didn't work. So I tried an arduino project (adapted version of telnet to serial). That worked fine.
The chip besides the 6pin connector is a ULN2003, which inverts 4 pins of the 6pin connector to the ESP8285.
FWIW, these are the pins (square pin = pin1) :
1 = GPIO0
2 = GPIO9
3 = GPIO10
4 = GPIO14
5 = 3V3
6 = GND
Unfortunately, the other chip has no marking. This is the pinout of that chip (J2 is the 4 pin connector) :
1 = ?
2 = ?
3 = Relay
4 = Relay
5 = GPIO10
6 = 3V3
7 = GPIO5
8 = GND
9 = J2-3
10 = J2-2
11 = GPIO9
12 = GPIO4
13 = GPIO0
14 = GPIO15
15 = GPIO12
16 = GPIO14
The LED is connected to GPIO13 (via a resistor).
The left switch is connected to GPIO0, so it could be used for flashing.
Middle switch is connected to GPIO9, right switch connected to 2nd chip pin 10 and pin2 of J2. Pin 1 and 4 of J2 are GND and 3V3 resp.
I suspect that second chip is some serial thing. I looked at some microchip chips, but there a few that have 16 pins ant then their power-pins are in other places. Also some AVR chips don't match.
I was looking for a schematic or photo's of the first version of the Sonoff Dual so I could compare that with this module but I don't find anything.

Any suggestions how to proceed ? I still got 4 modules with the original firmware.

closed 2018-08-05 by stale[bot]

james-fry, 2019-02-05:
Did you get this working?
Looks like 2ch version of this:
https://github.com/arendst/Sonoff-Tasmota/wiki/Sonoff-4CH-DIY
https://notenoughtech.com/home-automation/nodered-home-automation/hacking-esp8285-geekgerit-4-way-relay-controller/

KLelong, 2019-02-06:
I'm going to try this, watch for an update.
FWIW, I don't really need tasmota, as long as I have MQTT.

james-fry, 2019-02-06:
Me too :)
I have a device on my desk ready.
Its the mains AC85-250v version but I checked its working via +5v applied where the output of the buck converter is. Im not applying 240v until I have flashed and checked it etc.
It boots up etc.
Tonight (or soon) I will solder up gnd, 3v3, tx, rx and gpio0 dupont cables (thx for the pcb tracing btw) and try flashing tasmota per the 4ch DIY page I linked.

KLelong, 2019-02-10:
Hi,
I could flash the device and I could contact the initial web-interface (accesspoint-mode). But when I configure it and finish the configuration the device reverts to the "factory-defaults". Stranger still, I see the device contacted my accesspoint and got an address, but after a while, it disappears from the network to go back to accesspoint-mode. Sometimes even resulting in a Polish web-interface.
I don't understand :-(
A remark : I had to create a new SSID, because the password of the AFE firmware is max. 32 characters. Normally I use longer passwords.

james-fry, 2019-02-10:
When you put the bootloader in firmware upload mode (GPIO0-->GND when applying power) did you find that the device pulses one of the relays?
None of the flashing tools I have had success with previously (for sonoff basic and nodemcu) seem to want to flash this device.
Did you use FTDI/serial to flash? Or SonOTA?

KLelong, 2019-02-11:
I used esptool.py to flash the bin file, with a FTDI usb to serial adapter.
FWIW, I also flashed the latest tasmota-basic bin file. I could access the device and configure it to use my wireless network. Unfortunately, I can't get the relays to work. With MQTT I can see some buttons though.
I don't recall any action of the relays when flashing.

james-fry, 2019-02-11:
Haha total schoolboy error.
Connected tx-tx and rx-rx 
D'oh
Now I worked that out it flashed tasmota no probs.

So I had some success...
I can control (toggle) each of the relays from tasmota UI.

This is my config:

[image]

[image]

Note that you need to select the correct relay mode using left hand button (closest button to PSU).
This button seems to cycle through some relay modes - I guess supported by the external chips.

I actually want inching mode (pulse) to control a blind with up/down buttons.
So by using PulseTime1 10 and PulseTime2 10 commands I got it working perfectly for my use.
:)

james-fry, 2019-02-11:
Additional note. For me when the device restarts it starts in correct relay mode (or retains the mode that was correct before the restart)

KLelong, 2019-02-12:
Many thanks,
you got it working :+1: ! 
One drawback : that third button that toggles the relay-mode. Maybe I will make the plastic stud shorter.
Now I have to find a use for these devices :wink:. Since I couldn't make them work, I purchased some Sonoff Dual R2 and installed these on my shutters.

james-fry, 2019-02-12:
Agree - I think we can now state that this device is supported by tasmota.
With your great work tracing the pcb and the info I added on tasmota config I think there is enough for a wiki page...

BTW I have the opposite problem with regard to the buttons.
I got a version of this device that was PCB only and it has only short buttons.
I intend to 3d print an enclosure but I need to find some way to extend the buttons.
I am definitely not happy to press these buttons with my finger when 240v AC is used...

this is my device:
https://www.aliexpress.com/item/Sonoff-2CH-AC85-250V-Wireless-Switch-Timer-Smart-Wifi-Switch-Universal-Automation-Module-Remote-Control-for/32948157608.html

[image]

KLelong, 2019-02-12:
I see, [this](https://www.banggood.com/Geekcreit-2-Channel-AC-85V-250V-APP-Remote-Control-WIFI-Wireless-Switch-Socket-For-Smart-Home-p-1114958.html?rmmds=search&cur_warehouse=CN) is the device I own, that has an enclosure and "long" buttons.

james-fry, 2019-02-12:
Here is a nice diagram of the pinouts needed to flash and configure the device.
[image]

I am adding a wiki page, which will use this image.

james-fry, 2019-02-12:
Wiki page added:
https://github.com/arendst/Sonoff-Tasmota/wiki/Geekcreit-%22Sonoff%22-2-Channel-Relay-(AC-85V-250V)

KLelong, 2019-02-12:
Minor remark : maybe add GPIO0, 9 10 and 14 in the image (vertical connector).
You need to pull them to 3V3 to activate the switch. You will need to have a pull-down-resistor to able to use them because just a dangling wire of about 11cm on GPIO0 makes it pick up noise and toggle the relay unpredictable.

james-fry, 2019-02-12:
I can certainly add those.
But actually I didn't find any need to connect these pins/pads to flash or operate the relay.
I connected VCC, GND, Tx and RX. I used GPIO0 button to flash.

KLelong, 2019-02-12:
My current devices are operated via openhab, but I still can control them manually, hence the buttons.
But everyone has his own needs.

james-fry, 2019-02-12:
Ah go you.
You dont need to do this. :)
GPIO0 and GPIO9 are connected to button 1 and 2, and my tasmota config makes these buttons operate relay1 and relay2

james-fry, 2019-02-12:
Updated the pinout labels

KLelong, 2019-02-12:
Like I said before, the devices I currently use are Sonoff dual. Those have one button to control the two relays. And they are concealed in a wall. I have external buttons to open and close the shutters.
You made me think about extra functions : one of the shutters operates before a door. I could add a switch so the shutter does not go down when the door is open :smile:.

KLelong, 2019-02-12:
Very off-topic. How do you draw those blue boxes  with the text ? Or is that completely manual ?

james-fry, 2019-02-12:
PowerPoint. Because I was at work.
:D

nadynet, 2019-03-13:
Hi guys,
my module does not operate the switches. The phisical and webui buttons shows statechange in webui and also the main LED blinks on buttonpress but none of the two relay actually switch: Neither can I hear the click-sound nor the smd leds belonging to each relays light up. Every gpio binding is made via the wiki image. So according to the main LED flashes and the webui the modul works. Actually it does not SWITCHING.

After I bought I didn't try out with original fw. I connected the FTDI adapter VCC (set to 3,3V) and GND to the board pins - as there are two pinouts I used VCC and GND on the 6-hole-line, not on the 4-hole-line. TX and RX was pushed to the noted pins hardly and flash is managed... for the 3rd time.
I've successfully uploaded the firmware_sonoff.bin (modded version of sonoff.bin by **[stefanbode](https://github.com/stefanbode/Sonoff-Tasmota/tree/master/bin)**) via ESPeasy 2.0. During flash all of the LED-s behaved weirdly, smd leds for a given Switch status turned on off randomly and all the leds dimmed. I thought that the USB power via the FTDI adapter isn't enough (mains voltage wasn't connected). After flashing the module started to behave normally: own SSID, wifi setup, restart, module and gpio setup. Since then none of the combinations I tried was successful to toggle the relays. Pushing the mysterius 3rd button doesnt help. What can You advise? Is it possible that I somehow fried the board partially (during the flashing) to behave just like this? I've upgraded fw via webui to official sonoff-minimal.bin, and after that sonoff.bin and started over with numerous resets. Nothing helped. Thanks in advance.

KLelong, 2019-03-13:
I was going to ask if you read the wiki-page, but it seems it's gone. The link a few posts above does not work any more and I seem unable to find another one.
So did you read the wiki-page of the device ?
You need to set option 51 to 1 : 
Setoption51 1
And the configuration of the device : 
[image]

james-fry, 2019-04-30:
> but it seems it's gone

The picture and some info was added here:

https://blakadder.github.io/templates/geekcreit_2ch.html

KLelong, 2019-04-30:
Now it's officially supported, it's next besides the sonoff 2ch.

farpudo, 2019-05-05:
> Hi guys,
> my module (w/ enclosure version) does not operate the two relays. The physical and webui buttons show statechanges in webui and also the main LED blinks on buttonpress/clicks but none of the two relays actually switch over: Neither can I hear the click-sound nor the smd leds belonging to each relays light up. Every gpio binding is made via the wiki image. So according to the main LED flashes and the webui the modul works. Actually it does not SWITCHING.
> 
> After I bought I didn't try out with original fw. I connected the FTDI adapter VCC (set to 3,3V) and GND to the board pins - as there are two pinouts I used VCC and GND on the 6-hole-line, not on the 4-hole-line. TX and RX was pushed to the noted pins hardly and flash is managed... for the 3rd time.
> I've successfully uploaded the firmware_sonoff.bin (modded version of sonoff.bin by **[stefanbode](https://github.com/stefanbode/Sonoff-Tasmota/tree/master/bin)**) via ESPeasy 2.0. During flash all of the LED-s behaved weirdly, smd leds for a given Switch status turned on off randomly and all the leds dimmed. I thought that the USB power via the FTDI adapter isn't enough (mains voltage wasn't connected). After flashing the module started to behave normally: own SSID, wifi setup, restart, module and gpio setup. Since then none of the combinations I tried was successful to toggle the relays. Pushing the mysterius 3rd button doesnt help. What can You advise? Is it possible that I somehow fried the board partially (during the flashing) to behave just like this? I've upgraded fw via webui to official sonoff-minimal.bin, and after that sonoff.bin and started over with numerous resets. Nothing helped. Please advise. Thanks in advance.

Have you solved you problem, i'm in the same situation?

KLelong, 2019-05-06:
I used the "stock" tasmota version of a few version back. Since then I upgraded OTA.
Can't you try the "stock" version ?
At first it didn't work as expected, but pressing that third button fixed things.
Toggle your relays via the web-interface. If it doen't work, press the button once. Try again, etc.

farpudo, 2019-05-06:
> I used the "stock" tasmota version of a few version back. Since then I upgraded OTA.
> Can't you try the "stock" version ?
> At first it didn't work as expected, but pressing that third button fixed [...]

Yeah i'm using the lastest release of tasmota "stock" version. Tried to press that 3rd button but no success on relay activation. the ligh blink's when i press the web-interface button but the relay's won't trigger.

KLelong, 2019-05-07:
Not entirely sure, but it seems a cycle of 3, i.e. after three presses of that third button you get the same result as the start.
When OK, the next press results in the big LED flashing when toggling a relay via the web-interface, nothing else. 
The next press activates one of the relays a number of times, without action via the web-interface. Activating via the web-interface toggles a relay, but it turns on and off a number of times, with no logic that I can tell.
A next press results in the desired configuration : reaction via the web-interface in a predictable manner.
Are you sure you see the same configuration as my image of march 13 ? Only Switch 1 & 2 and Relay 1 & 2 are needed.

james-fry, 2019-05-07:
I agree with this.
We’re not really sure what button three does - presumably it configures the external IC between the esp and the relays. (It’s probsbly for inching etc that are not available with Tasmota fw).
There is only one state out of many that allows the tasmota fw to directly control the relays.
Just keep pressing button 3 and testing the relays work. If not press 3 and test again.

farpudo, 2019-05-08:
> I agree with this.
> We’re not really sure what button three does - presumably it configures the external IC between the esp and the relays. (It’s probsbly for inching etc that are not available with [...]

The problem persists, maybe some component is damaged or something.

farpudo, 2019-05-09:
> Not entirely sure, but it seems a cycle of 3, i.e. after three presses of that third button you get the same result as the start.
> When OK, the next press results in the big LED flashing when [...]

Do you have the sonoff stock firmware that you can provide? so i can reflash with the stock firmware so i can test it? ou is not possible?

james-fry, 2019-05-09:
Yes I have it. I dumped it before I flashed tasmota.
I have attached...

[PSF-B04.zip](https://github.com/arendst/Sonoff-Tasmota/files/3162879/PSF-B04.zip)

james-fry, 2019-05-09:
BTW I didnt have need to test it, so no guarantees, sorry

farpudo, 2019-05-09:
> BTW I didnt have need to test it, so no guarantees, sorry

Thanks, i will try that.

meingraham, 2019-05-09:
It is not possible to flash the stock Itead firmware from one Sonoff device onto another Sonoff device, even of the same model. There is some kind of device specific ID encode in each device's firmware.

nerdosity, 2019-12-04:
Just wanted to connect with you to share my experience.
I flashed a similar module, AC/DC 7-32V, same layout, same third button with strange behaviour.
The flash was correct (I used ESPEasy flasher to flash firmware, I flashed at least 30 devices with that application), I was able to flash it keeping the button on GPIO0 pressed, then I loaded tasmota on it, but it didn't went up anymore.
I tried with ESPHome, same issue. The module lights up correctly, but it doesnt' operate anymore, neither it does show an AP, neither it does connect to my wifi network via the preset ESPHome settings.
It seems dead.
I can flash it again as much as I want, I even flashed a 1mb blank file before flashing it again, but it stay dead.
I can solder pretty good, I have a semiprofessional soldering station with a really fine tip solder point, I checked with multimeter and no shorting / bad solder (RX/TX works perfectly).
Anyone has got an idea?

nerdosity, 2019-12-04:
Command: esptool.py --port COM5 --baud 115200 --after no_reset write_flash --flash_mode dout 0x00000 S:\Utente\Desktop\tasmota-basic.bin --erase-all

esptool.py v2.6
Serial port COM5
Connecting....
Detecting chip type... ESP8266
Chip is ESP8285
Features: WiFi, Embedded Flash
MAC: c4:4f:33:9c:bf:c5
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Auto-detected Flash size: 1MB
Erasing flash (this may take a while)...
Chip erase completed successfully in 0.0s
Compressed 469840 bytes to 326168...
Wrote 469840 bytes (326168 compressed) at 0x00000000 in 28.7 seconds (effective 130.9 kbit/s)...
File  md5: d7d0498bc19887f9ed3d7e5c405b8234
Flash md5: 8e1a0a1a2265d630456326799256576c
MD5 of 0xFF is 7c8555989de18cb3607b333ce723d8ad

nerdosity, 2019-12-04:
Seems OK, mine needed 12v power supply during flash:

Command: esptool.py --port COM5 --baud 115200 --after no_reset write_flash --flash_mode dout 0x00000 S:\Utente\Desktop\tasmota-basic.bin --erase-all

esptool.py v2.6
Serial port COM5
Connecting....
Detecting chip type... ESP8266
Chip is ESP8285
Features: WiFi, Embedded Flash
MAC: c4:4f:33:9c:bf:c5
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Auto-detected Flash size: 1MB
Erasing flash (this may take a while)...
Chip erase completed successfully in 5.2s
Compressed 469840 bytes to 326168...
Wrote 469840 bytes (326168 compressed) at 0x00000000 in 28.7 seconds (effective 130.9 kbit/s)...
Hash of data verified.

Leaving...
Staying in bootloader.

Firmware successfully flashed. Unplug/replug or reset device 
to switch back to normal boot mode.

FrancYescO, 2020-03-09:
someone know a back to stock procedure?
my module was this https://github.com/mirko/SonOTA/issues/168 actually i have tasmota on it but i need again a cloud-managed app

kirbah, 2020-09-13:
It took me two hours to flash. Maybe these details will be helpful for somebody else.
I flashed AC85-250V version.
1. I need to **connect GPIO0 and GND** during power up and pressed the button
[image]
2. Connect power and ground from CH340
[image]
3. Run tasmotizer and select com port and flash bin. I used recent sonoff lite bin file.
4. Connect TX and RX
[image]
5. Tasmotize! and make sure you TX and RX are connected all the time.

Few more notes:
- Relays leds might blink once you connect 3.3v and ground. Mode switch button 3 may be used to change mode and in some mode it will stop blinking.]
- Without connection GPIO0 and GND during poweron I got initial bright led near power board but it quickly became dim and my COM port disapppeared from the PC devices list.

FrancYescO, 2020-09-13:
SonOTA will flash this device very easily, but the tasmota management of this device is really bad, mainly cause of the button that switch modes and is not connected to the board logic

Bibinsa, 2020-11-21:
Hello !
I have a fairly similar PCB, tasmotized :+1: 
I would like to solder a DHT22 but I don't know where Vcc, GND and GPIOs are (GPIO14 GPIO10
[image]
).

Can anyone guide me?

Thanks

cite79, 2020-12-25:
> Ciao !
> Ho un PCB abbastanza simile, tasmotizzato👍
> Vorrei saldare un DHT22 ma non so dove siano Vcc, GND e GPIO (GPIO14 GPIO10 ).
> [image]
> 
> Qualcuno può guidarmi?
> 
> Grazie

Come hai fatto a tasmotizzare? Ho la stessa scheda ma non trovo dove collegare i fili per il flash grazie

FrancYescO, 2020-12-25:
SonOTA

kirbah, 2021-10-03:
SonOTA is not working for the new sw versions so it is actually useless for anything that I bought during the past few years.
Recently I got new module with the same PCB as on the recent screenshots above. SonOTA is not working. 
I had to solder 3.3V and GND pins to flash Tasmota fw.
1. Connect 3.3V (red on screenshot) and GND wires from CH340 and push the GPIO0 before connecting the CH340.
[image]
2. Once connected release the GPIO0. There should be COM port detected on your PC.
3. Open Tasmotizer and select file to flash. I've used the lite version.
4. Connect TX and RX pins. The pins are the same as in the old PCB version.
[image]
5. Tasmotize! and make sure TX and RX are connected all the time. It took around a minute.
[image]