arendst/Tasmota #1971: Random reboot

Chipz0r, 2018-02-19:
Hi
I have a sonoff sv on test here , just noticed it rebooted itself with this in the log

00:00:03 RSL: INFO1 = {"Module":"Sonoff SV","Version":"5.12.0","FallbackTopic":"DVES_95A5AE","GroupTopic":"sonoffs"}
00:00:03 RSL: INFO2 = {"WebServerMode":"Admin","Hostname":"sonoff-1454","IPAddress":"192.168.2.158"}
00:00:03 RSL: INFO3 = {"RestartReason":"Fatal exception:28 flag:2 (EXCEPTION) epc1:0xc2784000 epc2:0x00000000 epc3:0x00000000 excvaddr:0x00000298 depc:0x00000000"}

Any ideas ?

Thanks

Chipz0r, 2018-02-19:
and another

00:00:03 RSL: INFO3 = {"RestartReason":"Fatal exception:0 flag:2 (EXCEPTION) epc1:0xf4804001 epc2:0x00000000 epc3:0x00000000 excvaddr:0x00000298 depc:0x00000000"}

Chipz0r, 2018-02-19:
i setup a serial log and got this

Exception (0):
epc1=0xffff4025 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000298 depc=0x00000000

ctx: sys 
sp: 3ffffd00 end: 3fffffb0 offset: 01a0

>>>stack>>>
3ffffea0:  00063fff 4db40000 8cd53fff ffff4023  
3ffffeb0:  adcc7fff adcc3ffe 00013ffe 00010000  
3ffffec0:  7fff0000 00200000 62440000 00083fff  
3ffffed0:  00000000 00200000 0e2e0000 a5184010  
3ffffee0:  0ad93ffe 00024010 00220000 00050000  
3ffffef0:  00080000 098e0000 00024010 00020000  
3fffff00:  078f0000 00024010 e6600000 00053ffe  
3fffff10:  00000000 00200000 0e2e0000 eb820010  
3fffff20:  4f0c4023 cdc23fff efe43ffe 51b83ffe  
3fffff30:  f5a04024 fed03ffe 00163fff 3fff4d00  
3fffff40:  3ffec716 00000000 3fff4754 4023900d  
3fffff50:  2c02a8c0 00000054 00000000 0000001f  
3fffff60:  00000002 00000018 4022862b 3ffedfa0  
3fffff70:  3ffec6f0 3fffdcc0 3ffea558 3ffea558  
3fffff80:  4022859e 3ffedfa0 00000000 3fff4474  
3fffff90:  3fffdc80 00000000 3fff4754 40231533  
3fffffa0:  40000f49 3fffdab0 3fffdab0 40000f49  
<<<stack<<<

 ets Jan  8 2013,rst cause:4, boot mode:(3,7)

wdt reset
load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v401052cc
~ld

Anyone ?

arendst, 2018-02-19:
Considering the amount of different exceptions check your power supply.

Chipz0r, 2018-02-19:
Will do . Thanks