arendst/Tasmota #1979: Sonoff not start

DMP2-Tech, 2018-02-20:
Hello,
I made flash sonoff with tasmota but i have an error

Auto-detected: /dev/cu.usbserial-A8004Yy8
Uploading .pioenvs/sonoff-ds18x20/firmware.bin
warning: espcomm_sync failed
error: espcomm_open failed
error: espcomm_upload_mem failed
*** [upload] Error 255

I have change RX/TX and TX/TX but never.

My sonoff not start before...

Can you help me please ?

Thank's

reloxx13, 2018-02-20:
https://github.com/arendst/Sonoff-Tasmota/wiki/Beginner-Guide---Create-your-own-Firmware-Build#building-the-firmware

https://github.com/arendst/Sonoff-Tasmota/wiki/Upload#platformio

https://github.com/arendst/Sonoff-Tasmota/wiki/Hardware-Preparation#bringing-the-module-in-flash-mode

DMP2-Tech, 2018-02-21:
Hi,
Thank's for you response.

The problem is the next, on windows all ok but on mac i have an error 255.

DMP2-Tech, 2018-02-21:
Hi guys,

Solved https://github.com/arendst/Sonoff-Tasmota/issues/957

closed 2018-02-21 by DMP2-Tech