arendst/Tasmota #1980: mqtt message not send

tarbax, 2018-02-20:
Status wont change in home assistant when i set my sonoff manually on from the webgui.
there is no message being send to the broker when there is a change that did not happen true mqtt.
is there a setting i need to set with command line ?

SodoffBaldrick, 2018-02-21:
So I'm guessing that in HA you have the line
    _command_topic: "cmnd/Switch1/POWER"_   
Or whatever your topic is.

Do you also have the line
    _state_topic: "stat/Switch1/POWER"_
?
Again, adjust to match your topic. If you don't, then add it and restart HA.

tarbax, 2018-02-21:
home assistant config works if i controll it true HA itself On/off with status.
but when i controll it thue tasmota webgui itself there is no mqtt message being send to the broker do HA knows that it is on or off.

SodoffBaldrick, 2018-02-21:
Have you done as I suggested with the state topic?