arendst/Tasmota #1984: Sonoff Touch always toggles the relay every time it connects to mqtt broker.

jnherm, 2018-02-21:
Does someone using Touch and V5.12 experiencing Sonoff Touch always toggles the relay every time it connects to mqtt broker.

LucaAghemo, 2018-02-21:
I have the same problem with a Magic Home RGB LED strip controller and on a Kesen 2 gang touch switch, basically a Sonoff touch clone, and both are based on 8285 like the Sonoff touch. 
Something is "disturbing" my mqtt broker on my RPI, so it continue to go up and down. Tho this has to be solved, it doesn't give me problems on the other Sonoff and 8266 devices, that hold their state while reconnecting, but I have this problem on the 8285. 
Maybe the reason is connected to this? I flashed the Magic Home LED controller setting the Arduino IDE on "Generic 8285", while I flashed the Kesen switch, where I have also other problems discussed in another topic, with SonOTA, but both have this problem.
I hope someone can help :-)

bobindah, 2018-03-07:
I have the same issue. Using an older firmware 5.11.1 as don't think upgrading will help. Every time my rpi mqtt server restarts it the sonoff touch turns on then off again in very quick succession.
22:43:40 MQT: Connected
22:43:40 MQT: tele/bathroom/LWT = Online (retained)
22:43:40 MQT: cmnd/bathroom/POWER = 
22:43:41 MQT: stat/bathroom/RESULT = {"POWER":"ON"}
22:43:41 MQT: stat/bathroom/POWER = ON
22:43:41 MQT: stat/bathroom/RESULT = {"POWER":"OFF"}
22:43:41 MQT: stat/bathroom/POWER = OFF
22:43:41 MQT: stat/bathroom/RESULT = {"POWER":"ON"}
22:43:41 MQT: stat/bathroom/POWER = ON
22:43:41 MQT: stat/bathroom/RESULT = {"POWER":"OFF"}
22:43:41 MQT: stat/bathroom/POWER = OFF