arendst/Tasmota #1987: Kesen 2 gang switch went... almost fine! Need some help to make it perfect.

LucaAghemo, 2018-02-21:
I got this 2 gang Sonoff clone from a friend of mine:
https://www.amazon.it/gp/product/B076CJHPQ6/ref=oh_aui_detailpage_o06_s00?ie=UTF8&psc=1
basically a clone of Sonoff Touch 2 gang EU version, based on ESP8285.
I decided to open it, and see if I could flash it but... no luck. It has no reset button, the pin are forwarded on the PCB but... under the touch button 1! 
Before sending it ot the trash bin, or trying to sell it on ebay for 5$, I tryed a last, desperate thing: flashing it with SonOTA, under OSX 10.9.5 and Python 3.6.4.
Guess what: it worked! Almost...
Everything seems to work ok, mqtt connection, switching, I even connected it to Home Assistant. The only weird thing is that the touch button 2 doesn't work anymore. The relais 1 and 2 can be controlled via software or via the webserver page, everything connect and exchange informations, if I switch in the webserver page mqtt read and Home Assistant update, button 1 work perfectly, button 2... dead. This happen whether I power up the logic board alone with 3.3v or the whole thing assembled with 220v.
Quite a mistery to me, it used to work when it had standard firmware, I tried 1 min before flashing, can't understand why.
The debug log shows nothing strange, it seems to me that everything went well in flashing.
Maybe something has to be changed in the code, for this specific switch, maybe some GPIO input or similar, I'd like to do some test, but I don't know how to do. The brigth side is that now I can flash ota via the Tasmota web page :-)
Anyone can help? It would be nice to add another device to the compatibility list! :-)
Thanks in advance.

LucaAghemo, 2018-02-22:
Nobody answer?
What I need is some advice on what change I can try on the Tasmota code, to see if I can flash OTA a new firmware and make it work. Tasmota has hundreds of lines of code, it's a bit confusing for me. I'd really use some help.
Thanks

jupp70, 2018-02-22:
Hi. Don't know if it helps, 
Today I have successfully tasmota'ed the 3 gang version of the KESEN touch wall switch and all works as expected.
Settings : Device Type 30 Sonoff T1 3CH via SONOTA ( November Version )

LucaAghemo, 2018-02-23:
Thank you, this is very interesting. 
Of course I tried the setting "T1 2CH" in the web page, but it doesn't work. Maybe my version was a slightly more recent than november.
I guess I will have to find a use for it the way it is, and buy a Sonoff 2 gang for my actual need, but it's a shame... Those kesen switches are not bad after all, apart from being almost impossible to flash via serial.
I thought I might try some tweak in the code and reflashing via web, but I don't know where to look.

jupp70, 2018-02-23:
Yes. Right. Buying another might be the cheaper solution... But when you have ( too much ) time, you can give a try to ESPEays (https://www.letscontrolit.com/wiki/index.php/ESPEasy) , which AFAK supports the standard sonoff models and also a very generic model, where you can literally control each port as you want ( via Web Interace )

LucaAghemo, 2018-02-23:
Thanks again, I tried downloading the ESPEasy bin for 8285 and upload it over Tasmota web page OTA, but the upload fails, ging the error "Upload buffer miscompare".
I'm afraid that overwriting Tasmota firmware with EASPEasy cannot be done via Tasmota OTA update, which makes sense probably.
I checked the Tasmota source and find the place where pins are assigned, but it doesn't look like this is the problem. Anyway I think the only thing to do is trying, and see what happen.
I'll try re-uploading a Tasmota bin generated by Arduino and let you know.

jupp70, 2018-02-23:
The "Upload buffer miscompare" problem is known to be that that original image is too big. Try to OTA the Tasmota Minimal, which should work. After then, you can OTA the preferred NLS Tasmota image.

LucaAghemo, 2018-02-24:
Thank you, fumbling in the code I discovered the Minimal, I happily succeded in OTA that, then I OTA the final image... and the device is bricked! Lol! 
Ok, it was his destiny, the poor bastard has passed too many...
As soon as I have to order something on Amazon or similar I'll buy a couple of Sonoff.
Thanks a lot for your help, btw!

LucaAghemo, 2018-02-27:
I post another update, just in case somebody still reads this post and want to help me.
I was able to solder 2 wire on the board, under one of the button (the one that still works... lol) and reflash Tasmota. Yhaa!
But then again, button 2 doesn't work. It takes the command via mqtt, it switches the relay 2, but the button is dead.
I'd be very grateful if somebody could tell me where should I search in the code to try some tweaks. I tried changing some pin in sonoff_template.h, in the "Sonoff T1 2CH" section, but it seems that the firmware generated has something wrong, as trying OTA update doesn't work and give a restart with reason: {"RestartReason":"Blocked Loop"}.
This has become a challenge, I really would like to go to the end and see this damn button 2 working!

LucaAghemo, 2018-02-27:
And another small update, I tried to set as GPIO_KEY2 any available pin, leaving the one marked Flash or TX and RX, but it didn't work. Either the button 2 didn't work or the whole "Sonoff T1 2Ch" setting was refused.
If someone has a good idea that's the moment, otherwhise I'll use it as a "one gang", and I'll be damned to look at this second button with hate for the rest of my life :-)