arendst/Tasmota #1988: Feature request: adding Obi Socket

Frickelmaster1, 2018-02-21:
German DIY store "OBI" is selling ESP8266-based sockets (very similar to Sonoff S20) for less than 10 EUR. 

I got two of these sockets. It's quite easy to flash Tasmota on it but it was neccesary to modify sonoff/sonoff_template.h to make it work correctly. The same reason as with other clones: they assigned the GPIOs randomly to the available functions and chose others than ITEAD with their Sonoffs.

It would be nice if this configuration snippet could be added to the official sonoff_template.h.

I suggest adding these lines to sonoff_template.h for making the Obi Socket work:

`{ "Obi Socket",   // Obi socket (ESP8266) - https://www.obi.de/hausfunksteuerung/wifi-stecker-schuko/p/2291706
     0,             // GPIO00 Flash jumper - not available
     0,             // GPIO01 
     0,             // GPIO02
     0,             // GPIO03
     GPIO_LED1,     // GPIO04 LED on top and in switch button
     GPIO_REL1,     // GPIO05 Relay 1 (0 = Off, 1 = On)
     0, 0, 0, 0, 0, 0, // Flash connection
     0,             // GPIO12
     0,             // GPIO13
     GPIO_KEY1,     // GPIO14 switch button
     0,
     0,        // GPIO16 Led (1 = On, 0 = Off)
     0         // ADC0 A0 Analog input
  },`

reloxx13, 2018-02-21:
could you add images from the pins with description which pin is what?

Frickelmaster1, 2018-02-21:
The descriptions of the pins are printed on the PCB. They are similar to the pin names on any FTDI adapter.

reloxx13, 2018-02-21:
you could aso try to use the generic module on teh we interface and set up your gpio config.

[image]
(note: this is not from obi)

reloxx13, 2018-02-21:
btw @arendst , is their a reason why not all GPIOs are user selectable? would make more sense for a "generic" module.

martin-ger, 2018-02-21:
Some pics from the opened device can be found here: https://github.com/martin-ger/ESP8266-WiFi-Socket

martin-ger, 2018-02-21:
It works somewhat different than the Sonoff basic: a low pulse on GPIO12 switches the relay on, a low pulse on GPIO5 switches it off.

Frickelmaster1, 2018-02-21:
In my Tasmota there is no "Generic" type so I had to edit the templates...

tobox, 2018-02-21:
"Generic" has been added during the last days, replacing Wemos D1 Mini and NodeMCU variants.

Jason2866, 2018-02-21:
@Frickelmaster1 
Does it work with your setup? 
I ask this because martin-ger describes the on off function different

Frickelmaster1, 2018-02-22:
My first Obi socket works fine, the second doesn't. I flashed the binary with ESPloader to the sockets. The first is OK, the second (with exactly the same firmware) is accessible via web interface, the on-off works (including the blue LED) but the relay isn't switching at all.

Maybe Martin can explain how he modified Tasmota to make it work with the Obi sockets?!

martin-ger, 2018-02-22:
I did not use Tasmota but my onw software - works fine. I am not into the Tasmota code and platform.io, but use the esp_open_sdk and my MQTT framework.

The control of the relay works as follows: 
- a low pulse (150 ms works) on GPIO12 switches the relay on
- a low pulse on GPIO5 switches it off

The two ports are even named "RELAY ON" and "RELAY OFF" on the PCB of the ESP.

martin-ger, 2018-02-22:
In the sources I read something about "latch" - should be similar the this issue #58.

Frickelmaster1, 2018-02-22:
Although I can't imagine maybe there are differences between my sockets I'll compare them tonight. 

Strange... one is working fine, the other one doesn't switch the relay - with identical firmware.

I'll try to use a modified EXS relay configuration on the second socket. Thanks, Martin!

thxthx0, 2018-02-22:
I have solved the  'pulse issue' by setting GPIO12 to always 0 (as LED) and works fine.

[image]

Gulaschcowboy, 2018-02-22:
Got mine today and can confirm @thxthx0 solution. Thanx

HudecityDave, 2018-02-22:
Same here. Confirm @thxthx0. Firmware flash with GND/GPIO0 pin method, update to tasmota 5.12, shown config above. works. THANKS! What a great community!

reloxx13, 2018-02-23:
added to wiki https://github.com/arendst/Sonoff-Tasmota/wiki/Obi-Socket

etamtlosz, 2018-02-23:
How did you manage to access Tasmota on the OBI plug? I'm struggling since yesterday evening. I'm able to flash it, but no  wifi, no interface, nothing.

reloxx13, 2018-02-23:
build the firmware yourself, enter wifi credentials in the user_config.h and flash again.

dont forget to reset the cfg_holder

enter wifi data
change cfg_holder
flash
change cfg_holder back to default
flash again.

etamtlosz, 2018-02-23:
I'm doing something wrong.

edited my wireless settings, including ip
changed cfg_holder to the actual day
flashed
reconnected
changed the cfg_holder to 0x20161209
flashed

It won't connect to my wireless, no ping, etc.
I think I'll wait until some more detailed tutorials are going to be published for this plug.

reloxx13, 2018-02-23:
their is nothing more to add, its an esp like in all other devices.

do you got other devices with tasmota which are connected to your wifi?

ip on 0.0.0.0 for dhcp.

no special chars in wifiname and password.

how you want to send a ping without an ip or connection to your router wtf dude...

even if it says its flashed 100% it can be not flashed due a defect usb flasher.

etamtlosz, 2018-02-23:
My ip was set to a fix IP, this is why I was monitoring the IP with ping.
I'll try with DHCP, now.

I'm using the correct flashing method, I assume - Generic ESP8266 Module.

gunnwo, 2018-02-26:
I have the latest firmware flashed. As the wlan was preconfigured access was possible. But I'm struggeling with the OBI socket and generic Type. Comparing the screen shot from above with the settings I only get 4 entries in the module configuration: GPIO1/3/4/14. So how do you enable the rest of the fields and enter the settings?

etamtlosz, 2018-02-26:
Change it to generic, save. It will restart an you'll have the other options as well.

thxthx0, 2018-02-27:
If someone wants to confgure more sensors:

the onboard ESP8266 Module with 1MB is similar to '**WT8266-S1**' 

[image]

cyablo, 2018-02-27:
Got mine flashed with tasmota and used the settings suggested by thxthx0, but my relay does not turn on. Only Red & Blue LED's turn on. @Frickelmaster1 : Did you get it working on yours?

Steakschen, 2018-04-24:
I tryed to flash it with esptool.
Connected gpio0 to grnd
it flashes but then dosent boot up.

after 2-3 trys i got:

00:00:09 APP: starte neu

 ets Jan  8 2013,rst cause:4, boot mode:(1,6)

wdt reset

and nothign happend further, i used 5.12

Jason2866, 2018-04-24:
@Steakschen PreBuild Image or self compiled?

Steakschen, 2018-04-24:
@Jason2866  pre-build
https://github.com/arendst/Sonoff-Tasmota/releases

Steakschen, 2018-04-24:
i flashed 2 times, and booted it 5 minutes, now it works 
solved :-) wait 2min+ to boot

renne, 2018-05-14:
The current [c't magazine 11/2018](https://www.heise.de/ct/entdecken/?jahr=2018&ausgabe=11) presents the Obi/Euromate power socket on [page 48](https://www.heise.de/ct/ausgabe/2018-11-WLAN-Zwischenstecker-Obi-Wifi-Stecker-Schuko-4040742.html) and explains how to open the power socket, solder the serial port and flash and configure Tasmota on [page 152](https://www.heise.de/ct/ausgabe/2018-11-WLAN-Steckdose-aus-dem-Baumarkt-mit-alternativer-Firmware-Tasmota-4038695.html). The vendor is [Euromate](http://www.euromate.de/). The price at Obi is still 9,99 €.

etamtlosz, 2018-05-14:
It would be nice if someone could post some screenshots.

Jason2866, 2018-05-14:
Just my humble opinion. 
If a OBI is near you, this is a deal if you want the device fast.
If you not in a hurry i would choose sonoff. The build quality of a sonoff S20 is better and it is cheaper.

Jason2866, 2018-05-14:
[image]

Jason2866, 2018-05-14:
@renne Do you think the OBI device is really tested to be VDE compliant? (hüstel)
CE this means nothing, every crab has this....

cremofix, 2018-05-23:
hey folks,

I build the binaries with Atom and flashed the OBI Socket with the latest Tasmota.
worked fine
I used the workaround described in ct magazine and by @thxthx0 (GPIO12=LED2 and GPIO5=Relais1)
it works fine. I can access the sockets IP and switch it on/off via browser

But I fail to acces the socket via MQTT

I have mosquitto installed on a raspberry as mqtt-broker
I can connect to mosquitto with mqtt.fx
unfortunately it seems that I do not have any connection between the socket an the mqtt broker
this are my mqtt settings:
[image]
did I forget something?
any idea?
thanks

dzett, 2018-05-23:
Hi there, 

as per the described behaviour of the stock firmware, the Obi socket seems to use a latching relay only triggered via a short pulse. 

Such a relay cannot be triggered from the "Generic" configuration because even though there is a capability to deal with latching relays introduced with 533855c , this is hardcoded against the module type EXS_RELAY which unfortunately has the Two relay outputs on different GPIOs.

I would prefer to use the relay as intended (hence not giving a constant high output on the "On" side of the relay). Is there any easy way to refactor this to be not bound to the module type but have some other "configuration switch" to indicate that GPIO_REL1 and GPIO_REL2 are actually the two latching "ends" of the relay configured?

renne, 2018-05-24:
@dzett
You're right. Firing one coil constantly not only increases wear of that coil but also makes the switching of the relaiy unreliable and increases switching time which can lead to burned contacts.

@arendst 
I suggest to add `xx Relay N ON` and xx `xx Relay N OFF` in the GPIO list of device configuration. Before and after firing a coil both GPIOs should be set to low.

cremofix, 2018-05-24:
hey folks,

I do not know what I changed but now everything works fine
I can subscribe and publish via MQTT and the OBI socket does everything correctly.
I agree with dzett and renne. the workaround with GPIO=LED2 and coinstant fire on the coil does not seem to be ideal. 
I found out that the socket needs more than 2 Watts when switched on and only 0.6 W in standby
It gets about handwarm. So running the latching relay without constant current should be a big improvement
@arendst : would it be a big job to add a profile for this OBI socket?
probably after the article in the ct magazine there will be a lot of persons buying this gadget.

arendst, 2018-05-24:
Get me one for testing (I cannot order by obi) and I will see  how to get it implemented the correct way (using the pulses).

dzett, 2018-05-25:
@arendst After playing around with the socket a bit more yesterday (trying to use the existing latching implementation for module EXS_RELAY which I assume is working and remapping the GPIOs), I am not so sure that the Obi socket really has a latching relay, as it would turn off right away when the pulse was over. 
Also, I looked for some sort of datasheet for the relay that is used (same type as seen on the pictures in earlier posts) and from what I found it seems to be just a plain standard relay. Does anybody still have the stock firmware on it and actually look at the outputs?

cremofix, 2018-05-25:
I just opened my socket
and found this relais
[SMIH-NingBoSongle.pdf](https://github.com/arendst/Sonoff-Tasmota/files/2038388/SMIH-NingBoSongle.pdf)

this is not a latching relais. perhaps they use different models?

thxthx0, 2018-05-25:
That's right, it is not a latching relay.

There is a standard 16 amp relay, 
but driven by 'some kind of latching' transistor circuit.

I couldn't measure any differences in power consumption 
by using stock firmware, Tasmota (GPIO12 always 0) or martin-ger's ino.

The relay is driven at about 140mA and 5.4 V

dzett, 2018-05-25:
Hi all,

@thxthx0, thanks for clarification - should have checked my inbox a bit earlier :-)

I just picked up another one at Obi and can confirm that across the pins driving the relay coil, there is a constant voltage visible while the output is active. The "Relay On" trace from the ESP8266 board goes into something that looks like a circuit around two transistors (flip flop, eh? good old times). So I guess we don't need any code changes after all. Thanks to all!

palto42, 2018-06-01:
There seems to be a latching relay option in the Tasmota code for the EXS_Relay (see also #58). Can this be used with Generic model option (e,.g, enabled via Console) or would this require the code to be compiled with specific settings?
I could not find detailed information about the latching relay option on the Tasmota wiki.

thilokru, 2018-06-05:
Is it normal that the devices get hot? When turned off, it is warm to the touch, but when on it realy heats up.

thxthx0, 2018-06-08:
Yes, seems to be normal (and a 'feature' of this relay ;).

Power Consumption of the relay is about 0.75W
(data sheet says 0.72W).

It also works reliable @ 4V and 0.4W,
but temperature isn't much different  
(~ 40°C at the solder joints & case opened).

rudi48, 2018-06-17:
Unfortunately the Button1 wired to GPIO14 no longer allows to use the **Button usage** (usually wired to GPIO0), described on https://github.com/arendst/Sonoff-Tasmota/wiki/Button-Usage

Does anybody know how to regain this Button-Usage, e.g. for providing an **Access Point** with IP address 192.168.4.1?

This is needed to change to another local network, **without opening the case**.

rudi48, 2018-06-17:
A practical solution for the **original Button1 usage** is to connect the **Button1 pin to GPIO0** with a plug. If it is no longer wanted, just pull the plug off GPIO0. See the **picture** and the **screen shot:**
Sonoff_OBI_GPIO0_Button1_DSC07687.jpg
Sonoff_OBI_Config_GPIO-jumper.png
[image]
[image]

josch, 2018-08-05:
Are the tasmota developers against adding a template for the OBI socket and would prefer that one uses the Generic setting?

Or would a pull request be appreciated?

nobirdnobird, 2018-08-13:
Yes, I will also really appreciate to add the adapter configuration to the template !

BR

Jason2866, 2018-09-30:
@renne  OBI plug is a cheap china device too... 
https://github.com/arendst/Sonoff-Tasmota/issues/1988#issuecomment-388914281
See: https://github.com/arendst/Sonoff-Tasmota/pull/3944
Model tested: WFG-1
Models maybe working: WFF-1, WFE-1, WFI-1, WFD-1, WFR-1, WFA-1, WFU-1
Manufactured by: Cixi Yidong Electronics Co. Ltd - http://www.cn-yidong.com/
Sold by: OBI DIY market in Germany - https://www.obi.de

dreamlayers, 2018-11-29:
OBI Socket setting also works for Orvibo S25US https://fccid.io/2ACLPS25US/Internal-Photos/Internal-Photos-2970651

Button, relay and blue LED work. The S25US LED can also be red if the blue is turned off and GPIO12 is on. Though red and blue cannot be on at the same time. I experimented with this using Sonoff 4CH module setting.

mkormendy, 2019-07-18:
I wonder if this Works for the Orvibo S31 (which has power monitoring) as well. I have opened up the outlet and here's what it looks like:

[image]
[image]
[image]

roger-, 2019-11-02:
Interested in the S31 as well.  Any try this yet?

ascillato, 2019-11-02:
@roger- 
Please, check https://github.com/arendst/Tasmota/blob/development/README.md#note

roger-, 2019-11-30:
> @roger-
> Please, check https://github.com/arendst/Tasmota/blob/development/README.md#note

Yeah, I’m just curious if the template is the same before I flash it.

KSumwalt, 2019-12-13:
@roger- Did you figure out how to connect to this and flash it?  This is the first unit I am trying to flash and I don't see labeled points to connect to.

mkormendy, 2019-12-14:
The new tuya convert with https should be able flash this over the air.

roger-, 2019-12-14:
@KSumwalt no, haven’t tried yet. 

@mkormendy really? Do you have a link to someone who’s tried it?

[5 more replies]