arendst/Tasmota #1989: Feature request: RF Bridge Custom Firmware EFM8BB1 (RF Part)

Jason2866, 2018-02-21:
Hello Theo Arends,
thank you for your incredible firmware for sonoff/esp32 devices!
I use a RF Bridge and some of my RF 433 devices are not recognized.
This is a limitation in the stock RF firmware. Could you maybe support this alternative
Firmware: https://github.com/Portisch/RF-Bridge-EFM8BB1
Thank you in advance for taking a look

ErikAndren, 2018-02-22:
One problem is that you need to flash the EFM8BB1 device which currently involves soldering two pins, building a separate arduino application on an arduino board, hooking it up to the pins, possibly adding a level shifter in between if you are not using a 3.3 V arduino.
If one could flash the EFM8BB1 from the ESP8266 life would be much simpler.

Jason2866, 2018-02-22:
@ErikAndren 
Whats the problem with soldering pins and adding a level shifter? At Ali tnis parts costs nearly nothing

ErikAndren, 2018-02-22:
It's not a problem as much as it is a hassle and a large risk for the unexperienced user to break his stuff.
If we could find a solution where it is enough to wire the C2 and C2K pins to the ESP8266 and add support for flashing the EFM8BB from the ESP8266 preferably via the web interface I would think that is the ideal solution

Jason2866, 2018-02-22:
Yes that would be nice. But lets walk before run. If Theo would implement support for the alternative firmware i would buy a second RF Bridge for testing purpose. I wouldnt touch my installed working and needed one....

arendst, 2018-02-24:
Sounds nice but from the Portisch website I understand it's still in development.

@Jason2866 why not get your second bridge and try it yourself? From Portisch site I understand the current commands are still supported so You can start with uploading the firmware. After take a PR and edit file xdrv_04_snfbridge.ino to support the new commands...

Jason2866, 2018-02-24:
@arendst Will order a second one "for playing"....

ErikAndren, 2018-02-24:
I have flashed mine to this firmware and I have very mixed results. Sometimes it works, sometimes it doesn't. Currently I'm trying to debug it and rewrite the code to be a bit more clean.

Jason2866, 2018-02-25:
@ErikAndren Sounds Good. Could you keep us informed about that?
If it get my 2nd one i would be happy to test your improvements.
My PM is obermeier.johann(at)googlemail.com

KurtMar, 2018-02-25:
I'm looking forward to this, so I might get my Nexa and Telldus products to work.