arendst/Tasmota #1990: Feature Request: RCSwitch input from 433 mhz

kenmcc, 2018-02-21:
Hi, I've got some way along this myself, but am now stuck.

The idea: select one of the GPIOs as the input from an rf433mhz receiver, which will use RCSwitch to interpret the data received, and if it's recognised will publish it on mqtt

So I've managed to add this option to the dropdown, and have even added it to the xsns structure, and I can see log messages from the init, prep, json_append and web cases; the only issue is that the PREP one is only called very infrequently, it looks like every 5 minutes or so. 

How might I increase the frequency of this so that it would check for data on that pin much more frequently (e.g. millisecond interval)?

Presumably this must be done for buttons etc already, but I can't find where
Thanks
Ken

kenmcc, 2018-02-22:
Oh i think i might have found something this morning. I'm possibly on old version of firmware, but looking at the latest code, I see a 
FUNC_EVERY_50_MSECOND
which is used in a call to XsnsCall ; presumably if I put a case handler in my XSNS driver for this, it will get called every 50ms also? 

I'll upgrade tonite and try that, but if I'm on the wrong track maybe someone will have a better suggestion?

kenmcc, 2018-02-22:
yep, that's got it being called plenty frequently. now to get it checking the rcswitch. tomorrow i think

kenmcc, 2018-02-26:
I've managed to get it working, listening for data from the 433 receiver, and pushing out an MQTT message as a status. Openhab should be able to act on that message, and do stuff - (e.g. i'm using a 433 PIR sensor unit from ebay, so it could be used to activate the hall lights. can also be used to (unreliably of course) indicate the status of any 433 power switches i have in the house, but i'm less concerned about that for now)

moritzj29, 2018-05-04:
Hi Ken, I just stumbled onto this topic. I was searching if someone already implemented a 433mhz (universal) receiver for the tasmota firmware. Since I'm running a couple of Sonoffs, I would rather use these ESPs than an additional microcontroller for publishing 433MHz signals over MQTT to openhab. I think it is very attractive for wireless switch buttons since they are way cheaper than any WiFi or Zigbee remotes.

Are there any plans of adding your code to the tasmota firmware? Unfortunately, I'm more of a hard- than a software guy so I probably won't be able to code it myself like you did ;)

kenmcc, 2018-05-04:
Hi, 
i forgot all about this! Loads on my plate at the moment, so it's slipped to the bottom of the pile. I'll see if I can dig it out and get it into a working state again and publish it. Can't promise it'll be done this weekend but I'll try.

Onil09, 2018-05-08:
I will be very happy if you find and post the firmware

kenmcc, 2018-05-13:
updated now to allow any pin to be used (as selected in the configuration page) rathter than defaulting to pin 14. 

I've fixed the pull request up also

Onil09, 2018-05-14:
collected, but does not work yet, maybe the console is not suitable (although I used ttl, the diode blinked when pressed), I will find and try another one.
sorry for my english, I'm from Russia, I use an interpreter

kenmcc, 2018-05-15:
i've had different results with different rf remote control signals; some work better than others. I found the energenie one https://www.ie.screwfix.com/energenie-13a-wireless-remote-control-sockets-pack-of-3.html to be very good. also i have success with this PIR https://www.ebay.co.uk/itm/433Mhz-Wireless-PIR-Sensor-Motion-Detector-AutoDial-Home-Security-Alarm-System-/152210105399 

I'd suggest you firstly get a simple rc-switch sketch running on an arduino, using the RF receiver and the RF transmitter you have, and see if that works first.

huexpub, 2018-05-18:
Archiving built core (caching) in: C:\Users\ADMINI~1\AppData\Local\Temp\arduino_cache_856591\core\core_esp8266_esp8266_generic_UploadTool_esptool,CpuFrequency_80,FlashFreq_40,FlashMode_dout,UploadSpeed_115200,FlashSize_1M0,ResetMethod_ck,Debug_Disabled,DebugLevel_None_____5b25c9b537d1fb7226f87cd9d48cdd52.a
c:/arduino-1.8.5/portable/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld.exe: C:\Users\ADMINI~1\AppData\Local\Temp\arduino_build_128709/sonoff.ino.elf section `.text' will not fit in region `iram1_0_seg'

collect2.exe: error: ld returned 1 exit status

arduino1.8  this error with Sonoff-Tasmota-rcswitch,

kenmcc, 2018-05-18:
did you select the 1M (no SPIFFS) flash size? what board did you build for? did you enable any other option that's not normally enabled?

huexpub, 2018-05-18:
same error 4M(1M), 1M(512K)

kenmcc, 2018-05-18:
Weird. Arduino 1.8.4.

4M(1M) worked for me:
[image]

and 1M(no SPIFFS) also:
[image]

1M(512K): failed like yours

Onil09, 2018-05-30:
everything works fine, although it sometimes misses a signal. Thank you!

kenmcc, 2018-05-31:
great stuff. yes it will occasionally miss a signal, thats my main issue with 433mhz switches is that you can't reliably determine their state based on listening to broadcast signals.
ah well, better than nothing!

closed 2018-07-30 by stale[bot]

moneybag, 2018-11-04:
doesnt work for me. 106 RFrec on D6 and 
#define USE_RC_SWITCH                            // Add support for RF transceiver using library RcSwitch (+2k7 code, 460 iram) in my_user_config.h
console doesnt show ingoing codes. did i forget something?