arendst/Tasmota #1993: Change settings without reflashing

eporedieis, 2018-02-22:
Hello, I want to change any settings of the sonoff:
#define CFG_HOLDER
#define APP_PULSETIME
#define SWITCH_MODE PUSHBUTTON

How can I do it **without disconnect** Sonoff from the power and reflashing it? Could I use his console? 
Sorry for this simple question but I'm not able and I don't found the solution on the wiki. Thank you all!

SodoffBaldrick, 2018-02-22:
You can flash OTA which would allow you to upload a custom firmware without disconnecting it. It would reboot, but you could do it completely without touching it.

Or, look at this page
https://github.com/arendst/Sonoff-Tasmota/wiki/Commands

Variables:
  PulseTime
  SwitchMode

I can't see anything for CFG_HOLDER though, unless it's CfgDump?

eporedieis, 2018-02-22:
I read [in this issue](https://github.com/arendst/Sonoff-Tasmota/issues/894) that you have to change the Cfg_Holder every time or Sonoff will not view the changes. Isn't correct?

So, sorry for the question, to change the PulseTime I can use this command in the sonoff console:
tele/sonoff/PULSETIME 5
Is it correct?

Thank you very much!

curzon01, 2018-02-22:
> I read in this issue that you have to change the Cfg_Holder every time or Sonoff will not view the changes. Isn't correct?

No, simply fire the command and the setting will be stored and kept also on reboot, repower. Using change of CFG_HOLDER is not needed.

> So, sorry for the question, to change the PulseTime I can use this command in the sonoff console:
tele/sonoff/PULSETIME 5
Is it correct?

No, simply fire
`PULSETIME 5`

What you wrote is a MQTT topic which is send from Sonoff to a MQTT server. If you are not familary with MQTT use [Commands](https://github.com/arendst/Sonoff-Tasmota/wiki/Commands) as is within Web server Console.

eporedieis, 2018-02-22:
Greats, now it works!! Thank you very much, curzon01 😃

closed 2018-02-22 by eporedieis