arendst/Tasmota #1995: Communication RF-brigde and Smartwares SH5-TBD-02A Dimmer

mcoenen1994, 2018-02-22:
Hi all,

Currently I have tasmota flashed on my Sonoff RF-bridge and it is working pretty well with my motion sensor. Now I have bought an RF dimmer from Smartwares and googled a lot about the communication with these kinds of dimmers but was not able to learn/control the dimmer. Does anyone have experience with controlling the dimmer with tasmota? Or is it impossible to do?

Thanks in advance,

Maikel

Jason2866, 2018-02-22:
Needs #1989

tam481, 2018-06-28:
I've been having the same issue since updating to the latest 6.0.0a from 5.2.1
23:37:23 MQT: Attempting connection...
23:37:23 MQT: Connected
23:37:23 MQT: tele/TLamp/LWT = Online (retained)
23:37:23 MQT: cmnd/TLamp/POWER = 
23:37:25 MQT: stat/TLamp/RESULT = {"POWER":"OFF"}
23:37:25 MQT: stat/TLamp/POWER = OFF (retained)
23:38:34 MQT: Attempting connection...
23:38:34 MQT: Connected
23:38:34 MQT: tele/TLamp/LWT = Online (retained)
23:38:34 MQT: cmnd/TLamp/POWER = 
23:38:34 MQT: stat/TLamp/RESULT = {"POWER":"OFF"}
23:38:34 MQT: stat/TLamp/POWER = OFF (retained)
23:40:24 MQT: Attempting connection...
23:40:24 MQT: Connected
23:40:24 MQT: tele/TLamp/LWT = Online (retained)
23:40:24 MQT: cmnd/TLamp/POWER = 
23:40:26 MQT: stat/TLamp/RESULT = {"POWER":"OFF"}
23:40:26 MQT: stat/TLamp/POWER = OFF (retained)
23:53:13 MQT: Attempting connection...
23:53:13 MQT: Connected
23:53:13 MQT: tele/TLamp/LWT = Online (retained)
23:53:13 MQT: cmnd/TLamp/POWER = 
23:53:15 MQT: stat/TLamp/RESULT = {"POWER":"OFF"}
23:53:15 MQT: stat/TLamp/POWER = OFF (retained)
00:01:56 MQT: Attempting connection...
00:01:56 MQT: Connected