arendst/Tasmota #1996: PowerOnState for Dual POWER1:ON, POWER2:OFF

celeroll, 2018-02-22:
I use a sonoff dual to controll my living room light which has two channels. (yellow and dimmed blue light).
I'm not able to figure out how to controll PowerOnState for the dual, so that allways POWER1=ON and POWER2=OFF. In the PowerOnState option I cant different between this two channels. 
I need PowerOnState because the initial turn on on light is with a physical switch and it should usually be set only on yellow light.

reloxx13, 2018-02-23:
currently not supported for multi relays

celeroll, 2018-02-23:
Any idea for a workaround?

reloxx13, 2018-02-23:
Dont power them off ^^

reloxx13, 2018-02-23:
umm i just checked teh code,
did you try in console
PowerOnState1 1
PowerOnState2 0
?

please provide feedback, could be wrong documented in wiki.

reloxx13, 2018-02-23:
or
PowerOnState 10

celeroll, 2018-02-23:
Unfortunatelly it is not working. I have tried all the options

ageorgios, 2018-02-24:
setoption14 1
interlocking

celeroll, 2018-02-25:
It didn't work. On poweron the lamp was always off (both chanells). And when I wanted to turn one on or off, the other chanel didnt react at all. Maybe I need to change something like delay for interlocking.