arendst/Tasmota #1997: PlatformIO Compiling Issues

tango156157, 2018-02-22:
Hello, I want to re-compile the TASMOTA firmware using ATOM and PlatformIO. I am able to modify
user_config.h file and compile successfully the first time and generate .bin files - No issues there, however when i make subsequent changes to parameters in user_config.h file for say other parameters like MQTT etc, these changes do not reflect in the re-build action in PlatformIO. The updated .bin file which gets generated does not have the updates as expected and there are no compiler errors. What am i doing wrong here? This is my first experience using PlatformIO from the regular Arduino IDE platform.  Any advise will be appreciated here as been frustrated by this issue :-) Thanks

Jason2866, 2018-02-22:
Have you tried the menu point "Clean" Same menu as "Build"

tango156157, 2018-02-22:
Hi, thanks for your response. I did actually and i remember that option not helping really.

Jason2866, 2018-02-22:
Did you "Open Project Folder" to import the Sonoff Project?
Open New Project will fail!

jhlweb, 2018-02-22:
This is because all settings are stored in Flash.
There are multiple ways to reset. 
1) easiest is the reset button on the web-page under configuration
2) compile with an other value : `#define CFG_HOLDER             0x20161209` in the user_config.h file

reloxx13, 2018-02-23:
> compile with an other value : #define CFG_HOLDER 0x20161209 in the user_config.h file

do this, flash, change cfgholder back and flash again to stay on default config holder to avoid future updates to reset your config

tango156157, 2018-02-23:
Great thanks reloxx13 and jhlweb....your advise did the trick and now I am able to update changes and then re-compile. I however am stuck in one trival issue where I want to modify the Pulse time in the User_Config.h file.  Under #define APP_PULSETIME, I want to set a time of say 3 seconds for the Relay to be ON and then turn off. ShouldI be putting value of 0.30 here or 0.113? Please do guide and thanks again for excellent and quick support for a newbie :-)  Cheers~

reloxx13, 2018-02-23:
103

jhlweb, 2018-02-27:
@tango156157 no  30.0, the multiply factor is 0.1 that computes to 3 sec.