arendst/Tasmota #1999: Sonoff constantly restarts without WLAN.

danielpondruff, 2018-02-23:
If my WLAN is not available, the relay switches itself on and off very quickly. The Sonoff seems to restart over and over again. That doesn't seem to me at all healthy for the equipment that's attached to it. 

I would like to stop this behaviour.

reloxx13, 2018-02-23:
check wiki -> commands wifi -> wifi manager

isvedic, 2018-02-26:
Yea, by default it is set to WifiConfig 3, WPS causing reboot if not connected, try WifiConfig 5 in console

rcarmo, 2018-03-17:
Referencing #2162, where I added a note that this behavior fried a lightbulb. Loss of configuration and the relay clicking upon reboots in WiFiManager are probably best handled as two different issues, but both should be tackled.