arendst/Tasmota #2000: New Flashed

XtraLarge, 2018-02-23:
Hello,

I've flashed 3 Sonoff S20 devices with the "de" version. Two were normal as expected, but one shows an unexpected behavior.

After flashing I've pressed the button 4-times. The WLAN sonoff appears. I've entered the WLAN informations of my network and pressed the save button.

After that I've not seen an dhcp lease of the device. (my fault there was an static lease)
So I've pressed the button 4-times again and ... (Thats the Problem) ....

An AP named "ited-<mac>" appeard and this network was password protected.
After some googling, (<< nice word)I've found that the password is "12345678" and so I was able to ending my configuration.

2 Questions:
- Is this natwork name with password normal?
- How can I change this AP Network name and password?

best ragards
Willi

reloxx13, 2018-02-23:
im sure you got confused and bressed the button on a non tasmota flashed device.

itead is for orginal firmware.

build your firmware youself, enter wifi credentials in user_config.h, flash again and be happy.

XtraLarge, 2018-03-10:
OK, 
but explain me, how i got the tasmota webinterface after enter the  wlan key "12345678". 
And this is really the truth!!! I swear!