arendst/Tasmota #2001: Itead Si7021 not detected when temperature falls below +1C

mbolsteins, 2018-02-23:
I have a Sonoff TH10 with Itead's SI7021 temperature and humidity sensor. Flashed Tasmota 5.12 and was happy that sensor is immediately detected and measurements are displayed on webpage and published to MQTT as well.
Then I moved my Sonoff to the barn, where temperature is around 0C these days. After powering up there were no temperature/humidity readings anymore. I did some "investigation" and in syslog I could see a debug message "Timeout waiting for start signal high pulse":
```
Feb 22 16:21:51 192.168.11.107 sonoff-th-3044 ESP-WIF: Checking connection...
Feb 22 16:21:51 192.168.11.107 sonoff-th-3044 ESP-WIF: Connected
Feb 22 16:21:57 192.168.11.107 sonoff-th-3044 ESP-HTP: Main Menu
Feb 22 16:21:57 192.168.11.107 sonoff-th-3044 ESP-DHT: Timeout waiting for start signal high pulse
Feb 22 16:22:00 192.168.11.107 sonoff-th-3044 ESP-DHT: Timeout waiting for start signal high pulse
Feb 22 16:22:02 192.168.11.107 sonoff-th-3044 ESP-DHT: Timeout waiting for start signal high pulse
Feb 22 16:22:03 192.168.11.107 sonoff-th-3044 ESP-HTP: Configuration
```
I remember that when I tried Sonoff's original firmware I also faced some problems with temperature around 0C, but it was fine again when temperature fell below -1.

After that I checked the same sonoff with the same sensor in the house with normal room temperature and it worked, sensor was detected immediately. I've only not yet checked if it is detected in temperatures below -1.

reloxx13, 2018-02-23:
check the connection, sometimes its hard to plug it in completly.

if you removed it before the barn and plug it in again, could be it was not plugged in correctly.

mbolsteins, 2018-02-23:
Seems that you're right, now it works - at -0.6C.

closed 2018-02-23 by mbolsteins