arendst/Tasmota #2002: Less information in Console for RF Bridge

DerDom2010, 2018-02-23:
Hello

I have a question. Is it possible to change this line in **xdrv_04_snfbridge**

`snprintf_P(mqtt_data, sizeof(mqtt_data), PSTR("{\"" D_JSON_RFRECEIVED "\":{\"" D_JSON_SYNC "\":%d,\"" D_JSON_LOW "\":%d,\"" D_JSON_HIGH "\":%d,\"" D_JSON_DATA "\":\"%06X\",\"" D_CMND_RFKEY "\":%s}}"),`

from this Information

`19:52:29 MQT: tele/sonoff-bridge/RESULT = {"RfReceived":{"Sync":12570,"Low":420,"High":1220,"Data":"D2FD2E","RfKey":1}}`

to

`19:52:29 MQT: tele/sonoff-bridge/RESULT = {"RfReceived":Data":"D2FD2E","RfKey":1}}`

Best wishes and thank you
`

reloxx13, 2018-02-23:
reason?

DerDom2010, 2018-02-23:
When I get the Information from my PIR Sensor. The first values are always changing. The last ones are static.

With the changing values I am not able to create a working trigger via MQTT in my Io Broker.

This is just a personal issue for me. I am very bad at coding. Until now I always failed in my tries.

Sorry for asking.

curzon01, 2018-02-23:
Tasmota telegrams are in JSON format.

You have to extract the data you are interested in from the JSON string - then it doesn't matter which additonal values from JSON result are always changing.

reloxx13, 2018-02-23:
yup, and removing data output is always a bad idea, since other projects could already use it.

you should try to programm it correctly instead of avoiding it or looking for bad workarounds.

DerDom2010, 2018-02-23:
Thanks for the hint.

Working now

best wishes

ErikAndren, 2018-02-24:
Close?

closed 2018-06-02 by ascillato