arendst/Tasmota #2004: No DHT22 sensor selectable

dYna77, 2018-02-23:
Hi,

i am on 5.12.0c and use the "Generic Module" on a abre ESP8266.
On GPIO0 i have a DHT22 sensor, but i cant select it in the configuration menu.
Is this not available with the module i use?

Thanks!

curzon01, 2018-02-23:
Use "03 SI7021", that's the DHT22 sensor too.

dYna77, 2018-02-23:
Thank you.
I did not find a mention of this at https://github.com/arendst/Sonoff-Tasmota/wiki/Sensor-Configuration

curzon01, 2018-02-24:
There is: see subsection [Single wire sensor](https://github.com/arendst/Sonoff-Tasmota/wiki/Sensor-Configuration#single-wire-sensor) - DHT22 (3), AM2302 (3), AM2321 (3), SI7021 (3) are all the same sensor (=3) type.

**Hint**: Use also github _Search_ function
[image]

jhlweb, 2018-03-01:
@dYna77 if this is solved please close this issue

closed 2018-03-01 by dYna77

jamman9000, 2019-02-09:
Issue is closed, but answer should be corrected to say he/we should use the AM2301 selection when using the DHT22, if you go look at the issue referenced with the quote, the longer explanation helps better clarify which should be used for what. Apparently, the AM2301 uses the same handler as DHT22 so they "consolidated" them into one selection. Then took the old DHT22 selection and changed it to the SI7021.