arendst/Tasmota #2008: Flash Tasmota on self-locking Mode

tuga49, 2018-02-24:
Hi. First off all, I have no experience. Im new to this. I just flashed tasmota on two basic module and one s20. Now I want to introduce my garage door with HomeAssistant and my own server.
The problem is: I have this board: ESP8285 based 1Channel Inching/Locking, to control my garage door, trought eWelink app. But I want flash tasmota with this but I seen it is very hard to do that on this module because I need to do a several things. My qyestion is if any there any board that I can flash it was a normal one, with COM and NO port, and after flash, I have the function of self locking. Basiclly I want to integrate a board on home assistant, and connect it to my garage door, and press the button to turn on relay and it turns off automaticlly after a second or two. Thanks. Here is the board I have right now working with eWelink.

[image]
[image]

reloxx13, 2018-02-24:
http://forum.creationx.de/forum/index.php?thread/351-sonoff-1-channel-inching-self-locking/

tuga49, 2018-02-24:
reloxx13, I dont understand this guide. Im very new to this. How can I do that to get this sonoff working with tasmota?

asantos15, 2018-02-24:
Hello @tuga49 ,

I just flashed tasmota and it ran kind of smoothly. Not that difficult, indeed. However, there is a limitation/issue that I am trying to tackle.
Once it's flashed and working, whenever it restarts or reconnects to the WIFI network, it quickly turns on and off at startup and that would be enoug to open the garage door unintentionally.
This seems to be an issue with the firmware and how the relay starts...
As to the flashing itself, these tuts are the best there are so far. You should read and re-read them a couple of times to absorb different actions/aspects of this flashing process. You can do it!

tuga49, 2018-02-24:
@asantos15  you speak portuguese? Im asking that because of your name, Can you tell me the steps to flash tasmota? because its diferrent from flashing on sonoff basic... On sonoff basic I connect the pins from flashing pen to sonoff, then I press button and connect pen to pc. it detects and I flash it. On this self locking board its different I think. Because I cant do that by the same way I did before on sonoff basic

asantos15, 2018-02-25:
I'm Brazilian, yes!

Here are the steps:
Have your Arduino or FDTI ready to flash, following the steps found in this link:
https://github.com/arendst/Sonoff-Tasmota/wiki/Arduino-IDE

Then you can have the relay powered by a cell charger;
Press both buttons until the lights start blinking;
Press the outter button to have it stop blinking. That should put it in Flash Mode;
Hold TX and RX to the pins as shown on the picture;
Click Compile and Upload on Arduino IDE.

If your code is ok and configurations are right, you'll see its progress as it uploads Tasmota firmware.

Here is my setup:
[image]

tuga49, 2018-02-25:
@asantos15 qual é a posição do rx, tx, 3v3 e gnd? Preciso de soldar em algum sítio ou basta ligar nos  slots disponíveis? Pode mandar uma imagem com a conexão correta?

asantos15, 2018-02-25:
@tuga49 , veja que não é necessário soldar. 
Conecte o RX do Arduino ao quinto pinna linha mostrada na foto (fio verde). E TX do arduino conecte ao sexto. Isso, depois de pôr em modo de flash.
Note o fio azul, que está no GND do Arduino, mas confesso que não tenho ctz se tem algum efeito prático. 
The idea to have this discussion in English is because it gets easier for others to read and try. De onde é?

asantos15, 2018-02-25:
Um detalhe importante, quando for colocar a placa em modo de flash: Segurar os dois botões enquanto pluga energia. Mantenha-os apertados por uns 5 segundos. Depois, aperta uma vez o botão "de fora". Daí, entrará em flashing mode.

tuga49, 2018-02-25:
@asantos15, sou de Portugal. Falar português so para me facilitar um pouco :) o rx e o tx conecto onde. N estou a perceber. Pode fazer um desenho de esquema simples usando uma foto da board? Ja tentei e nao da. Devo estar a ligar mal algo. Obrigado

Jason2866, 2018-02-26:
The Arduino is 5 Volt. ESP use 3.3 Volt!  Dont do that.

asantos15, 2018-02-26:
Hello @Jason2866 ,

Did it a couple of times without any problem.
I think that once it gets fed by an arduino, it gets converted before it gets to the WiFi board, or it wouldnt have the alternative...

tuga49, 2018-02-26:
@Jason2866 I hqve that pen. How I make the connection?

Jason2866, 2018-02-26:
What type is it? What is written on the metal part of the ESP Chip?
Google for that ESP chip type pinout. You have to connect the pins 
gnd tx rx  GPIO 0 to gnd and 3.3v  Then plug in the USB seriell converter and start flash process
Here is a picture http://forum.creationx.de/index.php?attachment/488-angel%C3%B6tete-pinne-bild-2-forum-jpg/
and here
http://forum.creationx.de/index.php?attachment/489-pin-belegung-1-chanel-bild-1-forum-jpg/

tuga49, 2018-02-26:
Thats working. Thanks so much!

closed 2018-02-26 by tuga49