arendst/Tasmota #2009: Timer issue / garage door option

djtonyd, 2018-02-24:
hi all is there a way to set a timer on the sonoff just inside the software/web interface so when you press on  it turns the relay on for a split sec then off auto on the interface as im going to connect the load side to a contactor then use the NO to allow my garage to open as it needs a pulse function and i dont want constant 12v on it nor 240v on the sonoff coming out.

Oxyandy, 2018-02-24:
search google sonoff pulsetime

djtonyd, 2018-02-24:
i dont want to use the GPIO functions just the web on/off in order to make it pulse will this still work?

Oxyandy, 2018-02-24:
pulsetime X will mean anytime relay is activated, will stay on for x
whichever way you turn it on

djtonyd, 2018-02-24:
ok so i would type in the console pulsetime <0.1> ?

Oxyandy, 2018-02-24:
Yeah maybe, I have never gone below 1 before.. so not sure about the point should be in wiki

Oxyandy, 2018-02-24:
pulsetime 1
as console (web console window) will mean 1 second
right syntax

djtonyd, 2018-02-24:
just entered it works like a charm :) happy days thanks heaps

closed 2018-02-24 by arendst