arendst/Tasmota #2010: Feature Request: Quinled support

michel72, 2018-02-24:
Quinled by [Andries Faassen](http://blog.quindorian.org/2017/02/esp8266-led-lighting-quinled-v2-6-pcb.html/) is a PWM 2 channel LED board similar to the H801.

When using Tasmota as a generic device with GPIO00 set to PWM1 and GPIO2 set to PWM2 it works.
Better support for this great board would be appreciated as I can now only use it as a standard PWM light and that doesn’t offer all features that the H801 offers.

I have no programming knowledge whatsoever therefor changing the H801 template to a 2 channel module that uses GPIO00 en GPIO02 is a bit to steep for me ☹

arendst, 2018-02-24:
What happens after executing command ``SetoOtion 15 1``?

michel72, 2018-02-24:
I'm not really sure why you are asking :)

After SetOption15 1 the lights flash 2 times and the module seems to reboot.

17:34:52 CMD: SetOption15 1
17:34:52 MQT: stat/quinled-01/RESULT = {"SetOption15":"ON"}

shortly thereafter:

00:00:00 Project sonoff quinled-01 (Topic quinled-01, Fallback DVES_769CE7, GroupTopic sonoffs) Version 5.12.0
00:00:00 WIF: Connecting to AP1 lhmw in mode 11N as quinled-01-7399...
00:00:03 WIF: Connected
00:00:03 DNS: Initialized
00:00:03 HTP: Web server active on quinled-01-7399.local with IP address 192.168.2.243
00:00:06 MQT: Attempting connection...
00:00:06 MQT: Connected
00:00:06 MQT: tele/quinled-01/LWT = Online (retained)
00:00:06 MQT: cmnd/quinled-01/POWER = 
00:00:06 MQT: stat/quinled-01/RESULT = {"Command":"Unknown"}
00:00:06 MQT: tele/quinled-01/INFO1 = {"Module":"Generic","Version":"5.12.0","FallbackTopic":"DVES_769CE7","GroupTopic":"sonoffs"}
00:00:06 MQT: tele/quinled-01/INFO2 = {"WebServerMode":"Admin","Hostname":"quinled-01-7399","IPAddress":"192.168.2.243"}
00:00:06 MQT: tele/quinled-01/INFO3 = {"RestartReason":"Software/System restart"}
00:00:06 MQT: homeassistant/light/quinled-01/config =  (retained)
00:00:06 MQT: stat/quinled-01/RESULT = {"Command":"Unknown"}
17:35:01 MQT: stat/quinled-01/RESULT = {"POWER":"OFF"}
17:35:01 MQT: stat/quinled-01/POWER = OFF
17:35:08 MQT: tele/quinled-01/STATE = {"Time":"2018-02-24T17:35:08","Uptime":"0T00:00:14","Vcc":3.815,"POWER":"OFF","Dimmer":18,"Color":"2D2D","CT":500,"Fade":"OFF","Speed":1,"LedTable":"OFF","Wifi":{"AP":1,"SSId":"lhmw","RSSI":100,"APMac":"80:2A:A8:C7:9C:EA"}}

michel72, 2018-02-24:
Ah I see now.... ;)

I have a toggle on/off switch now and 2 sliders that indeed control the lights 👍

michel72, 2018-02-24:
Maybe not exactly what I was looking for. 

The one slider controls color temperature, the other brightness. I was looking for 2 individual sliders (and if possible 2 individual on/off switches) to control both channels separately.

rqleslie, 2018-06-09:
hi @michel72 can you give me more info on this, I have multiple Quinleds and desperately looking to use them with a dimmer / slider in home assistant. currently, I  set the brightness via scripts that is not ideal.

Did you flash the esp01 for the quinled board with tasmota firmware? and did you manage to get the sliders working for dimming the lights ?

Any help would be appreciated

michel72, 2018-06-10:
Hi. Flashing tasmota software on your ESP will enable you to control ON/OFF for both channels at the same time and control the brightness for both channels at the same time. The "color temperature" slider is nothing more then a slider between both channels. Sliding it from right to left, will bright up channel 1, while dimming channel 2.

I was looking for 2 single switches and 2 single dimming sliders since I would like to control both channels completely separately, but I have had no luck achieving just that.

Anyway, If you like the above, Flash tasmota, set te module type to "generic", GPIO0 to PWM1 and GPIO2 to PWM2.

If anyone has the answer as to how to set the tasmota to turn on/off and dim 2 seperate channels, please elaborate :)

closed 2018-08-09 by stale[bot]