arendst/Tasmota #2011: Reed and wall switch on one sonoff basic?

alvsky, 2018-02-24:
Hi all,
Now I have a following scenario:
I connected a reed sensor to GPIO14 of Sonoff1 which tells me if door is open or not (gave it switchtopic 
 "alarm" and disconnected it from the switch).
I connected a wall switch on GPIO14 of Sonoff2 to control the light. (switchtopic = sonoff1, switchmode = 0)
In Home Assistant I set that when I open the door a light (that is connected to Sonoff2) switches on.
Then I can turn it off with wall switch
When Home assistant is offline from some reason, I still can turn the light on and off.

Now... I would like to do it all with only one Sonoff. I tried connecting reed sensor to GPIO3 (with pull-up resistor) but I couldn't get it to work exactly as I want. The MUST is that I can always switch the light (otherwise I have well known wife problems ;) )

I have read lots of posts and lost a lot of time but I just could't figure it out so I need help.
And of course, it would be good to mention that I am new to all home automation/mqtt/... stuff :)

Thanks a lot

alvsky, 2018-02-25:
OK... I finally manage to do this so if someone else will need id here is how I have done it:
I connected reed switch to GPIO3 and wall switch to GPIO14.
This is Sonoff config:

[image]

In console I set it like this:

CMD: switchtopic
MQT: stat/sonoff1/RESULT = {"SwitchTopic":"sonoff1"}
CMD: switchmode1
MQT: stat/sonoff1/RESULT = {"SwitchMode1":0}
CMD: switchmode2
MQT: stat/sonoff1/RESULT = {"SwitchMode2":1}`

Now, when I switch on/off the light I get following messages in console (I honestly don't know why I see both POWER and POWER1 so if someone could explain it to me I would be grateful):

11:58:19 MQT: cmnd/sonoff1/POWER1 = ON
11:58:19 MQT: stat/sonoff1/RESULT = {"POWER":"ON"}
11:58:19 MQT: stat/sonoff1/POWER = ON
11:58:22 MQT: cmnd/sonoff1/POWER1 = OFF
11:58:22 MQT: stat/sonoff1/RESULT = {"POWER":"OFF"}
11:58:22 MQT: stat/sonoff1/POWER = OFF

When I open/close the door I get these messages:

12:02:34 MQT: cmnd/sonoff1/POWER2 = ON
12:02:34 MQT: stat/sonoff1/RESULT = {"Command":"Unknown"}
12:02:37 MQT: cmnd/sonoff1/POWER2 = OFF
12:02:37 MQT: stat/sonoff1/RESULT = {"Command":"Unknown"}

So, after some playing with combinations in Home Assistant I defined 2 switches in configuration.yaml like this:

`switch:
  - platform: mqtt
    name: "Sonoff1"
    command_topic: "cmnd/sonoff1/power"
    state_topic: "stat/sonoff1/POWER"
    qos: 1
    payload_on: "ON"
    payload_off: "OFF"
    retain: true

  - platform: mqtt
    name: "Door_Sensor"  
    command_topic: "cmnd/sonoff1/POWER2"
    state_topic: "cmnd/sonoff1/POWER2"
    qos: 1
    payload_on: "ON"
    payload_off: "OFF"
    retain: true
    optomistic: true`

and in automation.yaml:

`- action:
    service: switch.turn_on
    entity_id: 
    - switch.sonoff1
  alias: Ulazna vrata
  trigger:
  - entity_id: switch.door_sensor 
    from: 'off'
    platform: state
    to: 'on'`

Now when door is open the light goes on and I can switch it off with wall switch

I hope this will help some home automation noob like me so she/he does not loose too much time on basics :)

Enjoy

badgerhome, 2018-04-28:
@alvsky Thanks for this. Are you controlling the relay locally on the Sonoff with the wall switch on GPIO14?

I'm trying to setup and external switch that triggers the relay in the default way (would always control relay regardless of network connectivity) and a reed switch that ONLY reports to MQTT (does not change relay)

alvsky, 2018-04-29:
> Are you controlling the relay locally on the Sonoff with the wall switch on GPIO14?

Yes, I do

I have exact functionality that you described: External switch connected to GPIO14 for controlling relay and reed switch which reports to MQTT :) and it works without any issues

badgerhome, 2018-04-29:
Thanks @alvsky I will have to try this again. So you didn't change the switchtopic? You left default switchtopic0? 

Are the both connected to the same GND pin? 

Thanks again!

closed 2018-06-28 by stale[bot]