arendst/Tasmota #2016: Make sonoff digitally dim based on temp

andyd999, 2018-02-25:
I have a specific application and thinking one of you clever chaps may have a  solution. 
I have a few vivariums and wish to digitally dim the basking lamp based on temperature. So for example when the temperature reaches 40C the lamp attached to the sonoff dims to 75% brightness then when the temp drops the lamp brightena again. 

You can buy a commercial vivarium digital dimming thermostat but they are expensively and I was hoping to achieve it somehow using a sonoff and tasmota 

Any ideas ?

Frogmore42, 2018-02-25:
You can do that using NodeRed.  You would need an MQTT input node (listening to whatever is measuring the temperature), connect that to a function node to do the control you want and then have it output a message to an MQTT output node to control the lamp.  But you would need a module that supports dimming.  Most Sonoffs have a relay which would give anything other that full on or off.

andyd999, 2018-02-25:
I thought the sonoff relay was either on or off with no dimming? Any idea idea of a module that would support dimming? I guess like a th10 could be used to measure temp then just need to find a suitable dimming device that the mqqt link could speak to?

Frogmore42, 2018-02-25:
You are correct, there is a "not" missing in the last sentence.  I am not aware of anything from Itead that has dimming capability for high voltage.  You could use a Sonoff that has dual outputs and attach one directly and the other through a regular dimmer switch.  Or you could make something that uses the PWM output of a standard ESP and controls something like this: https://www.amazon.com/RobotDyn-controller-control-Arduino-Raspberry/dp/B072K9P7KH/

andyd999, 2018-02-25:
Thanks but why connect in every direct and the other through the dimmer? And with that module it would need an 8266 controller to run it I couldn't just attach to the sonoff? Sorry my electronics isn't very advanced

Frogmore42, 2018-02-26:
What you want to do has no easy solution today.  If your electronics knowledge is not advanced enough, there is a much higher possibility of electrocution. 

The first option was to try and make it easy.  When you turn one switch on, it connects the lamp through an external dimmer.  When you turn the first switch off and the second switch on, it connects the lamp directly, so full brightness.

The second option was to make what is not currently available.

There are zwave and zigbee dimmers available.  So using Smartthings to control them might be better for your needs.

andyd999, 2018-02-26:
Thanks I'll look for a zwave dimmer and explore that option