arendst/Tasmota #2018: Sonoff 4CH Pro (R2) Supported?

FamJohansson, 2018-02-25:
Hi,
Thank you for a wonderful job you've done with Sonoff Tasmota.

I'm trying to flash my Sonoff 4CH Pro R2 with the firmware but I do not get it uploaded. A little bit uncertain of what I'm doing wrong. 

I have no trouble flashing the Sonoff 4CH Pro. (Current version 5.12.0c)

I'm unsure of the following:
-Supports sonoff Tasmota Sonoff 4CH Pro R2?
-How do I put Sonoff 4CH Pro R2 in flash mode?

Jason2866, 2018-02-25:
Tried https://github.com/arendst/Sonoff-Tasmota/wiki/Sonoff-4CH

FamJohansson, 2018-02-25:
There are two versions of 4CH pro I have R2 version ..
see pictures
[image]
[image]

nmaggioni, 2018-03-07:
On the R2 model the GPIO0 pad is broken out to a small pad to the right of the ESP package, as per [my other comment](https://github.com/arendst/Sonoff-Tasmota/issues/2021#issuecomment-371227748):

[image]

FamJohansson, 2018-03-07:
@nmaggioni
Thanks this solved my question..

nmaggioni, 2018-03-07:
@FamJohansson You're welcome, consider closing this issue if it's fixed.

FamJohansson, 2018-03-07:
Hello again @nmaggioni 
Do yo have any clue how to flash this one?      
 
https://www.aliexpress.com/item/SONOFF-AC85-250V-4-Channel-Automation-Module-Jog-Inching-WIFI-Wireless-Smart-Switch-Relay-rf-receive/32835664788.html?spm=a2g0s.9042311.0.0.fbFZOL
[image]

sigalou, 2018-03-11:
For Flash, I did a tutorial on my site (in French). Good continuation.
http://sigalou-domotique.fr/index.php/17-sonoff-4ch-pro-r2

FamJohansson, 2018-03-11:
@sigalou
Have you tested this one?

https://www.aliexpress.com/item/32835664788/32835664788.html

Can’t see how to flash it,

Jason2866, 2018-03-11:
Looks interesting, but i am pretty sure it is not a genuine sonoff.
ITEAD had never used an extra power supply board (this looks like a cheap one)
and the integrate the ESP on the main PCB on actual board revisons.
Furthermore this board istnt listed on the website from ITEAD

ascillato, 2018-06-02:
Hi @FamJohansson 

If your issue is solved, please close it. Thanks! :+1:

closed 2018-06-02 by FamJohansson

KierkelsD, 2018-08-28:
@FamJohansson , could you tell us how you have resolved your problem?

Thanks

Frigorio, 2019-04-19:
> Hello again @nmaggioni
> Do yo have any clue how to flash this one?
> 
> https://www.aliexpress.com/item/SONOFF-AC85-250V-4-Channel-Automation-Module-Jog-Inching-WIFI-Wireless-Smart-Switch-Relay-rf-receive/32835664788.html?spm=a2g0s.9042311.0.0.fbFZOL
> [image]

https://www.letscontrolit.com/wiki/index.php/Sonoff_DIY_4ch

itajackass, 2019-04-27:
Hi i'm little bit OT but i've found some pic here... Is it possible to install an external antenna on R2 version? In normal version was a pad where we can solder an ipex connector. But in R2 i don't see anything. Can someone help me?