arendst/Tasmota #2019: unstable connection Witty Cloud Module ESP8266

Igor122017, 2018-02-25:
Hi, 
uploaded a new version 5.12 all ok, everything works. The problem is that there is a connection every minute or more. What could be the problem???
00:00:00 Project sonoff ESP8266-2 (Topic ESP02, Fallback MD client08, GroupTopic ESP8266) Version 5.12.0c-2_4_0
00:00:00 WIF: Connecting to AP1 MGTS_GPON_3233 in mode 11N as ESP02-0138...
00:00:08 WIF: Connected
00:00:08 DNS: Initialized
00:00:08 HTP: Web server active on ESP02-0138.local with IP address 192.168.1.78
00:00:10 MQT: Attempting connection...
00:00:11 MQT: Connected
00:00:11 MQT: tele/ESP02/LWT = Online (retained)
00:00:11 MQT: cmnd/ESP02/POWER = 
00:00:11 MQT: tele/ESP02/INFO1 = {"Module":"Witty Cloud","Version":"5.12.0c","FallbackTopic":"MD client08","GroupTopic":"ESP8266"}
00:00:11 MQT: tele/ESP02/INFO2 = {"WebServerMode":"Admin","Hostname":"ESP02-0138","IPAddress":"192.168.1.78"}
00:00:11 MQT: tele/ESP02/INFO3 = {"RestartReason":"Software/System restart"}
18:49:22 MQT: tele/ESP02/STATE = {"Time":"2018-02-25T18:49:22","Uptime":"0T00:00:19","Wifi":{"AP":1,"SSId":"MGTS_GPON_3233","RSSI":54,"APMac":"78:81:02:C7:73:C2"}}
18:49:22 MQT: tele/ESP02/SENSOR = {"Time":"2018-02-25T18:49:22","Switch1":"0","Analog0":94}
18:49:52 MQT: Attempting connection...
18:49:52 MQT: Connected
18:49:52 MQT: tele/ESP02/LWT = Online (retained)
18:49:52 MQT: cmnd/ESP02/POWER = 
18:50:30 MQT: Attempting connection...
18:50:30 MQT: Connected
18:50:30 MQT: tele/ESP02/LWT = Online (retained)
18:50:30 MQT: cmnd/ESP02/POWER = 
18:50:45 MQT: stat/ESP02/RESULT = {"POWER":"1"}
18:50:45 MQT: stat/ESP02/POWER = 1
18:50:53 MQT: Attempting connection...
18:50:53 MQT: Connected
18:50:53 MQT: tele/ESP02/LWT = Online (retained)
18:50:53 MQT: cmnd/ESP02/POWER = 
18:51:29 MQT: stat/ESP02/RESULT = {"POWER":"0"}
18:51:29 MQT: stat/ESP02/POWER = 0
18:52:16 MQT: Attempting connection...
18:52:16 MQT: Connected
18:52:16 MQT: tele/ESP02/LWT = Online (retained)
18:52:16 MQT: cmnd/ESP02/POWER = 
18:53:09 MQT: Attempting connection...
18:53:09 MQT: Connected
18:53:09 MQT: tele/ESP02/LWT = Online (retained)
18:53:09 MQT: cmnd/ESP02/POWER = 
18:53:32 MQT: Attempting connection...
18:53:32 MQT: Connected
18:53:32 MQT: tele/ESP02/LWT = Online (retained)
18:53:32 MQT: cmnd/ESP02/POWER = 
18:54:22 MQT: tele/ESP02/STATE = {"Time":"2018-02-25T18:54:22","Uptime":"0T00:05:19","Wifi":{"AP":1,"SSId":"MGTS_GPON_3233","RSSI":62,"APMac":"78:81:02:C7:73:C2"}}
18:54:22 MQT: tele/ESP02/SENSOR = {"Time":"2018-02-25T18:54:22","Switch1":"0","Analog0":74}

used witty cloud vodule...

scottdalton, 2018-02-25:
FWIW - I picked up a few witty cloud modules a long while back and found them completely unusable due to wifi connection issues.  Compared to a Wemos D1 Mini or generic NodeMCU v2 they had extremely poor signal strengths.  Even sitting a few feet away from an access point, they regularly wouldn't be able to connect to wifi on boot or would lose connection.  Perhaps I had a bad batch, but I gave up on them.

Igor122017, 2018-02-26:
thanks for the answer, the fact  I used this module with firmware 5.10 and there was no such problem .... I thought that firmware 5.12 was better and did it eventually faced this problem ...

Jason2866, 2018-02-26:
Maybe you try 5.12.0c with older Arduino SDK. You use 2.4.0. 
I had problems with these too. With version 2.3.0 (1.5.3) my problems dissapeared

Igor122017, 2018-02-26:
Many thanks for the answer, tell me how to check SDK Arduino? I will definitely check and write what has changed ....

Igor122017, 2018-02-26:
I'm flashing ESP 8266 through - the platformio, atom

Igor122017, 2018-02-26:
my model ESP12-F...

Igor122017, 2018-02-26:
as I understand, the problem is not WI-FI the connection, but the connection with MQTT....
00:00:00 Project sonoff ESP8266-2 (Topic ESP02, Fallback MD client008, GroupTopic ESP8266) Version 5.12.0c-2_4_0
00:00:00 WIF: Connecting to AP1 MGTS_GPON_3233 in mode 11N as ESP02...
00:00:05 WIF: Connected
00:00:05 DNS: Initialized
00:00:05 HTP: Web server active on ESP02.local with IP address 192.168.1.78
**00:00:07 MQT: Attempting connection...
00:00:08 MQT: Connected**
00:00:08 MQT: tele/ESP02/LWT = Online (retained)
00:00:08 MQT: cmnd/ESP02/POWER = 
00:00:08 MQT: tele/ESP02/INFO1 = {"Module":"Sonoff Basic","Version":"5.12.0c","FallbackTopic":"MD client008","GroupTopic":"ESP8266"}
00:00:08 MQT: tele/ESP02/INFO2 = {"WebServerMode":"Admin","Hostname":"ESP02","IPAddress":"192.168.1.78"}
00:00:08 MQT: tele/ESP02/INFO3 = {"RestartReason":"Software/System restart"}
00:00:08 MQT: homeassistant/light/ESP02/config =  (retained)
00:00:08 MQT: homeassistant/switch/ESP02/config =  (retained)
00:00:09 MQT: stat/ESP02/RESULT = {"POWER":"0"}
00:00:09 MQT: stat/ESP02/POWER = 0
00:00:16 MQT: tele/ESP02/STATE = {"Time":"1970-01-01T00:00:16","Uptime":"0T00:00:16","POWER":"0","Wifi":{"AP":1,"SSId":"MGTS_GPON_3233","RSSI":58,"APMac":"78:81:02:C7:73:C2"}}
00:00:16 MQT: tele/ESP02/SENSOR = {"Time":"1970-01-01T00:00:16","Switch1":"0"}
**17:46:29 MQT: Attempting connection...
17:46:30 MQT: Connected**
17:46:30 MQT: tele/ESP02/LWT = Online (retained)
17:46:30 MQT: cmnd/ESP02/POWER =

Jason2866, 2018-02-26:
in platformio.ini 
you have delete the ; 
if the entry is not there copy these line from here.

[env:sonoff]
**platform = espressif8266@1.5.0  ; v2.3.0**
;platform = espressif8266@1.6.0  ; v2.4.0
platform = espressif8266
framework = arduino
board = esp01_1m
board_flash_mode = dout
build_flags = -Wl,-Tesp8266.flash.1m0.ld -DMQTT_MAX_PACKET_SIZE=1000
lib_deps = PubSubClient, NeoPixelBus, IRremoteESP8266, ArduinoJSON
extra_scripts = pio/strip-floats.py

Igor122017, 2018-02-26:
now I have in platformio.ini
env:sonoff]
platform = espressif8266
framework = arduino
board = esp01_1m
board_flash_mode = dout
build_flags = -Wl,-Tesp8266.flash.1m0.ld
lib_deps = PubSubClient, NeoPixelBus, IRremoteESP8266, ArduinoJSON
extra_scripts = pio/strip-floats.py

; *** Serial Monitor options
monitor_baud = 115200
Espressif 8266 now 1.6 
1.6.0 released about a month ago

Finally! Release of Arduino Core v2.4.0
Add new boards: WeMos D1 mini Pro and 4D Systems gen4 IoD Range
Update espota.py to v1.1.0
Update mkspiffs tool to v0.2.0
Allow overriding PROGNAME (issue #54)
Improve support of FQDN for OTA target (issue #56)

Igor122017, 2018-02-26:
MQTT Client & Fallback Topic | MD  client008 - why Fallback Topic - this is right or wrong?
-- | --

Igor122017, 2018-02-26:
i am do in platformio.ini
[env:sonoff]
platform = espressif8266@1.5.0 ; v2.3.0
;platform = espressif8266@1.6.0 ; v2.4.0
platform = espressif8266
framework = arduino
board = esp01_1m
board_flash_mode = dout
build_flags = -Wl,-Tesp8266.flash.1m0.ld -DMQTT_MAX_PACKET_SIZE=1000
lib_deps = PubSubClient, NeoPixelBus, IRremoteESP8266, ArduinoJSON
extra_scripts = pio/strip-floats.py

here results:
00:00:00 Project sonoff ESP8266-2 (Topic ESP02, Fallback MD client08, GroupTopic ESP8266) Version 5.12.0c-2_4_0
00:00:00 WIF: Connecting to AP1 MGTS_GPON_3233 in mode 11N as ESP02...
00:00:06 WIF: Connected
00:00:06 DNS: Initialized
00:00:06 HTP: Web server active on ESP02.local with IP address 192.168.1.78
00:00:09 MQT: Attempting connection...
00:00:09 MQT: Connected
00:00:09 MQT: tele/ESP02/LWT = Online (retained)
00:00:09 MQT: cmnd/ESP02/POWER = 
00:00:09 MQT: tele/ESP02/INFO1 = {"Module":"Sonoff Basic","Version":"5.12.0c","FallbackTopic":"MD client08","GroupTopic":"ESP8266"}
00:00:09 MQT: tele/ESP02/INFO2 = {"WebServerMode":"Admin","Hostname":"ESP02","IPAddress":"192.168.1.78"}
00:00:09 MQT: tele/ESP02/INFO3 = {"RestartReason":"Software/System restart"}
00:00:09 MQT: homeassistant/light/ESP02/config =  (retained)
00:00:09 MQT: homeassistant/switch/ESP02/config =  (retained)
00:00:10 MQT: stat/ESP02/RESULT = {"POWER":"0"}
00:00:10 MQT: stat/ESP02/POWER = 0
19:22:33 MQT: tele/ESP02/STATE = {"Time":"2018-02-26T19:22:33","Uptime":"0T00:00:17","POWER":"0","Wifi":{"AP":1,"SSId":"MGTS_GPON_3233","RSSI":46,"APMac":"78:81:02:C7:73:C2"}}
19:22:33 MQT: tele/ESP02/SENSOR = {"Time":"2018-02-26T19:22:33","Switch1":"0"}
19:22:50 MQT: Attempting connection...
19:22:50 MQT: Connected
19:22:50 MQT: tele/ESP02/LWT = Online (retained)
19:22:50 MQT: cmnd/ESP02/POWER = 
19:23:06 MQT: stat/ESP02/RESULT = {"POWER":"1"}
19:23:06 MQT: stat/ESP02/POWER = 1
19:24:28 MQT: stat/ESP02/RESULT = {"POWER":"0"}
19:24:28 MQT: stat/ESP02/POWER = 0
19:24:35 MQT: stat/ESP02/RESULT = {"POWER":"1"}
19:24:35 MQT: stat/ESP02/POWER = 1
19:25:13 MQT: Attempting connection...
19:25:13 MQT: Connected
19:25:13 MQT: tele/ESP02/LWT = Online (retained)
19:25:13 MQT: cmnd/ESP02/POWER = 
19:25:18 MQT: stat/ESP02/RESULT = {"POWER":"0"}
19:25:18 MQT: stat/ESP02/POWER = 0
19:26:21 MQT: Attempting connection...
19:26:21 MQT: Connected
19:26:21 MQT: tele/ESP02/LWT = Online (retained)
19:26:21 MQT: cmnd/ESP02/POWER = 
19:27:01 MQT: Attempting connection...
19:27:01 MQT: Connected
19:27:01 MQT: tele/ESP02/LWT = Online (retained)
19:27:01 MQT: cmnd/ESP02/POWER = 
19:27:33 MQT: tele/ESP02/STATE = {"Time":"2018-02-26T19:27:33","Uptime":"0T00:05:17","POWER":"0","Wifi":{"AP":1,"SSId":"MGTS_GPON_3233","RSSI":50,"APMac":"78:81:02:C7:73:C2"}}
19:27:33 MQT: tele/ESP02/SENSOR = {"Time":"2018-02-26T19:27:33","Switch1":"0"}

Jason2866, 2018-02-26:
It seems that you are still on Version 1.6 (2.4.0) this a new version with new features and fixes..
But this version seems to have problems with wifi. Maybe you cannot install the old version 1.5 (2.3.0) through Atom.io. I had this on my PC before i installed Atom.io. Atom.io installed  the version 1.6 and asked me if i want to keep 1.5. I did so. So i dont know how you get the old one, sorry.
The message fallback topic is no fault

Igor122017, 2018-02-26:
thank you very much for any help. 
I am return versia 1.5 like you wriht...  this is in the section PlatfotmiO Home i am change with 1.6 to 1.5...

Platform has been successfully installed
PlatformManager: Installing espressif8266 @ 1.5.0
espressif8266 @ 1.5.0 is already installed
PackageManager: Installing toolchain-xtensa @ ~1.40802.0
toolchain-xtensa @ 1.40802.0 is already installed
PackageManager: Installing tool-esptool @ ~1.409.0
tool-esptool @ 1.409.0 is already installed
The platform 'espressif8266@1.5.0' has been successfully installed!
The rest of packages will be installed automatically depending on your build environment.

Igor122017, 2018-02-26:
I probably need to look at the platfirnio used version, but I do not know how to do it

Igor122017, 2018-02-26:
Dear Jason2866< thanks you very, very, very much...  i am delete 1.6.... 

now all OK... thanks you...

00:00:00 Project sonoff ESP8266-2 (Topic ESP02, Fallback MD client08, GroupTopic ESP8266) Version **5.12.0c-2_3_0**
00:00:00 WIF: Connecting to AP1 MGTS_GPON_3233 in mode 11N as ESP02...
00:00:06 WIF: Connected
00:00:06 DNS: Initialized
00:00:06 HTP: Web server active on ESP02.local with IP address 192.168.1.78
23:16:12 MQT: Attempting connection...
23:16:13 MQT: Connected
23:16:13 MQT: tele/ESP02/LWT = Online (retained)
23:16:13 MQT: cmnd/ESP02/POWER = 
23:16:13 MQT: tele/ESP02/INFO1 = {"Module":"Sonoff Basic","Version":"5.12.0c","FallbackTopic":"MD client08","GroupTopic":"ESP8266"}
23:16:13 MQT: tele/ESP02/INFO2 = 
{"WebServerMode":"Admin","Hostname":"ESP02","IPAddress":"192.168.1.78"}
23:16:13 MQT: tele/ESP02/INFO3 = {"RestartReason":"Software/System restart"}
23:16:13 MQT: homeassistant/light/ESP02/config =  (retained)
23:16:13 MQT: homeassistant/switch/ESP02/config =  (retained)
23:16:14 MQT: stat/ESP02/RESULT = {"POWER":"0"}
23:16:14 MQT: stat/ESP02/POWER = 0
23:16:21 MQT: tele/ESP02/STATE = {"Time":"2018-02-26T23:16:21","Uptime":"0T00:00:17","POWER":"0","Wifi":{"AP":1,"SSId":"MGTS_GPON_3233","RSSI":58,"APMac":"78:81:02:C7:73:C2"}}
23:16:21 MQT: tele/ESP02/SENSOR = {"Time":"2018-02-26T23:16:21","Switch1":"0"}

closed 2018-02-26 by Igor122017