arendst/Tasmota #2021: Noob Question - New 4ch Pro (R2) looks different than online pictures

amercat37, 2018-02-25:
I purchased my first 4ch Pro (R2) on amazon.  I'm attempting to flash Tasmota to integrate with Domoticz to control my door strikes.  This will replace my old X10 relays.  I think that I have everything figured out, except my 4ch Pro (R2) does not match any online pictures, so I'm not sure where GPIO0 is located.  Unlike all of the online pictures, where the board reads, "sonoff_4ch_pro V1.1 2017-4-26", my board reads "sonoff_4ch_pro V1.0 2017-11-10."  I also have a long bar for my 433mhz antenna, not the black coil of wire.  The ESP8285 is also soldered directly to the board, it's not on the little riser board.  Is this just because I have the R2 model?  Can I just locate GPIO0 on the chip specs and continue?
Thanks

Frogmore42, 2018-02-26:
You might get more responses if you include some pictures.  No one can see what you are seeing.  But, the simple answer is that yes you can find GPIO 0 by looking at the chip.  That is usually hard to solder to, so finding an easier solder point is usually a good idea.  But, even if you get it flashed it might not work if they changed what the GPIOs do.

nmaggioni, 2018-03-07:
This issue could be a dupe of #2018.

@amercat37 Look closely at the version number, you might have missed the _R2_ bit; mine reports `sonoff_4ch_pro`**`R2`**`V1.0 2017-11-10`.

[image]

amercat37, 2018-03-07:
Thanks.  I did miss the R2.  I have also tested and flashing was successful.  Everything appears to be working great.  A few things that I came across that may help others.  1) S6 has to be set to Interlocking, not Locking/Inching to enter programming mode for the chip. If it's set the wrong way, two red leds come on. 2) The relays can not be controlled from the web interface, but everything else appears to work.

nmaggioni, 2018-03-07:
You just need to select the correct Sonoff model in the web config panel _(`Configuration` > `Configure Module` > set `Module Type` to 23)_, then everything will be fine. Switches can be set [as suggested in the wiki](https://github.com/arendst/Sonoff-Tasmota/wiki/Sonoff-4CH-and-4CH-Pro#4ch-pro-dipswitch-configuration) at any point - before or after flashing.

sigalou, 2018-03-11:
For Flash, I did a tutorial on my site (in French). Good continuation.
http://sigalou-domotique.fr/index.php/17-sonoff-4ch-pro-r2

neandr, 2018-05-18:
Good to have found this threat ... also having that new board. Great!
But my problem is to get the flash sw over to the board. It seems the Dell Laptop running Ubuntu and PlatformIO (using Atom) looks for the com port and finds 'only' 
```
Building .pioenvs/sonoff-DE/firmware.bin
Configuring upload protocol...
Looking for upload port...
Use manually specified: COM5
Uploading .pioenvs/sonoff-DE/firmware.bin
error: cannot access COM5
```
Changing the setting upload_port = COM5 on platformio.ini to COM1 .. COM5 doesn't help!?
From different other videos I noticed some 'automatic' search of the com port. But can't figure out how to set that with platformio files.

Any suggestion how to get it successful flashed?

closed 2018-07-10 by ascillato2

howroyd, 2018-09-20:
Similar issue here.
Board recently purchased only seems to have the ESP8285 on it, no other micros.  Also, Button 1 is connected to GPIO0 so you can enter bootloader mode by holding down the bottom button.
Cannot get the thing to flash, however.  Using the Arduino IDE and tasmota release 6-2.
Settings I'm currently trying to no avail are:
Generic ESP8285 module
1M (no SPIFFs)
Debug disabled
IwIP Variant v1.4 Prebuilt
Reset nodemcu
Xtal 26MHz
CPU 80MHz
Upload 115200

Using a buspirate as ISP as I have done for loads of other devices; basic, TH, 2ch, etc.

Grateful for any help.

[image]

howroyd, 2018-09-20:
fixed it, as above but 40MHz xtal.

meingraham, 2018-09-20:
@howroyd

Happy to hear you solved you problem.

FYI - You posted to this closed 4Ch **_Pro_** track. Your device is a 4Ch (not Pro model).