arendst/Tasmota #2022: Open Push Button T1 2CH

ynuyasha, 2018-02-25:
Hi, i have installed a T1 2CH and correctly flashed Tasmota inside.
Second Button works correctly as a Normal Switch and Toggle Light.
First original Button that i removed instead was a Open Push Button that pilot a Relay and if i toggle Light doesn't works.
I tried to Set different PulseTime but without success. The original button is https://www.vimar.com/it/it/catalog/product/index/code/14008

Anyone as successfully works with a Open Push Button configuration?

ynuyasha, 2018-02-25:
my relay is finder 27.01.8.230 http://www.farnell.com/datasheets/2238047.pdf

ynuyasha, 2018-03-05:
Nobody can help me? i think configure one button as open push button is a main feature, so i want to know if someone as successfully do it.

ynuyasha, 2018-06-06:
Up this discussion, hope to have an answer

ascillato, 2018-06-06:
Hi

Sorry that your issue was left unresponded. There were lots of issues.

I don't understand completely your configuration. Please, can you share your status 0?
You removed the original pushbutton and you replaced it with a pushbutton for wall outlet. right? And that pushbutton is configured as button1? and controlling relay1?

ynuyasha, 2018-06-06:
Hi @ascillato ,
this is my status  0

```
13:11:58 CMD: status 0
13:11:58 MQT: stat/sonoff-corridoio/STATUS = {"Status":{"Module":29,"FriendlyName":"IngressoLight","Topic":"sonoff-corridoio","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
13:11:58 MQT: stat/sonoff-corridoio/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://domus1:80/api/arduino/sonoff.ino.bin","Uptime":"6T06:43:43","Sleep":0,"BootCount":22,"SaveCount":111,"SaveAddress":"F9000"}}
13:11:58 MQT: stat/sonoff-corridoio/STATUS2 = {"StatusFWR":{"Version":"5.12.0","BuildDateTime":"2018-02-09T14:56:37","Boot":6,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
13:11:58 MQT: stat/sonoff-corridoio/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"domus1","LogPort":514,"SSId1":"Wifi","SSId2":"Wifi","TelePeriod":300,"SetOption":"55800009"}}
13:11:58 MQT: stat/sonoff-corridoio/STATUS4 = {"StatusMEM":{"ProgramSize":486,"Free":516,"Heap":22,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3}}
13:11:58 MQT: stat/sonoff-corridoio/STATUS5 = {"StatusNET":{"Hostname":"sonoff-corridoio-5247","IPAddress":"192.168.1.108","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"60:00:94:00:54:7F","Webserver":2,"WifiConfig":2}}
13:11:58 MQT: stat/sonoff-corridoio/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.107","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_A1547F","MqttUser":"DVES_USER","MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
13:11:58 MQT: stat/sonoff-corridoio/STATUS7 = {"StatusTIM":{"UTC":"Wed Jun 06 12:11:58 2018","Local":"Wed Jun 06 13:11:58 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":1}}
13:11:58 MQT: stat/sonoff-corridoio/STATUS10 = {"StatusSNS":{"Time":"2018-06-06T13:11:58"}}
13:11:58 MQT: stat/sonoff-corridoio/STATUS11 = {"StatusSTS":{"Time":"2018-06-06T13:11:58","Uptime":"6T06:43:43","Vcc":3.143,"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":1,"SSId":"Wifi","RSSI":74,"APMac":"28:00:82:00:BF:5B"}}}
```

Yes button 1 is attached to relay Finder and configured as pushbutton doesn't works, button 2 is configured as normal switch attached to a light bulb and works very well.

ascillato, 2018-06-06:
> Yes button 1 is attached to relay Finder and configured as pushbutton doesn't works, button 2 is configured as normal switch attached to a light bulb and works very well.

Ups. That confuses me. Please expand your explanation. Can you share a screenshot of the module config page? and also a picture or diagram how do you connect your wires? What is a relay Finder? Button 2 is attached to a light bulb? directly?

ynuyasha, 2018-06-06:
relay finder is this http://www.farnell.com/datasheets/2238047.pdf
[image]
L1 is configured as push button, L2 as switch
If i remove L1 wire from T1 and connect to a normal push button as https://www.vimar.com/it/it/catalog/product/index/code/14008 relay works
[image]

ascillato, 2018-06-06:
ah, finder is the brand! :laughing: I though that was a kind of feature or something. sorry. now I understand that. An external AC relay. Ok.

> L1 is configured as push button, L2 as switch

How do you configure that? switchmode command?

ynuyasha, 2018-06-06:
yes i tried SwitchMode from 3 to 7 with different combination of PulseTime for impulse

ascillato, 2018-06-06:
va bene,

In the main page of Tasmota. If you click on Toggle1, it turn your relay?

ynuyasha, 2018-06-06:
Yes, if i click on toggle 1 animation of push Button works, but bulb Light
stay off. Same if use physical button of T1.

Grazie :)

ascillato, 2018-06-06:
Ok, So pressing the button on T1 makes the main tasmota webpage to toggle?

So the problem seems to be in your relay. If you connect to L1 the light bulb that you have in L2, The light goes on?

ynuyasha, 2018-06-06:
Yes it toggle, and viceversa too.
Yes, if i connect L1 directly to light goes on

ascillato, 2018-06-06:
Great, So the problem seems to be on the connection to the relay.

Can be splitted on the relay the LINE and NEUTRAL wires? Can you measure that with a Tester? (carefully)

ynuyasha, 2018-06-07:
Sorry i don't understand well your questions.
what you mean with can be splitted on the relay? and what you want that i measure with the tester (Voltage,Current)?

If i attach Neutral and L1 wires from T1 to normal push button https://www.vimar.com/it/it/catalog/product/index/code/14008 relay works.

ascillato, 2018-06-07:
Hi,

If your L1 output works from the webpage when you have a direct connection to a bulb, seems to be a problem in the connection of the relay (your Sonoff and Tasmota are woking ok). You have said that if you power the relay from an external pushbutton it works. So may be your relay is expecting neutral and you are sendind Line.

Can you check if you have your connections as the following diagram? Because seems to be that there is a mix in your neutral wires

[image]

Please, be safe. Turn off your main breaker for checking.

ynuyasha, 2018-06-09:
This is the complete schema, there isn't a Netrual on the relay
[image]

ascillato, 2018-06-09:
Hi

Thanks a lot for sharing the complete circuit.

That confirms what I was trying to tell you. In your connection, your relay will turn on only when in the A2 you put NEUTRAL. Your sonoff gives you LINE.
Now you have LINE in A2 and also in A1, so that will NOT turn on the relay. With your physical buttons you are putting NEUTRAL on the relay so that why it works with them.

You need to change your circuit **_INMEDIATELLY_**. As your diagram, if the sonoff turn on the L1Output and you use also the physical button at the same time, everything will blow up (SHORT CIRCUIT)

You need to change several things to make it work. It is a connection issue, not a software issue.
**_(If you do not how to connect it, please please be safe and call an electrician. A sonoff is NOT A TOY)_**

[image]

If your light is below 10 Amps, a better or simplier connection could be the following:

[image]

ynuyasha, 2018-06-11:
I don't understand why you continue to talk about Neutral on the relay?

The relay works only with Line and i don't do it circuit but was made by an electrician.
I only substitute one original push button with T1.
I tested personally wires with a phase detector (screwdriver with little bulb inside) after cut off T1 from circuit.

have you read page 2 of pdf of relay datasheet that i share before where is described wiring diagram?

i'm not an electrician but the black line of the diagram i think was a Phase (and i tested it)

http://www.farnell.com/datasheets/2238047.pdf

I share you the original circuit (all phisical button are open push button )
[image]

ascillato, 2018-06-11:
Hi @ynuyasha 

Yes, I read all the documentation you provide.

But please, don't get mad, I have to say that I'm very Sorry, but I realize that you are missing the basics of electricity. Your actual connection is VERY UNSAFE. Please call an electrician to install the Sonoff. It is not worth to risk yourself, your family and your home for saving just some money and not calling a certified electrician.

Your issue here is a wiring issue, not software issue.

Again, sorry and hope you can manage to install it correctly with an electrician.

closed 2018-07-10 by ascillato2