arendst/Tasmota #2023: Sonoff RF Bridge (5.12.0) doesn't send learnt codes

kovaccs, 2018-02-25:
The bridge have learned PT2260 like remote codes and original Sonoff remote codes but neider send them from web interface nor mqtt.
...
18:22:20 CMD: rfkey16 2
18:22:20 RSL: Received Topic /rfkey16, Data Size 1, Data 2
18:22:20 RSL: Group 0, Index 16, Command RFKEY, Data 2
18:22:20 MQT: stat/sonoff433br_2/RESULT = {"RfKey16":"Start learning"}
18:22:20 SER: Received A0 55
18:22:27 SER: Received A3 27 74 01 40 03 D4 55 45 14 55
18:22:27 MQT: stat/sonoff433br_2/RESULT = {"RfKey16":"Learned"}
18:22:28 CFG: Saved to flash at FA, Count 82, Bytes 1648
18:22:40 WIF: Checking connection...
18:22:40 WIF: Connected
18:23:02 WIF: Checking connection...
18:23:02 WIF: Connected
18:23:05 CMD: rfkey16 4
18:23:05 RSL: Received Topic /rfkey16, Data Size 1, Data 4
18:23:05 RSL: Group 0, Index 16, Command RFKEY, Data 4
18:23:05 MQT: stat/sonoff433br_2/RESULT = {"RfKey16":"Saved"}
18:23:06 CFG: Saved to flash at F9, Count 83, Bytes 1648
...
It seems to be sending, but the remote device not operate.
...
18:59:03 WIF: Connected
18:59:11 RSL: Received Topic /RfKey16, Data Size 0, Data 
18:59:11 RSL: Group 0, Index 16, Command RFKEY, Data 
18:59:11 MQT: stat/sonoff433br_2/RESULT = {"RfKey16":"Learned sent"}
18:59:11 SER: Received A0 55
18:59:18 HTP: Console
18:59:24 WIF: Checking connection...
18:59:24 WIF: Connected
...
Is there anybody's suggestion for the solution? 
Thank's
Kovaccsaba

Jason2866, 2018-02-26:
Try the latest dev 5.12.0c. With that you can show the learned codes with command  RfKey(x) 5
What does this command show? Is the learned code correct? I have Remotes where the Bridge fails to learn these (correct). I dont use the function learning codes anymore. I send the RF codes via mqtt with the command: -> Backlog RfSync Value, RfLow Value, RfHigh Value, RfCode #HexValue

kovaccs, 2018-02-26:
Ok! Thank you your advice, but. 
I can't find binary image 5.12.0.c because I could not compile the project (several errors occurred with ArduinoIDE) despite the fact that I carefully observed the terms of the translation. Sorry! 

The offered RfKey(x) 5 is this a new feature in 5.12.c?
In my version the consol log is:
...
10:33:00 CMD: rfkey11 5
10:33:00 RSL: Received Topic /rfkey11, Data Size 1, Data 5
10:33:00 RSL: Group 0, Index 11, Command RFKEY, Data 5
10:33:01 MQT: stat/sonoff433br_2/RESULT = {"RfKey11":"Learned sent"}
10:33:01 SER: Received A0 55
10:33:18 WIF: Checking connection...
10:33:18 WIF: Connected
...
The result does not sound too good :-(
There is a lack of opportunity to see the codes learned.
Thank's
Kovaccsaba

Jason2866, 2018-02-26:
May you install Atom.io? It is really a nice development plattform. And very easy to handle complex projects. So for Tasmota it is only download the zip extract this complete folder. In Atom.io "open project folder" and thats it. No manually copying files.

Jason2866, 2018-02-26:
Different approach, do you know the codes of the remotes? 
Are the maybe programmable Transmitter and Receivers? 
If yes you could it programm via the RF Bridge.

kovaccs, 2018-02-26:
Ok Jason2866 and thank you!
First: I'm not a developer, just an enduser, who interested in sonoff hardvare stuff and wanted to use them controlled in Node-RED flows. That's why I find Tasmota project and Espurna project also. I tried both of them, but I have experienced that the Tasmota MQTT is far better than Espurna MQTT client. Tasmota MQTT is reliable while the other is not because of the socket error:

1519672536: New connection from 172.30.1.81 on port 1883.
1519672536: Client ESPURNA_82 disconnected.
1519672536: New client connected from 172.30.1.81 as ESPURNA_82 (c0, k60, u'pi').
1519672711: Client ESPURNA_82 has exceeded timeout, disconnecting.
1519672711: Socket error on client ESPURNA_82, disconnecting.
1519673127: New connection from 172.30.1.81 on port 1883.
1519673127: Client ESPURNA_82 disconnected.
1519673127: New client connected from 172.30.1.81 as ESPURNA_82 (c0, k60, u'pi').
1519673599: Client ESPURNA_82 has exceeded timeout, disconnecting.
1519673599: Socket error on client ESPURNA_82, disconnecting.
1519673720: Saving in-memory database to /var/lib/mosquitto/mosquitto.db.
1519673900: New connection from 172.30.1.81 on port 1883.

So. I definitly want to use Tasmota image on Sonoff 433RF Bridge controlled by Node-Red MQTT.

Second: I had installed  platform.io developper projet containing Atom, but there was more comfortable for me flashing with binary image. I have v5.12.0 uploaded to the device and today I can't find  source newer then this one. (Source: https://github.com/arendst/Sonoff-Tasmota/releases )
If you are so kind to translate a better version for me I would be happy. 
I'm not using the maddox buid yet....    (appologize beeng my poor english!)
Kovaccsaba

Jason2866, 2018-02-26:
No need to excuse for your english. My english is bad...
I will compile a universal build for you this weekend.
I am on a business trip during the week.

kovaccs, 2018-02-26:
All right! Waiting for you and have a nice business trip!  :-) 
I hope your version is more useful about RF codes...
Kovaccsaba

kovaccs, 2018-02-26:
If you could help me, I would like to know these and how to get them:
1. stored RF codes, and that is they are realy stored. In the consol message somthing appear after Rfkey(x) 2. Unfortunately this code can not be used control remote device.

2. exactly how to send them trough mqtt message (RfCode) What does it mean 1..8388607 or #1..7FFFFF. That's only the "Data" part of RF codes, is it? What about Sync, Low, High?

3. Your are included something one of your previous notes:

"I dont use the function learning codes anymore. I send the RF codes via mqtt with the command: -> Backlog RfSync Value, RfLow Value, RfHigh Value, RfCode #HexValue". 

If I undersand correctly this is a broker (mosquitto?) pub cli command? How write it exactly (correct syntax example pls.).
And many-many more..... sorry!  ;-)

Jason2866, 2018-02-27:
Look at the wiki from @arendst to OpenHab
https://github.com/arendst/Sonoff-Tasmota/wiki/openHAB
It is a good starting point for learning and get the answers to your questions.

kovaccs, 2018-02-27:
oh yes, I almost learned the whole wiki....

Jason2866, 2018-02-27:
I needs some time, yes. But without basic understanding of mqtt tasmota is nearly useless or just
the wrong solution....
You find the commands here
https://github.com/arendst/Sonoff-Tasmota/wiki/Commands

kovaccs, 2018-02-27:
I'm using this mqtt command and love them, but unfortunately I can't send out (mqtt pub) rf codes  at all.
that's my problem.

Jason2866, 2018-02-27:
Which mqtt broker do you use? On a RasPi?

kovaccs, 2018-02-27:
Mosquitto on Raspberri Pi. It was always reliable.

kovaccs, 2018-02-27:
I have other Sonoff devices (4ch, 4ch pro) working well with mqtt commands. Just this rf bridge annoing me .

Jason2866, 2018-02-27:
Ok, good so mqtt is working fine, so just the Rf Bridge is the problem.
Try this command to send (via mosquitto)
cmd: RfCode #150551
What happens?

kovaccs, 2018-02-27:
Ok! Try it.....
One of my device is switched ON  #9/16 button!  Hurray!!!

Node-RED said: 
2/27/2018, 8:04:18 PMnode: b4773236.c77a58
stat/sonoff433br_2/RESULT : msg : Object
{ topic: "stat/sonoff433br_2/RESULT", payload: "{"RfCode":"#150551"}", qos: 0, retain: false, _msgid: "4253a5f6.bdac5c" }

kovaccs, 2018-02-27:
The #150550 switch OFF the sane device.... GOOOOOD!

kovaccs, 2018-02-27:
Then the question is how do I know these codes?

Jason2866, 2018-02-27:
😀
So everthing can now be done. Just press the remote and watch the code on the RfBridge.
Just send this code via mqtt broker and thats it. 
See here
20:22:34 MQT: tele/sonoff-8E8329/RESULT = {"RfReceived":{"Sync":11330,"Low":370,"High":1080,"Data":"15155F","RfKey":"None"}}

kovaccs, 2018-02-27:
Ok, I will...

kovaccs, 2018-02-27:
Trouble!!!!
The first four remote data is same! #0575D5  (1/16, 2/16, 3/16, 4/16) 
and if I sent them trough mosquitto nathing happened.
Top of all that the 9/16 and 10/16 both  #1575D5, this is different than your offered code. :-O

Jason2866, 2018-02-27:
What happens when you send the code  #1575D5. It is ok thats the code differs. It is a other protocol.
Can you copy the fully received results from the web console?

kovaccs, 2018-02-27:
If I sent this code, none f them divider responds

The first buttun used:
21:03:07 SER: Received A4 27 7E 01 4A 03 CA 05 05 15 55
21:03:07 MQT: tele/sonoff433br_2/RESULT = {"RfReceived":{"Sync":10110,"Low":330,"High":970,"Data":"050515","RfKey":"None"}}
21:03:08 SER: Received A4 23 50 01 40 03 48 05 05 15 55
21:03:09 SER: Received A4 23 1E 01 54 03 3E 05 05 15 55
21:03:09 SER: Received A4 23 28 01 54 03 3E 05 05 15 55
21:03:09 MQT: tele/sonoff433br_2/RESULT = {"RfReceived":{"Sync":9000,"Low":340,"High":830,"Data":"050515","RfKey":"None"}}
21:03:10 SER: Received A4 23 1E 01 54 03 3E 05 05 15 55
21:03:24 WIF: Checking connection...

The 9th button pushed:
21:07:34 SER: Received A4 27 9C 01 40 03 D4 15 05 55 55
21:07:34 MQT: tele/sonoff433br_2/RESULT = {"RfReceived":{"Sync":10140,"Low":320,"High":980,"Data":"150555","RfKey":"None"}}
21:07:34 SER: Received A4 23 5A 01 36 03 5C 15 05 15 55
21:07:34 MQT: tele/sonoff433br_2/RESULT = {"RfReceived":{"Sync":9050,"Low":310,"High":860,"Data":"150515","RfKey":"None"}}
21:07:35 SER: Received A4 23 32 01 40 03 48 15 05 15 55
21:07:35 SER: Received A4 23 32 01 40 03 48 15 05 15 55
21:07:36 SER: Received A4 23 32 01 40 03 48 15 05 15 55

Jason2866, 2018-02-27:
The timing is different here
RfReceived":"Sync":10110,"Low":330,"High":970,"Data":"050515"
You have to send the parameters sync low and high before you send the Data = RfCode

Jason2866, 2018-02-27:
You can do it all in one command with backlog
Send multiple MQTT commands at once
To change connectivity configuration over MQTT you can (should) use the Backlog command to send multiple commands to your Sonoff(s) at once and prevent reboot between the commands. For example you can change the wifi SSID and Password in one step by sending the following MQTT message to your sonoff's Backlog topic:

mosquitto_pub -t 'cmnd/yoursonoff/Backlog' -m 'ssid1 yournewssid; password1 yournewpassword'

Your Sonoff will be rebooted after both settings are written to the configuration, so your device will connect to the new network. You can also change the MQTT connection parameters the same way and of course any other settings.

kovaccs, 2018-02-27:
I don't understand why have to give ssid, and password?   Is it possible to give parameters as a json string because of I not handle mosquitto its command line than from Node-RED mosquitto output nodes.
This nodes accept the msg.payload containig mosquitto pub message. I don't know it is possible to handle Backlog topic, but I'll try it.
Could you show me how to format the multiply command payload in json string?

kovaccs, 2018-02-27:
Ok! I sent paylod {"Sync":10140,"Low":320,"High":980,"Data":"150555"} to topic cmnd/sonoff433br_2/Backlog

My device replyed (trough Node-RED consol) two messages:
2/27/2018, 9:54:44 
PMnode: b4773236.c77a58stat/sonoff433br_2/RESULT : msg : Object
object
topic: "stat/sonoff433br_2/RESULT"
payload: "{"Backlog":"Appended"}"
qos: 0
retain: false
_msgid: "a340b149.5cbf5"

2/27/2018, 9:54:44 
PMnode: b4773236.c77a58
stat/sonoff433br_2/RESULT : msg : Object
object
topic: "stat/sonoff433br_2/RESULT"
payload: "{"Command":"Unknown"}"
qos: 0
retain: false
_msgid: "c255f2b9.3daa1"

Something happened with backlog, but missformed the payload.

kovaccs, 2018-02-27:
Looking at the mosquitto log file, there is no record.

Jason2866, 2018-02-27:
At JSON Path i have to learn. Here i am a beginner!
This works on the web console 
Backlog RfSync 11320; RfLow 360; RfHigh 1090; RfCode #150551

Jason2866, 2018-02-27:
You can send every command in sequence this works too. Last command has to be the data = rfcode

kovaccs, 2018-02-27:
Oh! It really works, but sorry this does not help me because the Node-RED flow has to issue commands .... :-(

Jason2866, 2018-02-27:
Sorry for the wifi ssid thing. I copied that from the wiki. This is just the example for the command Backlog how it works...

kovaccs, 2018-02-27:
Next I'll learn more from mosquitto broker, and try to construct expression for working.
You helped a lot - more then lot!  Thank you. I'll refer if i find solutoin, because of it'll be useful others fighting this problem....

kovaccs, 2018-02-27:
Yes exactly!

kovaccs, 2018-02-27:
Ok! Solution so symple. Send same data then in webconsol to backlog topic
MQTT Topic:  cmnd/sonoff433_2/backlog
msg.payload = "RfSync 11320; RfLow 360; RfHigh 1090; RfCode #150551"

Node-RED debug consol said:
2/27/2018, 11:06:47 
PMnode: b4773236.c77a58stat/sonoff433br_2/RESULT : msg : Object
object
topic: "stat/sonoff433br_2/RESULT"
payload: "{"Backlog":"Appended"}"
qos: 0
retain: false
_msgid: "3a53bed9.c5ac42"
2/27/2018, 11:06:48 
PMnode: b4773236.c77a58stat/sonoff433br_2/RESULT : msg : Object
object
topic: "stat/sonoff433br_2/RESULT"
payload: "{"RfSync":11320}"
qos: 0
retain: false
_msgid: "f82b452d.07d4b8"
2/27/2018, 11:06:48 
PMnode: b4773236.c77a58stat/sonoff433br_2/RESULT : msg : Object
object
topic: "stat/sonoff433br_2/RESULT"
payload: "{"RfLow":360}"
qos: 0
retain: false
_msgid: "a7ac1286.5853f"
2/27/2018, 11:06:48 
PMnode: b4773236.c77a58stat/sonoff433br_2/RESULT : msg : Object
object
topic: "stat/sonoff433br_2/RESULT"
payload: "{"RfHigh":1090}"
qos: 0
retain: false
_msgid: "582e1ffc.a7d1e"
2/27/2018, 11:06:48 
PMnode: b4773236.c77a58stat/sonoff433br_2/RESULT : msg : Object
object
topic: "stat/sonoff433br_2/RESULT"
payload: "{"RfCode":"#150551"}"
qos: 0
retain: false
_msgid: "a5e5e8cf.5a1a18"
2/27/2018, 11:06:51 
PMnode: b4773236.c77a58
tele/sonoff433br_2/RESULT : msg : Object
object
topic: "tele/sonoff433br_2/RESULT"
payload: "{"RfReceived":{"Sync":10150,"Low":330,"High":970,"Data":"1575D5","RfKey":"None"}}"
qos: 0
retain: false
_msgid: "ab6923d8.5496e"

And the remote device operate !  :-D :-D

kovaccs, 2018-02-27:
Many thank's for Jason2866! Solved.

closed 2018-02-27 by kovaccs