arendst/Tasmota #2025: Get the temperature reading from sensor in MQTT

L00000813, 2018-02-25:
Below is a publish command where I can switch my device on. (It works)

mosquitto_pub -h control_central -t cmnd/my_house_living_room/power -m 0

Is there a command which returns the temperature of the sensor connected to my Sonoff. I am using the TH16 with a DS18B20 sensor. I believe I have everything set up correctly, however I cant find a command to retrieve the temperature?

Is there a publish command like the one above which will return the temperature?

Cheers.

Frogmore42, 2018-02-26:
Every teleperiod time, the device will send an mqtt message with the status of the sensors that have been configured.  It has the advantage of automatically setting the loop time.

L00000813, 2018-02-26:
I do not get any data back from my Sonoff. 
Can you provide the command(s) which I need to type in order to get this data back?
Are they pub or sub commands?
What I really want here is a pub command which will return data when I request it? Is this possible?

FYI: I have taken the Sonoff basic code, appended the Network details and uploaded it to the Sonoff TH16 MCU(with temp sensor). I then opened the web interface and configured that. The device rebooted. I have not added or configured any other file. Is this correct? Am I missing something?
Any advice will be appreciated.
Gavin

curzon01, 2018-02-26:
See [Commands - Management](https://github.com/arendst/Sonoff-Tasmota/wiki/Commands#management) - Status 10.

If you publish `Status 10` it will immediately response with a `SENSOR` info.

With your MQTT topics it might be
`cmnd/my_house_living_room/Status -m 10`

should response with
`tele/my_house_living_room/SENSOR`

L00000813, 2018-02-26:
Ok, I have tried the /Status -m 10, The command is published, but I get nothing returned.

Do I have to set something in the 'code' so that the ESP can return messages. To Date I have only been switching on/off devices with success.  I have not received any feedback from the device, i.e status?

It seems that I have a problem getting any type of status back. Even checking if a device is on/off.

Any ideas what I have failed to do here to get this working?

As a side note; below is the configuration I made to the 'basic sonoff configuration' so I can get sensor data. I think I am correct?!

[image]

L00000813, 2018-02-26:
pi@raspberrypi:~ $ mosquitto_pub -h 192.168.1.15 -t cmnd/sensor/power -m 1
pi@raspberrypi:~ $ mosquitto_pub -h 192.168.1.15 -t cmnd/sensor/power -m 0
pi@raspberrypi:~ $ mosquitto_pub -h 192.168.1.15 -t cmnd/sensor/status -n

The top two command turn on/of my Sonoff. (working)

The 3rd command returns nothing?? I am just trying to find the relay state in this example.
pi@raspberrypi:~ $ mosquitto_pub -h 192.168.1.15 -t cmnd/sensor/status -n

Any ideas why i cannot get the status of my Sonoff? The relay state. (and/or the temperature)

curzon01, 2018-02-26:
#### Terminal 1
`$ mosquitto_sub -h 192.168.1.15 -t +/sensor/#` -v

#### Terminal 2
`$ mosquitto_pub -h 192.168.1.15 -t cmnd/sensor/status -m 10`
and now have a look what is returned on Terminal 1

I think you do not care about the correct MQTT topics:
* commands are send to **cmnd**/sensor/...
* status are returned as **stat**/sensor/*
* telemetry data are returned as **tele**/sensor/*

**Please** - before continue tell us what you have configure for MQTT. Use the web interface, click on "Console", type `STATUS 0` [return] in the input field and post the results (all STATUS messages).

L00000813, 2018-02-26:
Ok, so the above commands work, but i need two terminals. I want to get the status with only one terminal on my Raspberry Pi?

As for the temperature - here is the log you requested.

Hopefully you can help me further, thanks.

"Console", type STATUS 0 [return]"

00:00:00 Project sensor Sonoff (Topic sensor, Fallback DVES_870CD8, GroupTopic sonoffs) Version 5.12.0c-2_4_0_RC2
00:00:00 WIF: Connecting to AP2 vodafone-7AB7 in mode 11N as sensor-3288...
00:00:06 WIF: Connected
00:00:06 DNS: Initialized
00:00:06 HTP: Web server active on sensor-3288.local with IP address 192.168.1.10
17:45:02 MQT: Attempting connection...
17:45:04 MQT: Connected
17:45:04 MQT: tele/sensor/LWT = Online (retained)
17:45:04 MQT: cmnd/sensor/POWER = 
17:45:04 MQT: tele/sensor/INFO1 = {"Module":"Sonoff TH","Version":"5.12.0c","FallbackTopic":"DVES_870CD8","GroupTopic":"sonoffs"}
17:45:04 MQT: tele/sensor/INFO2 = {"WebServerMode":"Admin","Hostname":"sensor-3288","IPAddress":"192.168.1.10"}
17:45:04 MQT: tele/sensor/INFO3 = {"RestartReason":"Software/System restart"}
17:45:04 MQT: homeassistant/light/sensor/config =  (retained)
17:45:04 MQT: homeassistant/switch/sensor/config =  (retained)
17:45:05 MQT: stat/sensor/RESULT = {"POWER":"OFF"}
17:45:05 MQT: stat/sensor/POWER = OFF
17:45:12 MQT: tele/sensor/STATE = {"Time":"2018-02-26T17:45:12","Uptime":"0T00:00:18","Vcc":3.461,"POWER":"OFF","Wifi":{"AP":2,"SSId":"vodafone-7AB7","RSSI":84,"APMac":"78:6A:89:D2:7A:C8"}}
17:46:15 MQT: stat/sensor/RESULT = {"POWER":"ON"}
17:46:15 MQT: stat/sensor/POWER = ON
17:46:20 MQT: stat/sensor/RESULT = {"POWER":"OFF"}
17:46:20 MQT: stat/sensor/POWER = OFF
17:47:00 MQT: stat/sensor/RESULT = {"POWER":"ON"}
17:47:00 MQT: stat/sensor/POWER = ON
17:47:09 MQT: stat/sensor/RESULT = {"POWER":"OFF"}
17:47:09 MQT: stat/sensor/POWER = OFF
17:48:00 CMD: STATUS 0
17:48:00 MQT: stat/sensor/STATUS = {"Status":{"Module":4,"FriendlyName":"Sonoff","Topic":"sensor","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
17:48:00 MQT: stat/sensor/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://domus1:80/api/arduino/sensor.ino.bin","Uptime":"0T00:03:06","StartupUTC":"2018-02-26T16:44:54","Sleep":0,"BootCount":24,"SaveCount":88,"SaveAddress":"F4000"}}
17:48:00 MQT: stat/sensor/STATUS2 = {"StatusFWR":{"Version":"5.12.0c","BuildDateTime":"2018-02-25T10:24:41","Boot":6,"Core":"2_4_0_RC2","SDK":"2.1.0(7106d38)"}}
17:48:00 MQT: stat/sensor/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"domus1","LogPort":514,"SSId1":"Wi-Fi-Raspberry","SSId2":"vodafone-7AB7","TelePeriod":300,"SetOption":"00000009"}}
17:48:00 MQT: stat/sensor/STATUS4 = {"StatusMEM":{"ProgramSize":510,"Free":492,"Heap":15,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3}}
17:48:00 MQT: stat/sensor/STATUS5 = {"StatusNET":{"Hostname":"sensor-3288","IPAddress":"192.168.1.10","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"EC:FA:BC:87:0C:D8","Webserver":2,"WifiConfig":3}}
17:48:00 MQT: stat/sensor/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.15","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_870CD8","MqttUser":"","MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
17:48:00 MQT: stat/sensor/STATUS7 = {"StatusTIM":{"UTC":"Mon Feb 26 16:48:00 2018","Local":"Mon Feb 26 17:48:00 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":1}}
17:48:00 MQT: stat/sensor/STATUS10 = {"StatusSNS":{"Time":"2018-02-26T17:48:00"}}
17:48:00 MQT: stat/sensor/STATUS11 = {"StatusSTS":{"Time":"2018-02-26T17:48:00","Uptime":"0T00:03:06","Vcc":3.459,"POWER":"OFF","Wifi":{"AP":2,"SSId":"vodafone-7AB7","RSSI":68,"APMac":"78:6A:89:D2:7A:C8"}}}

L00000813, 2018-02-26:
I have managed to get the temperature reading at last. There was a bad connection within my Sonoff which was preventing the sensor from collecting/transmitting data.

Thanks for any helpful comments.

closed 2018-02-26 by L00000813

eadin79, 2018-12-17:
@ghost: How you discover and what... I faced same issues: in console is ok, in mosquitto_pub I can change statuses, but in mosquitto_sub I cannot get anything....

Frogmore42, 2018-12-17:
@eadin79 it sounds like you are not subscribed to the correct topic.

PepeTheFroggie, 2019-08-28:
tele/sonoff/SENSOR

Dont ask. Took me 2 hrs. Wildcards are nice to discover topics.
tele/+/+

wexivi, 2021-01-11:
This works for me:

mosquitto_pub:  -t "cmnd/sonoff/status" -m 10
mosquitto_sub -t "stat/sonoff/#"

Sonoff TH16 answered:
{"StatusSNS":{"Time":"2021-01-11T17:39:27","SI7021":{"Temperature":23.2,"Humidity":25.3,"DewPoint":2.2},"TempUnit":"C"}}

rundekugel, 2021-10-11:
> Dont ask. Took me 2 hrs. Wildcards are nice to discover topics. try "tele/+/+" or "+/+/+"

or simply: "#"